PiSCSI/piscsi #1098: Additional configuration options for DaynaPort emulation

akuker, 2023-02-18:
# Info

From Discord user "speakers#7740"

> I've been messing with piscsi Daynaport ethernet to write a driver for NetBSD. In particular, this is to provide ethernet on my SE30. The driver is based on the age-old se driver for Cabletron EA41x. I run NetBSD 3.1 on my 68k macs since it's reliable and reasonably fast -- although (amazingly) the latest version (9)  will install and run.
> 
> I have a functional driver, but I've also needed to modify the piscsi code in two respects. Firstly, the send delay workaround for MacOS needs to be disabled. Secondly, the minimum packet size needs to be raised from 64 to 128 so that small packets - particularly ARPs -- are exchanged correctly. The changes are dynamic and depend on the behavior of the client OS.  I can provide a diff.

akuker, 2023-02-18:
Note that this needs to be updated on the fly, since the Mac will boot into MacOS first, then switch over to BSD

speakers-k64, 2023-02-18:
Here's diffs that enables both MacOS and NetBSD to use the DaynaPort emulation:
[bsd.patch](https://github.com/PiSCSI/piscsi/files/10774924/bsd.patch)

benjamink, 2023-11-08:
> @benjamink I'm aware that you have already spent a lot of time on #1306, but it would be great if you could also help here. I would like to verify whether the Mac really needs a work-around for the packet size. Most work-arounds in the piscsi code have turned out not to be required and some have even made things worse, so we should verify this. In case you agree I would modify the existing "testing_daynaport" branch accordingly and would ask you for an additional test with this branch.

@uweseimet absolutely!  Let me know what you need.

uweseimet, 2023-11-08:
@benjamink Thank you! As far as I can tell two things have to be tested, and I created a new branch testing_daynaport_2 for this. This branch is identical to issue_1306, which you have tested earlier, except for a work-around I removed. I would like to know whether networking still works for you with this branch.

benjamink, 2023-11-08:
@uweseimet commit `0e77ab3fbdad14871d6ff0f02ddc63c935fd6c49` failed.  

The Mac boots quickly without getting stuck on the AFP share mount but once the desktop loads the share is not mounted & an error pops up on the screen (I'll attach screenshot  in a moment).

[piscsi-issue-1088.log](https://github.com/PiSCSI/piscsi/files/13299668/piscsi-issue-1088.log)

uweseimet, 2023-11-08:
@benjamink OK, thank you. I just reverted the change and now applied a different change in order to test the second work-around regarding the packe size. Please update your testing_daynaport_2 branch and test again. No "make clean" required, just "make".

benjamink, 2023-11-08:
@uweseimet commit `5ab222acfe26fb9ab9f342f3ec0413c0d0cac8a3` works

uweseimet, 2023-11-08:
@benjamink Once more, thank you for testing. This means that the Mac would be fine with unconditinally setting the packet size to a minumum of 128 bytes.
If this also works for the Atari this current value can simply be changed and it will work with all platforms without any platform-specific code.

benjamink, 2023-11-08:
No problem.  Just let me know if you need me to run any more.

uweseimet, 2023-11-08:
@benjamink Not currently, thank you for the offer.

uweseimet, 2023-11-08:
@anodynesoftware Just a quick question: Before we spend time on a trial-and-error approach, maybe you know by heart or can check whether your driver is fine with a minimum packet siize of 128 instead of 64 bytes?

uweseimet, 2023-11-10:
OK, with some help I have managed to use the daynaport emulation with an Atari. I have just tested transfers with a minimum packet size of 128 bytes and there has not been an obvious issue.

uweseimet, 2023-11-10:
Development note: For a work-around like the one suggested the initiator ID has also to be remembered. Otherwise there would be issues with more than one initiator on the same SCSI chain.

uweseimet, 2023-11-14:
@speakers-k64 In the latest develop branch the minimum packet size is 128 bytes now. The other change you suggested might also be added, but needs to take the initiator ID into account (see my previous comment).
Can you please test whether the current develop branch works when you remove the DaynaPort relay? Please note that we depend on your feedback for changing this code. Without feedback we cannot we cannot test this change and would have to discard this ticket.

anodynesoftware, 2023-11-16:
On 8 Nov 2023 at 14:08, Uwe Seimet wrote:

> @anodynesoftware Just a quick question: Before we spend time on a
> trial-and-error approach, maybe you know by heart or can check whether your
> driver is fine with a minimum packet siize of 128 instead of 64 bytes?
> 
Sorry for the delay in responding to this.  All the driver cares about is that 
the encapsulated Ethernet packet size is within the range specified by the 
Ethernet standard.  So if a packet is padded out to 128 bytes rather than 64 it 
should make no difference.

Roger

uweseimet, 2023-11-16:
@anodynesoftware Thank you for the feedback. Previous testing has confirmed this (eventually with a lot of help I was able to set up a running network on my TT). For PiSCSI the size has already been increased in the develop branch.

uweseimet, 2023-12-04:
I am quite sure that you can configure this for a Mac .With other platforms this is definitely possible, e.g. with regular PCs, Ataris and various workstations.
My goal is not to implement platform-specific solutions or work-arounds when a general general solution is possible.