PiSCSI/piscsi #1099: Request: Tree/Folder view for WebUI

grichter, 2023-02-18:
I have sorted my iso files into a sub folder of images and would like to have a collapsable tree view similar to how zip files are handled in the images section of the WebUI, please? :)

rdmark, 2023-02-18:
What I'd like to propose here is to generate one table for each directory including the images root, and display them sequentially. Each of them except for the images root is wrapped in a <details> block just like the help text. That structure should degrade gracefully for vintage browsers.

rdmark, 2023-02-20:
[image]

Here is a proof of concept. It can be styled better of course, but it does work pretty well!

rdmark, 2023-02-20:
Code in https://github.com/PiSCSI/piscsi/pull/1102

rdmark, 2023-02-20:
Thanks for testing! 

The upload dir request: The tricky part here would be to figure out a UI that makes sense, since the Upload form is used for three distinct destinations.

The move button request: This is already possible with the Rename button. If you prefix the target file name with a dir structure the Web UI will create those dirs if needed and move the renamed file there.

rdmark, 2023-02-21:
[image]

How about this for choosing upload subdir in images?

grichter, 2023-02-21:
Looks like a good place to start

fdanapfel, 2023-02-21:
> [image]
> 
> How about this for choosing upload subdir in images?

Looks good to me.

rdmark, 2023-02-25:
PR merged.

closed 2023-02-25 by rdmark