PiSCSI/piscsi #1101: PiSCSI web interface can't format 4GB HFS disk image... hformat: /home/pi/images/4GB.hda: invalid partition map (Invalid argument)

that-ben, 2023-02-19:
In PiSCSI web interface, trying to create a HFS disk image larger than 2GB results in this error: hformat: /home/pi/images/4GB.hda: invalid partition map (Invalid argument)

It looks like it's using <= System 7.1 limitations of the HFS format, which changed from 2GB to 4GB in System 7.5 and then changed again in System 7.5.2 (PPC only) which allowed a whopping 2TB (2000GB).

It would be nice if PiSCSI could at the very least support closer to 4GB.

In the meantime, for anybody coming here for a solution in creating a larger disk image, @rdmark suggested to create a disk image WITHOUT A FORMAT (do not select HFS) and then once you're booted on the old Mac OS, initialize it as HFS (through the Finder or through the DRIVE SETUP utility app).

Thx

rdmark, 2023-02-23:
The real error is in hfdisk. It can't presently handle file sizes >2GB on 32 bit systems.
```
$ hfdisk 2049.hds
hfdisk: can't open file '2049.hds'  (Value too large for defined data type)
```

Reading up on this, hfdisk currently uses `open()` to read the file in one go. Supposedly, another approach is to use `fopen()` to stream the file instead, but it's not a drop in replacement. Portability also becomes tricky potentially.

- https://www.man7.org/linux/man-pages/man2/open.2.html
- https://www.man7.org/linux/man-pages/man3/fopen.3.html

The (very bare bones) implementation is in `open_device()` in [io.c](https://github.com/bfranske/hfdisk/blob/a7659ed434abc8fb6994498e9e0e6e2c40bfbf51/io.c#L271)

rdmark, 2023-02-23:
And to answer your question, I've tested that hfdisk is able to operate on very large image files (10s of GB) on a 64bit OS (RPi Zero W2)

rdmark, 2023-02-24:
Upstream issue filed https://github.com/bfranske/hfdisk/issues/2

rdmark, 2023-02-24:
So unfortunately, even the error handling is due to hfdisk. Our code is doing the right thing to capture the stderr, but hfdisk errors go to stdout, which is why we log them but don't capture them as errors and display in the front end.

rdmark, 2023-02-24:
Let me close this ticket without action in piscsi. I'll look into improving hfdisk.

closed 2023-02-24 by rdmark