PiSCSI/piscsi #1110: Add the SCSI version as a configuration option

akuker, 2023-02-27:
# Info
There have been a couple point solutions (#828 and #1109) to add support for SCSI-1 in the inquiry response. These solutions work for HD's and CD's, but this should expanded to all device types and should not be based upon file name extension.

Proposed solution is to add a configuration parameter, similar to the Vendor/Product/Revision options, that allows the SCSI version to be specified. This should be an option that applies to all devices, not just HD and CD.

The web interface should also be updated to allow the user to select which SCSI version.

akuker, 2023-02-27:
We should look at making a generic configuration input that allows the user to specify key/value pairs as configuration options. We're getting a lot of command line arguments in scsictl.

Something like: (-a for argument, maybe... there aren't many letters left)
scsictl -c attach -i 3 -f ~/images/drive.hda -a SCSI_VERSION:1

uweseimet, 2023-02-27:
@akuker Are there any real-world scenarios (user reports) where the current solution does not work?

uweseimet, 2023-03-12:
I suggest to close this issue as "won't" do, based on my last comment: There are no other SCSI-1 device types than the ones already covered.

rdmark, 2023-04-16:
@uweseimet I agree that introducing a SCSI level selection right now isn't high priority, since the known corner cases have been covered by the special file ending code paths now. I suggest we revisit this if unhandled SCSI level corner cases are discovered.

I still think it's worth discussing *quirks mode* in general as @akuker suggests. We have one major pending compatibility issue with a known workaround (see https://github.com/PiSCSI/piscsi/issues/741 ) that may benefit from introducing a quirks mode option. Thoughts?

uweseimet, 2023-04-16:
@rdmark IMHO this is not primarly about priorities. PiSCSI should not emulate SCSI-1 device types that have never existed, because (unless somebody proves the opposite) there is no computer that would expect such a device.

Regarding any kind of quirks mode you might end up with a bunch of work-arounds where nobody knows which mode to use when, and what it means to have a combination of these modes. The original RaSCSI code contained many strange work-arounds, and instead of improving compatibility they just did the opposite.
Often when you think a work-around is needed in practice this indicates that the underlying problem has not fully been analyzed. In the case of https://github.com/PiSCSI/piscsi/issues/741, why would artifically slowing down the rate at which PiSCSI executes commands be an improvement? I doubt that changing MIN_EXEC_TIME is the right solution but rather think the core of the problem has not sufficiently been analyzed. And users are still waiting for feedback on their SCSIMON logs, as far as I can tell.

akuker, 2023-10-14:
Withdrawing this ticket.

closed 2023-10-14 by akuker