PiSCSI/piscsi #1118: Minor issue unzipping images

Pacjunk, 2023-03-04:
# Info

- Which version of Pi are you using: zeroW
- Which github revision of software: develop
- Which board version: fullspec
- Which computer is the PiSCSI connected to: n/a. WebUI

# Describe the issue

To reproduce:
   - unzip an image and its matching properties file
   - unzip the same image again
   - a properties files will now show up in the images list

What will happen is that the first unzip process will extract both files into the images directory, then move the properties file into the config directory. On the second unzip, the image is overwritten, but when the script goes to move the properties file into the config directory it will fail as there is already a file there of that name, so the properties file will stay in the images directory.

Maybe the code needs to be modified to overwrite any existing file in the config directory, rather than causing this error.

Cheers,

rdmark, 2023-03-05:
Good catch. This should fix it. Please test! https://github.com/PiSCSI/piscsi/pull/1120

As an added boon, I've improved the error handling as well.

Pacjunk, 2023-03-06:
Seems to be OK. Thanks

rdmark, 2023-03-06:
Very good, thanks!

closed 2023-03-06 by rdmark