PiSCSI/piscsi #1125: Kurzweil K2000RS OS3.87j Not Loading Drives

Cubeinthebox, 2023-03-13:
# Info

- Which version of Pi are you using: Linux 5.15.84+ armv6l
- Which github revision of software: 23.2.1
- Which board version: Raspberry Pi Zero W Rev 1.1
- Which computer is the PiSCSI connected to: Kurzweil K2000RS OS 3.87j

# Describe the issue
Accessing SCSI devices via PiSCSI causes OS to reboot. In some instances I am able to access the directory of the ID and very rarely am able to load from the device. This was only after changing the wait time as per [this issue](https://github.com/PiSCSI/piscsi/issues/741). 

ZuluSCSI may have fixed a similar issue in firmware 1.0.7:
[image]

Here's is a video of what occurs when the system reboots:

https://user-images.githubusercontent.com/78498382/224714022-2660a183-1a96-4ff2-9d4f-d259a99b00d9.mp4

Associated log files I pulled from various tests:

**Able to enter directory here**
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.847] [trace] (ID 1) - ++++ Starting processing for initiator ID 6
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.847] [trace] (ID 1) - Selection phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.848] [trace] (ID 1) - Message Out phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.848] [trace] (ID 1) - Receiving data, transfer length: 1 byte(s)
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.848] [trace] (ID 1) - Phase: msgout
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.848] [trace] (ID 1) - Received IDENTIFY message for LUN 0
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.848] [trace] (ID 1) - Command phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.848] [debug] (ID 1) - Controller is executing ReadCapacity10, CDB $25000000000000000000
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.848] [debug] (ID:LUN 1:0) - Device is executing ReadCapacity10 ($25)
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.849] [trace] (ID:LUN 1:0) - Device is ready
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.849] [trace] (ID 1) - Entering Data In phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.849] [trace] (ID 1) - Sending data, offset: 0, length: 8
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.849] [trace] (ID 1) - Moving to next phase: datain
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.849] [trace] (ID 1) - Status Phase, status is $00
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.849] [trace] (ID 1) - Sending data, offset: 0, length: 1
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.849] [trace] (ID 1) - Moving to next phase: status
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.849] [trace] (ID 1) - Message In phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.850] [trace] (ID 1) - Sending data, offset: 0, length: 1
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.850] [trace] (ID 1) - Moving to next phase: msgin
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.850] [trace] (ID 1) - Bus free phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.850] [trace] virtual bool GPIOBUS::PollSelectEvent()
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.850] [trace] static bool SBC_Version::IsBananaPi()
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.861] [trace] static bool SBC_Version::IsBananaPi()
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.861] [trace] (ID 1) - ++++ Starting processing for initiator ID 6
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.861] [trace] (ID 1) - Selection phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.861] [trace] (ID 1) - Message Out phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.861] [trace] (ID 1) - Receiving data, transfer length: 1 byte(s)
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.861] [trace] (ID 1) - Phase: msgout
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.861] [trace] (ID 1) - Received IDENTIFY message for LUN 0
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.861] [trace] (ID 1) - Command phase
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.862] [debug] (ID 1) - Controller is executing Read10, CDB $2800000f4d5f00000100
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.862] [debug] (ID:LUN 1:0) - Device is executing Read10 ($28)
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.862] [trace] (ID:LUN 1:0) - READ/WRITE/VERIFY/SEEK, start block: $000f4d5f, blocks: 1
Mar 13 09:13:16 piscsi PISCSI[360]: [2023-03-13 09:13:10.862] [trace] (ID:LUN 1:0) - Device is ready
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:10.871] [trace] (ID:LUN 1:0) - Length is 512
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:10.872] [trace] (ID 1) - Entering Data In phase
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:10.872] [trace] (ID 1) - Sending data, offset: 0, length: 512
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:13.872] [debug] (ID 1) - Error status: Sense Key $b, ASC $0
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:13.872] [trace] (ID 1) - Error (to status phase)
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:13.872] [trace] (ID 1) - Status Phase, status is $02
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:13.872] [trace] (ID 1) - Sending data, offset: 0, length: 1
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:16.872] [trace] (ID 1) - Bus free phase
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:16.873] [trace] virtual bool GPIOBUS::PollSelectEvent()
Mar 13 09:13:17 piscsi PISCSI[360]: [2023-03-13 09:13:16.873] [trace] static bool SBC_Version::IsBananaPi()
Mar 13 09:13:20 piscsi PISCSI[360]: [2023-03-13 09:13:20.571] [trace] Received SERVER_INFO command

**Immediate Reboot upon attempted access:**
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.603] [trace] (ID 1) - ++++ Starting processing for initiator ID 6
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.603] [trace] (ID 1) - Selection phase
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.603] [trace] (ID 1) - Message Out phase
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.603] [trace] (ID 1) - Receiving data, transfer length: 1 byte(s)
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.604] [trace] (ID 1) - Phase: msgout
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.604] [trace] (ID 1) - Received IDENTIFY message for LUN 0
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.604] [trace] (ID 1) - Command phase
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.604] [debug] (ID 1) - Controller is executing Read10, CDB $28000000000000000100
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.604] [debug] (ID:LUN 1:0) - Device is executing Read10 ($28)
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.604] [trace] (ID:LUN 1:0) - READ/WRITE/VERIFY/SEEK, start block: $00000000, blocks: 1
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.604] [trace] (ID:LUN 1:0) - Device is ready
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.604] [trace] (ID:LUN 1:0) - Length is 512
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.605] [trace] (ID 1) - Entering Data In phase
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:24.605] [trace] (ID 1) - Sending data, offset: 0, length: 512
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:27.605] [debug] (ID 1) - Error status: Sense Key $b, ASC $0
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:27.605] [trace] (ID 1) - Error (to status phase)
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:27.605] [trace] (ID 1) - Status Phase, status is $02
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:27.606] [trace] (ID 1) - Sending data, offset: 0, length: 1
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:30.606] [trace] (ID 1) - Bus free phase
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:30.606] [trace] virtual bool GPIOBUS::PollSelectEvent()
Mar 13 09:25:30 piscsi PISCSI[360]: [2023-03-13 09:25:30.606] [trace] static bool SBC_Version::IsBananaPi()

Let me know if you want any other tests, log or have any other hardware questions!

uweseimet, 2023-10-16:
@dialtr Correct me if I am wrong, but I think you mentioned that you have a Kurzweil. Maybe you can help with this ticket, e.g. verify it with your hardware?
The logs indicate that there is a profound issue with the SCSI bus phases/the data transfer. I would not be surprised if this was a termination issue.

@Cubeinthebox Can you please verify whether all your devices are properly terminated, and that the cables and connectors you are using are OK?

dialtr, 2023-10-16:
It make take me more than a day to verify this as my K2000R is in an inconvenient place. It also has a SCSI2SD installed and so I need to change settings on that to free up a virtual device for the PiScsi. I also may not have the same operating system installed on the system, but I will do my best to check.

Cubeinthebox, 2023-10-16:
> @Cubeinthebox Can you please verify whether all your devices are properly terminated, and that the cables and connectors you are using are OK?

I'm currently unable to test with my piscsi due to the [Issue](https://github.com/PiSCSI/piscsi/issues/1244#issue-1941124855)  surrounding the web editor. I had shelved the piscsi until very recently and reflashed. I can confirm though that I can successfully utilize the scsi port on the back via a Zuluscsi Mini RP2040. I attempted to utilize the piscsi in the same manner rather than internally mount.

uweseimet, 2023-10-17:
@Cubeinthebox Issue https://github.com/PiSCSI/piscsi/issues/1244#issue-1941124855 should be resolved in the develop branch. Would be great if you could use this branch for further testing.
Please note that even if you can use your hardware with a Zuluscsi does not mean that everything is properly terminated when PiSCSI is connected. Can you please tell us which of your devices (Samplerl, PiSCSI) you terminated and how long the cable cable between PiSCSI and the sampler is?

Cubeinthebox, 2023-10-17:
@uweseimet 
> Can you please tell us which of your devices (Samplerl, PiSCSI) you terminated and how long the cable cable between PiSCSI and the sampler is?
Both the sampler and PiSCSI were terminated and no other SCSI devices were in the chain, connected directly to the back of the unit.

> Would be great if you could use this branch for further testing.
I'm not sure how to do this, are there any instructions on how to compile with the develop branch as opposed to the main release?

uweseimet, 2023-10-17:
Just check out the respective branch. Everything else works like with a release branch.

Cubeinthebox, 2023-10-17:
I utilized the precompiled image to flash my SD card via the most recent release. I'm not familiar with how to specifically utilize the develop branch when flashing piscsi onto a fresh Pi install, and I couldn't find anything in the documentation to aid in that. I see where you clone the git but nothing beyond that. I was able to initially set up my PiSCSI via the documentation on the wiki but the precompiled image makes it even easier and it's not clear to me where I would invoke the develop branch during the setup steps. This is all a bit foreign to me, wish I could help more!

uweseimet, 2023-10-17:
Compiling the sources, even if it is for the very first time, is easier than you might think. There are detailed instructions on https://github.com/PiSCSI/piscsi/wiki/Setup-Instructions, see "Clone & build the PiSCSI software", steps 1 to 6.
Depending on your Pi the compilation process may take quite some time, but it does not need any manual intervention.

Cubeinthebox, 2023-10-17:
I understand and was able to compile before, I just don't understand how to utilize the develop branch of the code.

> Note that for the very special problem addressed in this ticket without somebody who can reproduce it and compiles and tests potential fixes from the sources there will not be any progress in the first place.

I appreciate this and had a working unit when I posted this issue in March 2023, but I just don't understand how to compile/flash my device utilizing the develop branch. I'll just close the ticket as it seems I'm not capable in utilizing PiSCSI for the purposes I need.

uweseimet, 2023-10-17:
@Cubeinthebox There is nothing to be flashed. For the develop branch, just like for a release branch, the resulting piscsi binary resides in bin/fullspec. So if you were able to compile before, compiling any (develop) branch exactly works the same.

Cubeinthebox, 2023-10-17:
I should mention that I did attempt to compile via steps 1-6, followed the instructions and still have the werkzeug error after waiting today for the compiling to occur on the device. I still don't understand how to utilize the develop branch vs. the main branch when compiling no matter if it "works the same". I'm keeping this closed as I'm fine with just utilizing Zuluscsi and keeping the PiSCSI in a drawer while I wait for a more stable release to test on and a bit more direction on how to compile the develop branch vs the release branch if it comes to actually testing any fixes (if needed) to this issue.

dialtr, 2023-10-18:
Gentlemen, I pulled out my K2000R tonight (It is not an RS from the factory, but has the sample option installed_ and it so happens that my unit is on the same operating system version as @Cubeinthebox, v3.87j. Adding to the similarities, I have a SCSI2SD installed (not a Zuluscsi.)  I attached a virtual hard disk to an unused SCSI ID (1). The K2000R identified the PiSCSI hosted virtual disk, but attempting to load anything caused the device to reboot.  The PiSCSI dashboard reports version "PiSCSI software version: 23.4.1" and my PiSCSI board is v2.7. I also saw different behavior when trying to mount an ISO: the device hangs.

I'd say we have a good test platform on our hands. I am going to buy a different SD card so that I can image it with a dev build. I would prefer not to touch the working/stable image since I use that with my Akai sample successfully.

uweseimet, 2023-10-18:
@dialtr Very interesting! This means that there is a chance now to have a closer look at the problems piscsi has with this hardware. I am going to reopen this ticket, assuming that @Cubeinthebox won't mind.

dialtr, 2023-10-18:
@uweseimet Yes, I feel like I must be experiencing the same issue! I will work towards setting up a dev image. Note that I will be taking some vacation time over the next week, so there may not be much movement on this ticket.

uweseimet, 2023-10-18:
@dialtr Any chance for movement on my pending PRs? There are other tickets already ready for more PRs, but I am blocked by the current PRs still waiting for review.