PiSCSI/piscsi #1132: No module named 'bjoern'

finalmillenium, 2023-03-20:
- Which version of Pi are you using: PI4b-4g
- Which github revision of software: image_2023-02-11-RaSCSI-v23.02.01-lite
- Which board version: 2.4a
- Which computer is the PiSCSI connected to: Quadra 800

Due to the major changes with the project since my last update, I decided to backup my configs and data, and perform a full re-image.
Upon starting up, I configured wifi, localizations, and changed default password.
When I attempt to load the web interface, I got the landing page for a long time.

This is the output of `journalctl -f` while attempting to load the config page on a fresh image.
```
Mar 20 14:23:12 piscsi systemd[1]: Started PiSCSI-Web service.
Mar 20 14:23:21 piscsi PISCSIWEB[1445]: Warning: Not running from a valid git repository. Will not be able to update the code.
Mar 20 14:23:23 piscsi PISCSIWEB[1451]: compiling catalog src/translations/sv/LC_MESSAGES/messages.po to src/translations/sv/LC_MESSAGES/messages.mo
Mar 20 14:23:23 piscsi PISCSIWEB[1451]: compiling catalog src/translations/zh/LC_MESSAGES/messages.po to src/translations/zh/LC_MESSAGES/messages.mo
Mar 20 14:23:23 piscsi PISCSIWEB[1451]: compiling catalog src/translations/fr/LC_MESSAGES/messages.po to src/translations/fr/LC_MESSAGES/messages.mo
Mar 20 14:23:23 piscsi PISCSIWEB[1451]: compiling catalog src/translations/de/LC_MESSAGES/messages.po to src/translations/de/LC_MESSAGES/messages.mo
Mar 20 14:23:23 piscsi PISCSIWEB[1451]: compiling catalog src/translations/es/LC_MESSAGES/messages.po to src/translations/es/LC_MESSAGES/messages.mo
Mar 20 14:23:23 piscsi PISCSIWEB[1445]: Starting PiSCSI Web UI ...
Mar 20 14:23:24 piscsi PISCSIWEB[1453]: Traceback (most recent call last):
Mar 20 14:23:24 piscsi PISCSIWEB[1453]:   File "/home/pi/piscsi/python/web/src/web.py", line 12, in <module>
Mar 20 14:23:24 piscsi PISCSIWEB[1453]:     import bjoern
Mar 20 14:23:24 piscsi PISCSIWEB[1453]: ModuleNotFoundError: No module named 'bjoern'
Mar 20 14:23:24 piscsi systemd[1]: piscsi-web.service: Main process exited, code=exited, status=1/FAILURE
Mar 20 14:23:24 piscsi systemd[1]: piscsi-web.service: Failed with result 'exit-code'.
Mar 20 14:23:24 piscsi systemd[1]: piscsi-web.service: Consumed 11.583s CPU time.
```

I am able to impport the python bjoern module in a quick idle session without any errors, and pip shows it properly installed in system level with correct permissions, so I'm scratching my head as to why isn't working. Figured after trying the normal troubleshooting/I'm an idiot stuff before I'd write up an issue.

rdmark, 2023-03-20:
This issue is usually caused by missing or incomplete python venv installations. You may have had unstable network connection while installing piscsi, or when launching piscsi-web for the first time.

I wrote a section on the wiki for this issue: https://github.com/PiSCSI/piscsi/wiki/Troubleshooting#user-content-Web_Interface

finalmillenium, 2023-03-20:
That did indeed fix it.
Though the fact that this is happening with the recommended image may still need to be looked at?

rdmark, 2023-03-20:
I guess one thing we could try is to pre-cache python libraries in the image to bypass the common fail state.

@akuker Is this something that's possible in the image generation script?

finalmillenium, 2023-03-28:
I'm going to close this since it's been resolved.

closed 2023-03-28 by finalmillenium

sqwirral, 2023-11-07:
It's not resolved because the same thing just happened to me.

rdmark, 2023-11-08:
@sqwirral It is resolved in the bleeding-edge development code. We're fixing a few last bugs now in preparation for a stable release. Please bear with us a little longer!