PiSCSI/piscsi #1133: Classic Mac (1 Minute) Hang on Shutdown

mitchkramez, 2023-03-21:
# Info

- Which version of Pi are you using: 3b+
- Which github revision of software: February Release
- Which board version: 2.5? I think (purchased in May of 2022)
- Which computer is the PiSCSI connected to: Macintosh LC3 (Though this happens on my Performa 475 and a few others)

# Describe the issue
- On shutdown, my macs hang for about a minute before showing the "It is now safe to shutdown your Macintosh screen"
- This only started after updating to the December release, prior to that I hadn't seen this behavior with any of my images.
- I upgraded in place (git) to the December update, and the images didn't change at all during this time
- I have since done a full re-image of the SD card to see if that was the issue
- If I create a blank HDA in the WebUI: https://screenshot.electricpulp.com/PKjNqz73p0bQhr6qKCcW

# Disk Image Created in the WebUI:
- After attaching this image, Mac OS 7.5.3 doesn't see it and ask me to initialize it by default.
- Initializing the disk with Patched Apple HD Setup on the RaSCSI bootstrap image takes several minutes to scan the SCSI bus with this image connected, but it does eventually find the Disk and let me initialize it.
- At this point, the system will hang on shutdown - this is the trace-level log output:
[message.txt](https://github.com/PiSCSI/piscsi/files/11029311/message.txt)
- The Relevant output appears to be:
```
Mar 19 02:42:31 piscsi PISCSI[763]: [2023-03-19 02:42:28.541] [warning] (ID:LUN 5:0) - Unknown MODE SELECT page code: $08
Mar 19 02:42:31 piscsi PISCSI[763]: [2023-03-19 02:42:28.541] [warning] (ID:LUN 5:0) - Unknown MODE SELECT page code: $2b
Mar 19 02:42:31 piscsi PISCSI[763]: [2023-03-19 02:42:28.541] [debug] (ID 5) - Error status: Sense Key $5, ASC $26
Mar 19 02:42:31 piscsi PISCSI[763]: [2023-03-19 02:42:28.541] [trace] (ID 5) - Error (to status phase)
Mar 19 02:42:31 piscsi PISCSI[763]: [2023-03-19 02:42:28.541] [trace] (ID 5) - Status Phase, status is $02
Mar 19 02:42:31 piscsi PISCSI[763]: [2023-03-19 02:42:28.541] [trace] (ID 5) - Bus free phase
```

# Disk Image Created with Disk Jockey for Windows
- Screenshot of image config: https://screenshot.electricpulp.com/nGJq5V71FpRb8Hv6tBDK
- Attaching this disk and booting the same mac, System 7.5.3 immediately sees an uninitialized disk and asks me to initialize it. This happens quickly, probably 1/10 of the time it took Patched Apple HD Setup to find and partition the WebUI image.
- There are no hangs at shutdown using this image once it's been initialized and mounted (or even before)

# Disk Image Created in the WebUI, attempting to use another disk driver (LIDO/SCSI Director)
- SCSI Director would not work initialize the disk. Here's a screenshot of the error on initialization: https://screenshot.electricpulp.com/2lhmBdKWhzs32H4MTNwq, and the error on format: https://screenshot.electricpulp.com/m3v7lyyy5qCVMZNcBGzw
- Lido was able to initialize and format the disk, and there was no delay at shutdown.

# Final Notes
- None of this lag happened before the December update, but it does continue in  the February release.
- The base disk images that come with the PiSCSI release image seem to work just fine (I only tested the 7.5.3 and 7.0 and Bootstrap Images)
- The images that I have been using since the beginning were all created with`dd` on a mac, so I'm guessing there's some correlation there with what the WebUI is doing
- It certainly seems to be somewhat driver related since LIDO worked just fine, but it's still odd that these images were fine with the pre-December release. I'll try and go back and re-flash the old release here and confirm that all is working as it should and follow up here with that result soon.

Thanks for the great project, and let me know if there's any additional information or testing you'd like me to do!

mitchkramez, 2023-03-29:
I’ve been able to spend a bit more time on this this evening. This only appears to happen on Mac OS 7.5.3 and earlier. My machines with Mac OS 8.6 didn’t have any issues with the drive images that cause hanging on the older OS.

Additionally, it appears that this is some kind of driver issue, since LIDO prepped images work just fine in both OS’s.

This still strikes me as weird that the Apple HD SC Setup would not behave properly with the images on PiSCSI, but it behaves fine with the images when they’re in a BlueSCSI.

I still plan to roll back to the pre December release and confirm there are no issues there as well, but this week hasn’t allowed much time.

rdmark, 2023-04-10:
@akuker The freeze seems to happen with `bool SBC_Version::IsBananaPi()` -- can you think of any scenario where this check can get stuck in a loop?

akuker, 2023-10-14:
Note: removing myself as the assignee

mitchkramez, 2023-10-15:
I think we can close this issue out. I'm no longer having any hangs in shutdown with the latest version 😀

rdmark, 2023-10-15:
That is great to hear! Must have been something environmental at one point...

closed 2023-10-15 by rdmark