PiSCSI/piscsi #1151: Web UI start.sh: Recover automatically from broken venv

rdmark, 2023-04-25:
We're yet again running into frequent messed up venv that start.sh isn't able to recover from. The typical error state is:

```
Apr 25 05:47:45 piscsi PISCSIWEB[1495]: Starting PiSCSI Web UI ...
Apr 25 05:47:46 piscsi PISCSIWEB[1579]: Traceback (most recent call last):
Apr 25 05:47:46 piscsi PISCSIWEB[1579]:   File "/home/pi/piscsi/python/web/src/web.py", line 12, in <module>
Apr 25 05:47:46 piscsi PISCSIWEB[1579]:     import bjoern
Apr 25 05:47:46 piscsi PISCSIWEB[1579]: ModuleNotFoundError: No module named 'bjoern'
```

sadguitarius, 2023-04-26:
I get this error on my setup as well. I'm not using the pre-built image, but instead building PiSCSI myself on Manjaro ARM. In general, everything works fine as long as I manually install all dependencies, but the one thing that doesn't work is the web interface. I've been "fixing" it by just creating a Conda environment with the proper dependencies and modifying the systemd file to use that environment. 

Are you using the pre-built image, easyinstall or doing a manual installation? It would be good to know if _anyone_ is successfully using the web interface when following the manual install instructions. If so, what's the difference that's causing this?

rdmark, 2023-04-27:
That's odd. Let me check if the manual install instructions on the wiki are outdated somehow.

Sorry for the lack of context in this issue ticket, but the specific scenario I'm alluding to here is when you use a pre-built release image, and flash a wifi-only RPi (Zero) which leaves a broken venv due to initial instability of the wifi network connection. At least this is the theory.

rdmark, 2023-04-27:
I made some updates to the wiki now https://github.com/PiSCSI/piscsi/wiki/Web-Interface adding a few more dependent packages, plus instructions to get HFS volume formatting working.

If your issue persists when using venv, please share the full logs from when you attempt to launch the webapp.

rdmark, 2023-05-01:
Regarding the particular fail state I have observed, it seems we end up with an incomplete, but otherwise functional venv. Problem is that only baseline libraries are actually installed, so our python code can't start obviously:

```
$ ls -al piscsi/python/web/venv-bad/lib/python3.9/site-packages/
total 40
drwxr-xr-x 9 root root 4096 May  1 02:14 .
drwxr-xr-x 3 root root 4096 May  1 02:14 ..
-rw-r--r-- 1 root root  126 May  1 02:14 easy_install.py
drwxr-xr-x 5 root root 4096 May  1 02:14 pip
drwxr-xr-x 2 root root 4096 May  1 02:14 pip-20.3.4.dist-info
drwxr-xr-x 5 root root 4096 May  1 02:14 pkg_resources
drwxr-xr-x 2 root root 4096 May  1 02:14 pkg_resources-0.0.0.dist-info
drwxr-xr-x 2 root root 4096 May  1 02:14 __pycache__
drwxr-xr-x 6 root root 4096 May  1 02:14 setuptools
drwxr-xr-x 2 root root 4096 May  1 02:14 setuptools-44.1.1.dist-info
```

rdmark, 2023-05-01:
One solution for this scenario is to do `pip3 list` and validate the output. Will be cumbersome to do with a shell script though...

Another more heavy-handed solution would be to always run `pip3 install -r requriements.txt` before launching web.py. It would add a minute to the startup time of the web UI though. We could introduce a parameter for bypassing this.

@akuker Do you have a preferred solution here?

ericlewis, 2023-07-30:
I am seeing the same issue on latest release (April 22) when using the official rpi sd flashing tool on rpi4. WiFi only connection currently. I tried to run easy install again but same issue.

rdmark, 2023-07-30:
@ericlewis To recover from this error you must delete the python/web/venv dir and restart the piscsi-web service.

A workaround that I learned yesterday is to add a wpa_supplicant.conf file to the file system before starting the RPi for the first time after flashing. See f.e. https://www.raspberrypi-spy.co.uk/2017/04/manually-setting-up-pi-wifi-using-wpa_supplicant-conf/

This makes sure that your RPi has a working network connection before pip tries to create the venv.

ericlewis, 2023-07-30:
Thanks. That definitely helped! Now just weirder issues.

rdmark, 2023-07-30:
@ericlewis Do you need help with the other issues? A good place for initial troubleshooting is our Discord server https://discord.gg/CNmfPmddDR

rdmark, 2023-07-31:
Here's an attempt to work around the issue happening in the first place: https://github.com/PiSCSI/piscsi/pull/1197

EriolGaurhoth, 2023-10-10:
> 
> A workaround that I learned yesterday is to add a wpa_supplicant.conf file to the file system before starting the RPi for the first time after flashing. See f.e. https://www.raspberrypi-spy.co.uk/2017/04/manually-setting-up-pi-wifi-using-wpa_supplicant-conf/
>

I've tried this workaround and it still gets me to the same place as everyone else, with the system not recognizing 'bjoern'.  It's because, with wpa_supplicant.conf included on first boot, I get the message "Wi-Fi is currently blocked by rfkill". I have to change the localization for it to work, and by then, it's too late. Just curiously, if I set the country to GB (which I believe is default) will my US wifi still recognize, even temporarily, so that the python libraries are set up correctly?

rdmark, 2023-10-22:
It's never "too late" -- you can always go and delete the `venv` dir and restart the service once network connection is established.

EriolGaurhoth, 2023-10-25:
> It's never "too late" -- you can always go and delete the `venv` dir and restart the service once network connection is established.

What I meant was too late to NOT have to delete venv. I did eventually get it working using the suggestions in this thread, though it certainly wasn't an "easyinstall" as the original script implied.

rdmark, 2023-11-04:
superbenk @ discord reported that upgrading to the current release candidate code in `develop` led to a broken venv for piscsi-web, so I propose that we do the brute force approach of having easyinstall.sh simply look for previous venv and deleting them outright, as part of the upgrade workflow.

PR here https://github.com/PiSCSI/piscsi/pull/1300

closed 2023-11-04 by rdmark