PiSCSI/piscsi #1154: OLED Script / requirements.txt - Contains error

GoodSpeed001, 2023-04-29:
# Info

- Which version of Pi are you using: 3B+
- Which github revision of software: April 2023 Release - v23.04.01
- Which board version: FULL
- Which computer is the PiSCSI connected to: NA

# Describe the issue
Hi found the an error on the script / requirements.txt

https://github.com/PiSCSI/piscsi/blob/main/python/oled/requirements.txt

pi@rascsi:~/piscsi/python/oled $ source venv/bin/activate
(venv) pi@rascsi:~/piscsi/python/oled $ sudo pip3 install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement pkg-resources==0.0.0
ERROR: No matching distribution found for pkg-resources==0.0.0

After removing the line 1 pkg-resources==0.0.0, the script can run.

GoodSpeed001, 2023-04-29:
Older version of Requirements.txt have the following:

```
Adafruit-Blinka==6.15.0
adafruit-circuitpython-busdevice==5.1.0
adafruit-circuitpython-framebuf==1.4.7
adafruit-circuitpython-ssd1306==2.12.2
Adafruit-PlatformDetect==3.17.2
Adafruit-PureIO==1.1.9
Pillow==9.0.1
pkg-resources==0.0.0
pyftdi==0.53.3
pyserial==3.5
pyusb==1.2.1
rpi-ws281x==4.3.0
RPi.GPIO==0.7.0
sysv-ipc==1.1.0
protobuf==3.19.1
unidecode==1.3.2
```

But I'm not too comfortable or have the know-how to apply it on this last version without breaking something :(

GoodSpeed001, 2023-04-29:
after checking this issue with @rdmark it might be cause by a mess python venv.

going to do a fresh installation to restart my tests.

additionally, @rdmark have updated the OLED wiki to contemplate renaming files to be executed manually

I'll provide feedback as soon as possible

rdmark, 2023-04-29:
Weird, `pkg-resources==0.0.0` has been in requirements.txt for years literally and hasn't been troublesome in the past. But a quick googling suggests that it is the result of a bug in Debian's pip. So let's get rid of it https://github.com/PiSCSI/piscsi/pull/1155

GoodSpeed001, 2023-04-29:
Installation of a clean build, have finished (main and oled option).

```
Created symlink /etc/systemd/system/multi-user.target.wants/piscsi-oled.service → /etc/systemd/system/piscsi-oled.service.

The piscsi-oled service will start on the next Pi boot.
Press Enter to reboot or CTRL-C to exit

Rebooting...
```

Had no luck, the OLED doesn't start automatically (expected to have the systemclt automatically).
Should I run the guide - https://github.com/PiSCSI/piscsi/wiki/OLED-Status-Display-(Optional)#create-a-systemd-service-to-automatically-run ?

The OLED only start after I ran the start.sh script

pi@piscsi:~/piscsi/python/oled $ ./start.sh
Starting OLED Screen...
No screen rotation parameter given; falling back to the default.
No screen height parameter given; falling back to the default.
Running with the following display:
<adafruit_ssd1306.SSD1306_I2C object at 0x75c6d5b0>

Will update the OLED display every 1000ms (approximately)

Still strange message, because I have setup the rotation parameter (select the default) and have selected the type of OLED to 128x32 (select the default).

GoodSpeed001, 2023-04-30:
I decided to follow the guide https://github.com/PiSCSI/piscsi/wiki/OLED-Status-Display-(Optional)#create-a-systemd-service-to-automatically-run

when copy piscsi-oled.service by doing

`sudo cp ~/piscsi/python/oled/service-infra/piscsi-oled.service /etc/systemd/system/`

and then run

`sudo systemctl enable piscsi-oled.service`

and

`sudo systemctl piscsi-oled.service`

get error cannt find piscsi-oled.service

```
pi@piscsi:~/piscsi/python/oled $ sudo systemctl enable piscsi-oled.service
pi@piscsi:~/piscsi/python/oled $ sudo systemctl piscsi-oled.service
Unknown command verb piscsi-oled.service.
```
in piscsi the file exists:

```
pi@piscsi:~/piscsi/python/oled/service-infra $ ls
piscsi-oled.service
```
and in /etc/systemd/system/

```
pi@piscsi:~/piscsi/python/oled/service-infra $ cd /etc/systemd/system/
pi@piscsi:/etc/systemd/system $ ls -l
total 84
drwxr-xr-x 2 root root 4096 Apr 25 03:36 bluetooth.target.wants
lrwxrwxrwx 1 root root   42 Apr 25 03:38 dbus-fi.w1.wpa_supplicant1.service -> /lib/systemd/system/wpa_supplicant.service
lrwxrwxrwx 1 root root   37 Apr 25 03:36 dbus-org.bluez.service -> /lib/systemd/system/bluetooth.service
lrwxrwxrwx 1 root root   40 Apr 25 03:36 dbus-org.freedesktop.Avahi.service -> /lib/systemd/system/avahi-daemon.service
lrwxrwxrwx 1 root root   40 Apr 25 03:38 dbus-org.freedesktop.ModemManager1.service -> /lib/systemd/system/ModemManager.service
lrwxrwxrwx 1 root root   45 Apr 25 03:32 dbus-org.freedesktop.timesync1.service -> /lib/systemd/system/systemd-timesyncd.service
drwxr-xr-x 2 root root 4096 Apr 25 03:28 default.target.wants
drwxr-xr-x 2 root root 4096 Apr 25 03:36 dev-serial1.device.wants
drwxr-xr-x 2 root root 4096 Apr 25 03:38 dhcpcd.service.d
drwxr-xr-x 2 root root 4096 Apr 25 03:28 getty.target.wants
drwxr-xr-x 2 root root 4096 Apr 25 03:32 getty@tty1.service.d
drwxr-xr-x 2 root root 4096 Apr 25 03:36 graphical.target.wants
drwxr-xr-x 2 root root 4096 Apr 25 03:35 halt.target.wants
drwxr-xr-x 2 root root 4096 Apr 30 00:45 multi-user.target.wants
drwxr-xr-x 2 root root 4096 Apr 25 03:38 network-online.target.wants
-rw-r--r-- 1 root root  443 Apr 30 01:01 **piscsi-oled.service**
-rw-r--r-- 1 root root  662 Apr 30 00:40 piscsi.service
-rw-r--r-- 1 root root  662 Apr 25 03:51 piscsi.service.old
-rw-r--r-- 1 root root  208 Apr 30 00:40 piscsi-web.service
drwxr-xr-x 2 root root 4096 Apr 25 03:35 poweroff.target.wants
drwxr-xr-x 2 root root 4096 Apr 25 03:37 rc-local.service.d
drwxr-xr-x 2 root root 4096 Apr 25 03:35 reboot.target.wants
drwxr-xr-x 2 root root 4096 Apr 25 03:36 remote-fs.target.wants
drwxr-xr-x 2 root root 4096 Apr 25 03:37 sockets.target.wants
lrwxrwxrwx 1 root root   31 Apr 25 03:36 sshd.service -> /lib/systemd/system/ssh.service
drwxr-xr-x 2 root root 4096 Apr 25 03:36 sysinit.target.wants
lrwxrwxrwx 1 root root   35 Apr 25 03:29 syslog.service -> /lib/systemd/system/rsyslog.service
drwxr-xr-x 2 root root 4096 Apr 25 03:30 timers.target.wants
```
I just don't understand the error 

```
pi@piscsi:/etc/systemd/system $ sudo systemctl piscsi-oled.service
Unknown command verb piscsi-oled.service.
```

rdmark, 2023-04-30:
The last issue was an error in the wiki docs which has now been fixed. You need to issue the `start` verb to start the service.

GoodSpeed001, 2023-04-30:
hello @rdmark thank you for the help and patience :) we can close the issue.

rdmark, 2023-04-30:
You're welcome!

closed 2023-04-30 by rdmark