PiSCSI/piscsi #1158: scsimon fails with error 'Unable to register event request. Is PiSCSI or RaSCSI already running? ; Unable to intiailize the GPIO bus. Exiting....

rreiner, 2023-04-30:
# Info

Which version of Pi are you using: Zero W
Which github revision of software: April 2023 Release - v23.04.01
Which board version: FULL
Which computer is the PiSCSI connected to: NA

# Describe the issue

`scsimon` produces an error message and exits.

```
pi@scrape:~/images$ sudo scsimon
[info] Reading live data from the GPIO pins
[info]     Connection type : FULLSPEC
[info]     Data buffer size: 1000000
[info]
[info] Generating output files:
[info]    log.vcd - Value Change Dump file that can be opened with GTKWave
[info]    log.json - JSON file with raw data
[info]    log.html - HTML file with summary of commands
[info]
[info] Now collecting data.... Press CTRL-C to stop.
[info]
[info] Detected device Raspberry Pi 1
[error] Unable to register event request. Is PiSCSI or RaSCSI already running?
[error] Unable to intiailize the GPIO bus. Exiting....
```

rdmark, 2023-04-30:
You can't run piscsi and scsimon at the same time. Shut down the former to run the latter. Should work then!

rreiner, 2023-04-30:
Ah, OK. I don't see any mention of that in the docs, so this is a documentation bug then :-)
Thanks1

rdmark, 2023-04-30:
Which documentation did you look at? The wiki is pretty clear about it I think. https://github.com/PiSCSI/piscsi/wiki/SCSI-Bus-Capture#capture-data

What may be missing is a line in the man page.

In addition, the error message does provide a hint, but perhaps it is ambiguous?
```
[error] Unable to register event request. Is PiSCSI or RaSCSI already running?
```

rreiner, 2023-04-30:
For cli tools I tend to rely on man pages or the output of `-h` / `--help`, so a mention there would be helpful.

Re. that error message, I interpreted it to mean "this tool needs PiSCSI to be running, are you sure it is?" rather than the opposite, so I think yes, it could be clearer.

akuker, 2023-04-30:
feel free to update the man page and submit a PR. Always glad to accept suggestions :)
https://github.com/PiSCSI/piscsi/blob/main/doc/scsimon.1

rdmark, 2023-05-21:
Here's a PR that hopefully will improve the docs: https://github.com/PiSCSI/piscsi/pull/1175

closed 2023-06-24 by rdmark