PiSCSI/piscsi #1159: FTP transfer get "Permission denied error"

antoniograzioli, 2023-05-01:
- Which version of Pi are you using: Raspberry Pi 3 Model B Plus Rev 1.3, Linux 6.1.21-v8+ aarch64  
- Which github revision of software: April 2023

I am trying to upload some files to the images using Filezilla and FTp.
Connection is OK, but any time I try to upload a file I get a "503 Permission Denied" on the /images folder
I have also tried "chmod -R 777 images" with no luck.
Any idea on how to overcome thsi (Webui upload is quite unreliable, most of the time the upload fails, ot the file is corrupted)

antoniograzioli, 2023-05-01:
Filezilla says:

```
Comando: 	CWD /home/pi/images
Risposta:  	250 Directory successfully changed.
Comando: 	PWD
Risposta:  	257 "/home/pi/images" is the current directory
Comando: 	TYPE I
Risposta:  	200 Switching to Binary mode.
Comando: 	PASV
Risposta:  	227 Entering Passive Mode (192,168,1,2,190,165).
Comando: 	STOR abcd.pdf
Risposta:  	550 Permission denied.
```

rdmark, 2023-05-01:
FTP is an insecure protocol. Please use ssh (filezilla calls it SFTP) instead.

As a side note: This issue is not related to the piscsi software at all. Please use the Discussions forum here on github, or join our Discord server for general troubleshooting.