PiSCSI/piscsi #1161: Web UI: before_first_request is deprecated in Flask 2.3.x

rdmark, 2023-05-02:
Finding an alternative solution for the `before_first_request` decorator that was removed in Flask 2.3.0

https://flask.palletsprojects.com/en/2.3.x/changes/#version-2-3-0

> The app.before_first_request and bp.before_app_first_request decorators are removed.

Where the decorator was removed: https://github.com/pallets/flask/issues/4605

rdmark, 2023-05-16:
Fixed with https://github.com/PiSCSI/piscsi/pull/1162

closed 2023-05-16 by rdmark

ZulqarnainB, 2023-06-10:
what is an alternative of before_first_request in the latest version?

rdmark, 2023-06-10:
@ZulqarnainB please see the PR I linked above for my solution. I used `before_request` with slightly different logic.

FredySchaible, 2023-10-02:
@ZulqarnainB, you can also use **before_app_first_request** instead of **before_first_request**