PiSCSI/piscsi #1165: Hardware version "akuker 2.7" runs transceiver out of spec with Vcc=3.3V

ManuTester, 2023-05-02:
Reviewing the schematic of the hardware version "akuker 2.7",
I found that the 74LS641 is now getting only 3.3V supply voltage.
According to the spec, the minimum voltage is 4.75V.
Has this design been fully tested?
I don't think this will work in all cases, the chip is running completely out of spec,
and there is - to my knowledge - no 74LVC641 or so to replace it.

The outputs are open-collector so not much power is drawn from the Vcc anyway.

Also 10uF caps have been added, one for each transceiver. One 10uF cap per voltage probably won't harm (general buffering of supply, put near pi connector), but what's desperately needed here are blocking/bypass caps, 100nF per chip, no more, no less. You put them as close as possible to the chips. 10uF caps do not work as bypass capacitors!

hemulen242, 2023-05-16:
Can this be confirmed as bug, or will these 74LS641 transceivers run on +3V3 ?

pascalsavigny, 2023-06-11:
Hello, interesting point. Can anyone confirm if it's ok or not ? Are we really out of spec ? 
Thanks in advance.
I'd like to place an order for the 2.7 revision....

akuker, 2023-06-20:
> Reviewing the schematic of the hardware version "akuker 2.7", I found that the 74LS641 is now getting only 3.3V supply voltage.
> According to the spec, the minimum voltage is 4.75V.
Good catch @ManuTester . We are running the transceivers outside of the datasheet limits. 

> Has this design been fully tested?
Yes, I've been testing with a few boards and haven't run into any issues. 

> I don't think this will work in all cases, the chip is running completely out of spec, and there is - to my knowledge - no 74LVC641 or so to replace it.

I did quite a bit of testing with my Macintosh SE/30, but it is possible that some chips may behave differently. Different lots of parts may have different characteristics. For the ones that I'm shipping on Tindie, I'm testing them out and haven't heard any complaints from the ones I've shipped out.  

> Also 10uF caps have been added, one for each transceiver. One 10uF cap per voltage probably won't harm (general buffering of supply, put near pi connector), but what's desperately needed here are [...]

Acknowledged. I'll make this update to the production files in the next few weeks to change the values. Proposed replacement:
https://jlcpcb.com/partdetail/Fojan-FCC0402N101J500AT/C5137573

akuker, 2023-06-20:
> Can this be confirmed as a bug, 

Bug? Technically, yes. 

> or will these 74LS641 transceivers run on +3V3 ?

I haven't had any issues running on 3.3v, but your mileage may vary. Since its being used outside of spec, it is possible that some chips may work, others may not.

akuker, 2023-06-20:
If anyone has any boards that you're having issues with, please comment here or reach out to me. I agree we are using the part outside of its specification which is not a good idea. I'd like to collect some data to see if folks are having issues. I've tested a hundred or so and haven't seen issues. BUT, it could be that how I'm testing them is tolerant of flakey signals.

uweseimet, 2023-06-20:
I'm not a hardware expert, but nevertheless I would like to comment. IMO running hardware out of spec is not a good idea. Users pay money for the board. If you buy a Pi and a board and then it turns out that the board does not work because some parts are run out of spec this definitely would not do the PiSCSI project any favor. Why taking such a risk?
Just my two cents.

ManuTester, 2023-06-20:
I think @akuker intended to reduce power consumption that's why he went to 3.3V, but:
The 74LS641 has open collector output, that means only very little power ist drawn from VCC, I don't think much power can be saved this way. It should be changed back. Sure, I believe it works and if you have such a board and it works, fine you got no problem. Underrunning the chip won't harm it, but the input threshold levels on the 5v input signals won't be correct and it may also draw more current from there.
I looked at the capacitor you chose as bypass cap, that one is 100p not 100n! Also why choose such a small one (0402) the 0805 are much easier to handle/solder by hand.

uweseimet, 2023-10-02:
What are going to be the next steps in order to resolve this ticket?

akuker, 2023-10-14:
Note: I have shut down my tindie store indefinitely. I don't think there is anything that needs to be done with this ticket, unless someone else is going to make hardware modifications. Therefore, I'm closing it.

closed 2023-10-14 by akuker