PiSCSI/piscsi #1167: Rename options in WebUI doesn't work with apostrophes in filenames

antoniograzioli, 2023-05-10:
I think I discovered a small bug.
if you have a filename containing apostropes, the "rename" button doesn't work and give an erroe ("A file with name /home/pi/images already exists")

if you rename the files using Filezilla as an example, then rename works correctly.
Here's a small video documenting the bug and how to reproduce it...

https://www.youtube.com/watch?v=8tLdNcLgKgo

rdmark, 2023-05-18:
I think this is because the apostrophe breaks the embedded javascript, namely the string delineation. We need to escape special chars in that string to make this more sturdy.

```
<form action="/files/rename" method="post" class="file-rename" onsubmit="var new_file_name = prompt('Enter new file name for: 1'0000.hds', '1'0000.hds'); if (new_file_name === null) event.preventDefault(); document.getElementById('new_file_name_images/_1').value = new_file_name;">
                    <input name="file_name" type="hidden" value="1'0000.hds">
                    <input name="new_file_name" id="new_file_name_images/_1" type="hidden" value="">
                    <input type="submit" value="Rename" title="Rename">
                </form>
```

rdmark, 2023-05-21:
Fixing with https://github.com/PiSCSI/piscsi/pull/1174

closed 2023-05-21 by rdmark