PiSCSI/piscsi #1178: Yamaha AW2816 - Issues when formatting device

procionefotone, 2023-05-23:
# Info

- Which version of Pi are you using: Zero W
- Which github revision of software: 23.4.1 (Pre-built image)
- Which board version: [akuker PiSCSI Zero 1.1](https://github.com/piscsi/piscsi-hw/tree/main/piscsi_zero_1p1)
- Which computer is the PiSCSI connected to: Yamaha AW2816 (Digital mixing console)

# Describe the issue

Hello,
I'm having an issue using the PiSCSI on a Yamaha AW2816.
The emaulated drive is successfully recognized by the console, but it needs formatting by the console before it's usable.
When excecuting the formatting, the communication seems fine for a few seconds (based on the indicator LEDs), then it halts without completing and the console is stuck on "formatting".

I've gathered some logs at the error phase, which are showing the following:
```
May 23 16:09:49 piscsi PISCSI[355]: [2023-05-23 16:09:40.605] [trace] (ID 0) - Command phase
May 23 16:09:49 piscsi PISCSI[355]: [72B blob data]
May 23 16:09:49 piscsi PISCSI[355]: [2023-05-23 16:09:40.606] [debug] (ID 0) - Error status: Sense Key $5, ASC $20
May 23 16:09:49 piscsi PISCSI[355]: [2023-05-23 16:09:40.606] [trace] (ID 0) - Error (to status phase)
```

I'm attaching the full log for reference.
[piscsi-aw2816-0523.txt](https://github.com/PiSCSI/piscsi/files/11545870/piscsi-aw2816-0523.txt)

Is it possible that this issue is related to #1133 ?

Let me know if you need more tests, I will be happy to help!

Thank you for the great work

Daniele

uweseimet, 2023-05-23:
According to the Sense Key and ASC there is an issue with the SCSI command sent by the device. The command is usually also logged, but where the relevant logging output should be your logfile says "[72B blob data]". This sounds as if there are binary data instead of normal logfile text data.
Regular log output for the respective piece of code (SCSI Command phase) looks like this:
```
[2023-05-23 19:13:28.444] [trace] (ID 1) - Command phase
[2023-05-23 19:13:28.444] [debug] (ID 1) - Controller is executing Read6/GetMessage10, CDB $0803fdde0400
[2023-05-23 19:13:28.444] [debug] (ID:LUN 1:0) - Device is executing Read6/GetMessage10 ($08)
[2023-05-23 19:13:28.444] [trace] (ID:LUN 1:0) - READ/WRITE/VERIFY/SEEK, start block: $0003fdde, blocks: 4
```
I cannot tell why, but your piscsi process does not appear to generate proper log data for some parts of the logfile. I have never seen something like that before, but I am afraid without the missing logfile data it is not possible to find out why the SCSI command was considered illegal.

uweseimet, 2023-05-23:
The piscsi logs are always in text format, i.e. there is no way/need to force them from binary into text. If there are binary data in your log this is because something not (only) related to SCSI formatting is not working as expected. As you can see from my logfile example a regular logfile entry for SCSI commands is plain text and lists the SCSI command block with the command bytes. This information is missing in your log, and as long as this is the case it is not possible to address any SCSI format issue. The binary data block further analysis.

Just a general note: SCSI commands (and also the SCSI format command) do not have to do anything with filesystems. Filesystems and SCSI are completely independent things. This is the same with IDE or ATA: They are also not aware of filesystems and essentially only know how to read and write data from/to a drive. The nature of these data depend on the OS, and only the OS knows about filesystems.

procionefotone, 2023-05-23:
Thank you for the in-depth explanation! I know a bit of C++, I'll try to look at the logfile problem

uweseimet, 2023-05-23:
The relevant logging code is in scsi_controller.cpp, the method is ScsiController::Execute, right at the beginning:
```
	stringstream s;
	s << "Controller is executing " << command_mapping.find(GetOpcode())->second.second << ", CDB $"
			<< setfill('0') << hex;
	for (int i = 0; i < BUS::GetCommandByteCount(static_cast<uint8_t>(GetOpcode())); i++) {
		s << setw(2) << GetCmd(i);
	}
	logger.Debug(s.str());
```
Something with these lines of code does not work as expected on your system.

uweseimet, 2023-10-16:
@procionefotone Any news on this? I'm afraid without your feedback keeping this ticket open does not make much sense.

uweseimet, 2023-11-05:
Closing this ticket because without feedback it cannot be worked on.

closed 2023-11-05 by uweseimet