PiSCSI/piscsi #1179: Prepare C++ code for Debian Bookworm, update C++ standard to C++-20

uweseimet, 2023-06-19:
Debian Bookworm was released recently, and it's very likely just a matter of weeks until there is an official Raspberry Pi OS based on Bookworm. This is a list of changes and code improvements that can be applied:

- Drop the support for gcc 10. The Bullseye default compiler is gcc 10, with Bookworm it's gcc 12. The available clang versions are even newer.
- With gcc 10 C++-20 can be used as C++ standard level, which provides several benefits
- Clean up the code and make it better readable/maintainable by using C++-20 features
- Fix general issues with C++-20
- Fix C++-20 issues reported by SonarCloud
- Make all enums values upper case and unambiguous

There is a test_bookworm branch that reflects these changes. The code in this branch so far passes the test suite on a Pi that was updated from Bullseye to Bookworm.

```
>lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
>gcc --version
gcc (Raspbian 12.2.0-14+rpi1) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

akuker, 2023-06-20:
Buster has been moved to "legacy" support, so seems like a reasonable change.

uweseimet, 2023-07-04:
The plan is to wait for the officlal Raspberry Pi OS based on bookworm before merging the changes. Testing can already be done on an upgraded bullseye distribution.

uweseimet, 2023-09-28:
I assume bookworm will be released together with the Pi 5, which is supposed to be released next month.

closed 2023-10-15 by uweseimet