PiSCSI/piscsi #1180: Use link time optimization with gcc >= 10 in order to reduce binary sizes?

uweseimet, 2023-06-28:
When compiling and linking with the -flto option the resulting binaries are about 20% smaller.
Stripped sizes on bullseye without -flto:
```
-rwxrwxr-x 1 us us 1211432 28. Jun 09:39 piscsi
-rwxrwxr-x 1 us us  321364 28. Jun 09:39 scsictl
-rwxrwxr-x 1 us us  320720 28. Jun 09:39 scsidump
-rwxrwxr-x 1 us us  357552 28. Jun 09:39 scsiloop
-rwxrwxr-x 1 us us  341216 28. Jun 09:39 scsimon
```
Stripped sizes on bullseye with -flto:
```
-rwxrwxr-x 1 us us 1069208 28. Jun 09:29 piscsi
-rwxrwxr-x 1 us us  277240 28. Jun 09:29 scsictl
-rwxrwxr-x 1 us us  301372 28. Jun 09:29 scsidump
-rwxrwxr-x 1 us us  317712 28. Jun 09:29 scsiloop
-rwxrwxr-x 1 us us  321888 28. Jun 09:29 scsimon
```
Especially on Pis with only 512 MiB reducing the memory footprint is a benefit. On the other hand, compiling/linking with -flto takes longer and on these Pis may not work because there might not be enough memory for the compiler or linker. This is why just making -flto the default setting for a release build (in addition to -O3) may not be a good idea.
Nevertheless, pre-built binaries should probably be built with this option, and maybe easysetup should offer an option to add this flag (via EXTRA_FLAGS, no Makefile change is needed) for those who are building on a Pi with sufficient memory. I wonder, though, whether it is a good idea to add yet another option to easyinstall ...

rdmark, 2023-07-04:
I don't have a strong opinion one way or the other. Running out of memory on a RPi Zero when using piscsi has never really been a problem for me, but at the same time saving memory doesn't hurt either.

Adding another command line option to the easyinstall script doesn't introduce much overhead to the code. In fact, I think @akuker relies on the script for building the images, so this would be the best option for the time being. We could call it something like `--optimize`

The appropriate way to enable the flags is with the `CXXFLAGS="-flto"` env variable, isn't it?

uweseimet, 2023-07-04:
You might run out of memory while *compiling/linking* with this option, not when running piscsi. This is why we should not add this option unconditionally for release builds.
Maybe it's better just to add a note on this option in the development notes on the Wiki?

rdmark, 2023-07-05:
As an experiment I tried building piscsi with -flto optimization on my RPi3B / Raspbian Bullseye. It finds these potential errors that aren't detected when building without optimization.

```
g++ -flto -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++17 -iquote . -D_FILE_OFFSET_BITS=64 -D_LARGEFILE64_SOURCE -MD -MP  -DFMT_HEADER_ONLY -DCONNECT_TYPE_FULLSPEC  -o bin/fullspec/scsictl obj/fullspec/scsictl_commands.o obj/fullspec/scsictl_display.o obj/fullspec/scsictl_parser.o obj/fullspec/scsictl_core.o obj/fullspec/scsictl.o obj/fullspec/piscsi_version.o obj/fullspec/piscsi_util.o obj/fullspec/protobuf_util.o obj/fullspec/protobuf_serializer.o obj/fullspec/piscsi_interface.pb.o -lpthread -lprotobuf
./devices/scsi_host_bridge.cpp: In member function ‘WriteFs’:
./devices/scsi_host_bridge.cpp:884:20: error: ‘dpb.sector_size’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  884 |  dpb.sector_size = htons(dpb.sector_size);
      |                    ^
./devices/scsi_host_bridge.cpp:885:19: error: ‘dpb.fat_sector’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  885 |  dpb.fat_sector = htons(dpb.fat_sector);
      |                   ^
./devices/scsi_host_bridge.cpp:886:17: error: ‘dpb.file_max’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  886 |  dpb.file_max = htons(dpb.file_max);
      |                 ^
./devices/scsi_host_bridge.cpp:887:20: error: ‘dpb.data_sector’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  887 |  dpb.data_sector = htons(dpb.data_sector);
      |                    ^
./devices/scsi_host_bridge.cpp:888:20: error: ‘dpb.cluster_max’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  888 |  dpb.cluster_max = htons(dpb.cluster_max);
      |                    ^
./devices/scsi_host_bridge.cpp:889:20: error: ‘dpb.root_sector’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  889 |  dpb.root_sector = htons(dpb.root_sector);
      |                    ^
lto1: all warnings being treated as errors
lto-wrapper: fatal error: g++ returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [Makefile:202: bin/fullspec/piscsi] Error 1
make: *** Waiting for unfinished jobs....
```

uweseimet, 2023-07-05:
That's interesting. Does this also happen with the issue_1179 branch? This is the one I have been using my tests.

rdmark, 2023-07-06:
It does not happen with your work branch, so we're good.

uweseimet, 2023-10-03:
I would like to either discard or resolve this ticket. So what are we going to do? Making -flto a standard option is definitely not possible because Pis with only 512 MB of memory will not be able to compile/link anymore. My favorite solution is to add a note to the Wiki.

rdmark, 2023-10-04:
I second the resolution of adding a note in the wiki about this option, and not make easyinstall.sh any more complex.

uweseimet, 2023-10-04:
I added a note on link time optimization at the bottom of https://github.com/PiSCSI/piscsi/wiki/Setup-Instructions, right after the existing note on using alternative linkers.

closed 2023-10-04 by uweseimet