PiSCSI/piscsi #1182: Update logging to use C++ strings, try to get rid of log.h

uweseimet, 2023-07-04:
This should be addressed *after* the bookworm related changes have been committed in order to avoid merge conflicts.

At least with recent versions of the spdlog library the logging methods (spdlog::info, spdlog::debug etc.) accept C++ strings/string_views as arguments. This means that instead of
```
LOGINFO("Renamed/Moved image file '%s' to '%s'", from.c_str(), to.c_str())
```
just like with any other library we can just write
```
spdlog::info("Renamed/Moved image file '" + from + "' to '" + to + "'");
```
There is no need for using any macros and log.h most likely becomes obsolete. Note that current versionf of C++ in general aim at reducing the need for using macros.

Before working on this ticket it should be verified that this change works for the spdlog releases provided by bullseye and bookworm.

uweseimet, 2023-09-25:
@rdmark Just in case you (or somebody else, of course) can spare some time: If everything goes to plan (also see https://github.com/PiSCSI/piscsi/pull/1213) I will create a PR for #1179 and then #1182 next week. #1182 includes all changes from #1179. I have cleaned up a lot of code (except for the legacy stuff), and by making better use of C++ features (not only C++-20 but also older) rather tricky SonarQube issues have been resolved and the code volume has decreased without losing functionality. Based on my changes I could also improve the test coverage.
I would welcome help with testing #1182 (not #1179, it's just an intermediate step). The respective branch is test_issue_1182. There are no known issues so far (I tested with bullseye and bookworm), but Murphy's law says that somebody else will find some ;-).

rdmark, 2023-09-25:
@uweseimet Thanks for the heads up! This week I'm in the middle of preparing a move from Seattle to Tokyo (for work) so I won't have time for piscsi for some time, unfortunately. Once I'm settled down, maybe 2-3 months later, I'll lean back into the project and see what I can do to help.

uweseimet, 2023-09-25:
@rdmark Seattle to Tokyo, wow! I hope everything works as planned.

closed 2023-10-15 by uweseimet