PiSCSI/piscsi #1188: Drives not detected

hydrohs, 2023-07-15:
# Info

- Which version of Pi are you using: 3B
- Which github revision of software: 23.04.01
- Which board version: 2.5b
- Which computer is the PiSCSI connected to: Macintosh SE and Power Macintosh 6500

# Describe the issue
I can’t get either of my Macs to detect disks via the PiSCSI. SCSIProbe on my Mac SE says the bus is not terminated if I attach a drive (if the PiSCSI is physically connected, but no drives are attached it’s fine) with the dip switches in either configuration. On the Power Mac, doing a scan with Lido just gives bus errors with a drive attached. Since it appeared to be a termination issue I checked the connections with my multimeter and everything seems to be going where it’s supposed to.

[image]
[image]

rdmark, 2023-07-15:
Did you go through the steps in https://github.com/PiSCSI/piscsi/wiki/Troubleshooting ?

Do other SCSI devices work with the same chain? There's always the chance that there are hardware issues, oxidized contacts, poor SCSI cables, etc.

Have you read through https://github.com/PiSCSI/piscsi/wiki/Drive-Setup and made sure you're using good images?

hydrohs, 2023-07-16:
I totally missed the troubleshooting page, thank you.

I don't have a loopback cable, though I think I have what I need to make one. However checking voltages with my multimeter; D5 is stuck low and D6 is stuck high, everything else is good. I see on the schematic both of those go through IC2, checking again for continuity from the IC to the GPIO pins everything seems connected. Is it just a bad chip, then?

akuker, 2023-07-18:
Its possible. I've had a few bad chips, but not many. Depending upon how good you are with a solder rework station, you could swap IC1 and IC2. IC2 is only used for the data parity bit (on this version of the board). So, if your host ignores parity (Macintosh's do), that might be a solution. Or, at least narrow it down to a specific part.

rdmark, 2023-09-25:
@hydrohs were you able to get to a working piscsi setup?

hydrohs, 2023-10-16:
Sorry, I have not been able to get it working, just not able to get the parts for testing so I will mark this closed. Appreciate the help.

closed 2023-10-16 by hydrohs

uweseimet, 2023-10-16:
@hydrohs Thank you for your swift response. Just re-open the ticket at some time in the future if needed.