PiSCSI/piscsi #1191: Error when attaching ethernet adaptor

timoog, 2023-07-26:
Raspberry Pi 2b + wifi usb dongle
PiSCSI software version: 23.4.1
Fullspec board
ssh to macbook pro

After running easy install option for wireless set up, in web interface attach ethernet adaptor and receive error:

`Configure the network bridge for enxb827eb9faba9 first: /etc/dhcpcd.conf, /etc/network/interfaces.d/piscsi_bridge`

also ran easy install option for wired network and again receive error:

`Configure the network bridge for enxb827eb9faba9 first: /etc/dhcpcd.conf`

Went to config and set _Enable predictable network names_ reboot and still receive 
`Configure the network bridge for enxb827eb9faba9 first: /etc/dhcpcd.conf`

rdmark, 2023-07-27:
So I think the immediate issue is solved by correctly configuring the dhcpcd.conf ipv4 deny setting for the enxb827eb9faba9 interface here.

I put up a PR to improve the way we report the network bridge status in the Web UI, which should help with the misunderstanding:
https://github.com/PiSCSI/piscsi/pull/1192

rdmark, 2023-07-27:
One additional improvement that could be made here, is to change the verbiage in easyinstall.sh to make it clearer that `eth0` and `wlan0` aren't the only potential options. Since the original confusion arose from the fact that when enabling "predictable interface names" for a Debian system, you get these `enx` and `wlx` + MAC address names instead.

rdmark, 2023-07-29:
The second improvement in this PR: https://github.com/PiSCSI/piscsi/pull/1193

With this change, the script will pick the first available (in the order that `ip addr show` lists them) linked network interface as the default. The grep filter now checks for the enx/wlx pattern as well so that it works with predictable names.

rdmark, 2023-07-31:
Improvements are merged to develop now. Please test & provide feedback if you find any other corner case that needs addressing!

closed 2023-07-31 by rdmark