PiSCSI/piscsi #1195: Buster deprecation: easyinstall.sh

rdmark, 2023-07-29:
This ticket is to track modifications to easyinstall.sh required for the Buster deprecation

rdmark, 2023-07-29:
At this moment I have identified one required update: The removal of the locking of clang to v11. This was required on Buster since it shipped with a too old version of clang (v7) as the default, and we had to specifically use the v11 package.

On Bullseye, v11 is the default, and on Bookwork the default is v14 which has no issues with our codebase in my testing.

PR that removes the locking to v11 and defaults to the distro default https://github.com/PiSCSI/piscsi/pull/1194

uweseimet, 2023-07-29:
@rdmark Please take note of my changes to easyinstall in the issue_1179 branch. What about adding any required changes to easyinstall to this branch?

rdmark, 2023-07-30:
@uweseimet Normally I prefer to make smaller stand-alone changes in separate PRs. But since you already have changes in your branch I can discard my branch.

BTW, what do you think about locking to a newer version (such as clang-13 in your branch) vs. letting it float with the default for the distro (as in my branch)? I can see pros and cons, but my choice would be to let it float since it removes the maintenance overhead of managing the package and compiler version.

uweseimet, 2023-07-30:
@rdmark If the default compiler works fine I agree that there is no need for using a newer version, unless it turns out that there is an issue with the old compiler.
In case you test with the default compiler, can you please also test whether the -flto option produces correctly working binares?

rdmark, 2023-07-31:
@uweseimet I tested Raspbian bullseye default clang (11.0.1-2+rpi1) with `-flto` using your work branch. The resulting piscsi and scsictl binaries seem to work well, although I did only cursory dynamic testing.

I'll test on Raspbian bookworm once it's released.

uweseimet, 2023-10-16:
@rdmark Is there still anything to be done for this ticket, or can we close it?

rdmark, 2023-10-22:
Nothing else that I can think of.

closed 2023-10-22 by rdmark