PiSCSI/piscsi #1198: Tweak the way DaynaPort inet is displayed to the user

rdmark, 2023-07-31:
Presently, when you attach a DP device without parameters you get a default `inet=10.10.20.1/24` parameter. However, this inet address range is irrelevant for DHCP / piscsi_bridge setups. We've had several instances now when users have been confused by this parameter and configured their clients with a static IP in this range rather than letting their router's DHCP server assign the address.

This ticket is for changing how we display the inet parameter in clients to decrease the risk of misunderstanding.

fdanapfel, 2023-11-25:
The network setting that is displayed for the DaynaPort device in the web-interface seems to be hardcoded, because even after changing the setup for the WLAN config as described in https://github.com/PiSCSI/piscsi/issues/1387 I still see 10.10.20.1/24 displayed on the web-interface.

rdmark, 2023-11-26:
@fdanapfel Did you pass `iface` parameters when attaching the daynaport device? Either with the `-f` parameter of scsictl or the FILE object of piscsi proper (upon startup).

One example of the syntax in the man page: https://github.com/PiSCSI/piscsi/blob/main/doc/piscsi_man_page.txt

fdanapfel, 2023-11-26:
@rdmark No, as you can see in https://github.com/PiSCSI/piscsi/issues/1387 I'm attaching the DaynaPort device with the following command:
`scsictl -i 6 -c attach -t scdp -f wlan0`

rdmark, 2023-12-03:
@fdanapfel Right, so if you want to override the default `10.10.20.1/24` you need to pass a different parameter to piscsi or scsictl as per the above.

fdanapfel, 2023-12-03:
@rdmark Yes, I'm aware of this. But the point is that if I manually remove the IP-Address from the piscsi0 interface and delete the bridge the web interface should no longer show the IP-address.