PiSCSI/piscsi #1200: Optimize build options per RPi model

rdmark, 2023-07-31:
We currently have a one size fits all default build option  in easyinstall.sh  (clang++-11 with one thread) which always works on any RPi, but isn't optimized for build time.

For instance, on RPi 3 or 4, using 2 or 3 threads cuts build time by 1/2 or 2/3. But on a Zero 2 W, we can only use one thread because otherwise the system runs out of memory.

Another example, the Zero W, gcc is notably faster than clang. Building with clang takes 4h at the time of writing, while gcc takes 3.5h.

This ticket is for exploring the potential of detecting particular models or capabilities and dynamically choose the default options.

rdmark, 2023-07-31:
One solution would be to inspect /proc/cpuinfo

```
$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000f14495c9
Model           : Raspberry Pi Zero W Rev 1.1
```

uweseimet, 2023-08-01:
We already had a simple solution (only) taking the number of processors into account. It used the `cpuinfo`command in order to determine the number of CPU cores. Directly using(parsing /proc/* is not the best choice if there are regular commands providing the desired information.
If there are platforms where clang is slower than gcc it is very likely due to swapping because of not enough physical memory.

Just adding a note to the Wiki may be best. One can simple configure the MAKEFLAGS environment variable with the number of course to use. On my Pi 4 I just set in my profile:
```
export MAKEFLAGS=-j4
```
This is a very simple approach. Spending time on more complicated solutions, which can never yield better results than just setting MAKEFLAGS once for the respective platform. may not be such a good idea, considering that there are still several open (bug) tickets.

rdmark, 2023-08-01:
The problem with using all available CPU cores, is that there are devices with multiple cores but too little memory to compile multi-threaded. The currently known example is the Zero 2 W, which has 4 cores but only 512MB RAM. Use any more than one thread and the system freezes up with both gcc and clang. Other such models may be manufactured or discovered in the future.

Long compile times when updating is continuously reported as a pain point by our users. I want to do something to improve the situation for them.

uweseimet, 2023-10-06:
@rdmark Any chance to resolve this issue? If you want to add a not too complicated rule of how many cores to use when compiling I suggest to set -j to the availabile physical RAM in MB divided by 512, but never to set it higher than the actual number of cores. This results in one more parallel process for each 512 MB of physical memory, as long as there are enough cores.

rdmark, 2023-10-07:
@uweseimet I like this idea. It seems scalable and safe. I don't have access to any RPis right now to test on real devices (being shipped across the Pacific ocean) but I can do the theoretical update to easyinstall.sh

rdmark, 2023-10-07:
PR in https://github.com/PiSCSI/piscsi/pull/1239

closed 2023-10-07 by rdmark