PiSCSI/piscsi #1201: DaynaPORT Issues With MacTCP

alexthecat123, 2023-08-02:
# Info

- Which version of Pi are you using: Pi 4 Model B 
- Which github revision of software: v23.04.01
- Which board version: akuker 2.5
- Which computer is the PiSCSI connected to: Mac Quadra 700

# Describe the issue

I'd like to start by saying that the PiSCSI is absolutely awesome! By far the best SCSI drive replacement that I've ever had, so great job to everyone who has helped make this possible!

Unfortunately, I've been having some problems with getting the DaynaPORT emulation to work with MacTCP. I'm using a Quadra 700, so I can run Open Transport (with DHCP) as well and that works perfectly, but MacTCP just doesn't. I've configured everything according to the steps on the DaynaPORT wiki page, but I always just get timeout errors in MacTCP Ping whenever I try to ping 8.8.8.8. Running "ip addr show" and "brctl show" on the Pi gives the same results shown on the wiki page too, so I'm pretty sure I have everything configured properly on the PiSCSI end of things. I've tried MacTCP with the PiSCSI on several different machines (including an SE) and encounter this same problem every time, so it doesn't seem to have anything to do with the specific computer either. What could be going on here?

The reason why I need MacTCP instead of just using Open Transport is because the eventual goal is to get this thing working with MacWorks Plus on the Apple Lisa, which can't run Open Transport thanks to its 68000 processor. I'm just trying to get a proof of concept working on the Quadra 700 before trying anything on the Lisa itself since the Quadra is WAY faster and easier to transfer files to than the Lisa. 

Thanks for the help!
Alex

rdmark, 2023-08-03:
@alexthecat123 Thank you for the kind words! We're doing our best to deliver a useful tool for old SCSI equipped computers.

So yes, it is notoriously tricky to get MacTCP to play ball with a modern network. It's a moody beast. From the network configuration, it seems like you're trying to connect to a LAN where a DHCP server controls the IP assignments, correct? Can you please share how you have configured MacTCP on the Quadra?

Also, if you haven't done so already, please read through all the steps and troubleshooting advice on our wiki: https://github.com/PiSCSI/piscsi/wiki/Dayna-Port-SCSI-Link

alexthecat123, 2023-08-03:
Thanks for the response! Yeah, MacTCP can be a real pain to deal with sometimes!

Yep, that's exactly right about my network; it's a LAN with a DHCP server for IP address assignment. It just sucks that DHCP doesn't work in MacTCP because that would probably make things way easier!

I've read that wiki page probably at least 25 times now and I've tried everything that it suggests. MacTCP is configured exactly the way that it specifies with an IP of 10.10.20.2, the router IP of 10.10.20.1, a subnet mask of 255.255.255.0, and the Google DNS server. But I still get the timeout error in MacTCP Ping. 

And I apologize in advance if the problem ends up being something really stupid that I've overlooked; networking is BY FAR my weakest area of knowledge in computing and I know very little about it!

Here are some photos showing the exact network configuration that's being used on the Quadra!

[image]
[image]
[image]

rdmark, 2023-08-03:
@alexthecat123 I can see the problem here immediately. You need to configure MacTCP with IP / subnet / router addresses that align with the address ranges of your router. Looking at your wired network interface, that means in the 192.168.0.2~254 range (netmask of 24).

The pitfalls here are that 1) assigning static IPs in a DHCP managed network can lead to duplicate assignment of IP addresses, and 2) the router may reject connections from MacTCP altogether. I've had mixed results myself. You may want to consider putting up a separate router just for your vintage network that is lenient with this kind of configuration.

An alternative is to set up NAT for your bridge, which isolates it from the rest of your LAN and enable you to use whatever status IP address ranges that you want. But with the drawback that you can't reach the rest of your LAN from MacTCP.

I suggest you try the former approach first! You may be lucky and have it just work for you.

alexthecat123, 2023-08-04:
Thanks so much for the help! That makes a lot of sense; I was wondering why the wiki listed those 10.10.20.X IP addresses, but I guess those were specific to the author's router!

Unfortunately, the first approach didn't work for me. I changed the gateway to 192.168.0.1 and the Quadra's IP to 192.168.0.25 and then went into the router and reserved that IP for the DaynaPORT's MAC address, but no luck. So I guess I just have one of those routers that doesn't like MacTCP? 

Although this does seem a little strange because MacTCP works fine with my router when I use a NuBus ethernet card, but not with the PiSCSI. Could that offer any clues as to what's going on?

This means that I'm probably stuck with the NAT option for now. Sorry if this is a dumb question, but could you tell me how to set up NAT for the bridge? Like I said, I know very little about networking stuff!

Thanks again for all of the help!

rdmark, 2023-08-04:
> Thanks so much for the help! That makes a lot of sense; I was wondering why the wiki listed those 10.10.20.X IP addresses, but I guess those were specific to the author's router!
> 

No, the 10.10.20.X range is just what I arbitrarily chose as the default for the NATed network configuration... I thought I was explicit about those being example IPs that have to be adjusted for your environment! Earlier today I made a few edits that hopefully reduce the chance of misunderstanding, but please help pointing me to any ambiguous statements in the wiki. I most certainly have blind spots!

> Unfortunately, the first approach didn't work for me. I changed the gateway to 192.168.0.1 and the Quadra's IP to 192.168.0.25 and then went into the router and reserved that IP for the DaynaPORT's [...]

Have you run the Dayna diagnostics apps to make sure that the hardware and drivers are working correctly on your system?

> This means that I'm probably stuck with the NAT option for now. Sorry if this is a dumb question, but could you tell me how to set up NAT for the bridge? Like I said, I know very little about networking stuff!
> 
> Thanks again for all of the help!

Sure! It should just be a matter of following the wiki instructions.

First revert the "wired" network bridge settings, i.e. deleting piscsi_bridge and removing the line in dhcpcd.conf as described in https://github.com/PiSCSI/piscsi/wiki/Dayna-Port-SCSI-Link#wired-raspberry-pi-setup

Then, follow the steps for the NAT configuration, but substitute wlan0 with whatever your wired network interface is called: https://github.com/PiSCSI/piscsi/wiki/Dayna-Port-SCSI-Link#wireless-raspberry-pi-setup

alexthecat123, 2023-08-06:
> Have you run the Dayna diagnostics apps to make sure that the hardware and drivers are working correctly on your system?

I have and the hardware seems to be working properly!

Thanks for the NAT instructions! Unfortunately, after following your instructions I now get the error message "Configure the network bridge for eth0 first: /etc/dhcpcd.conf, /etc/network/interfaces.d/piscsi_bridge" whenever I try to attach an ethernet adapter. What should I do to get around this?

uweseimet, 2023-11-08:
@rdmark What is going to happen with this ticket? Is this actually something the requires changes/fixes in PiSCSI?

rdmark, 2023-11-08:
@alexthecat123 Sorry for leaving you hanging there! The workaround there is to use the `scsictl` command line tool to attach the DP device. We have validation on the Web UI side to prevent incomplete network bridge configuration. But looking at your case, I think I will change that validation to a warning instead of error.

Please let me know if you are still available to keep troubleshooting this, and I can give you a few commands to try!

rdmark, 2023-11-08:
This PR will make the Web UI more lenient about attaching networking devices when it can't detect the "normal" bridge settings. https://github.com/PiSCSI/piscsi/pull/1319

alexthecat123, 2023-11-13:
@rdmark I'm pretty busy with college right now, but I'll try to make time for troubleshooting soon. You can go ahead and send me the commands and I'll try them out next time I get the chance!

alexthecat123, 2023-11-14:
@benjamink I haven't tried that yet, but I'll give it a shot as soon as I can!

silvi89UD, 2023-12-16:
Hi! I'm silvano from italy. I have an issue with my Powerbook 520c which runs system 7.5. I've been able to run the BlueSCSI pico wifi model but I can't have an internet connection. From OpenTrasnport 1.3 menu I've been selected Alternate ethernet and DHCP server. When I try to use an internet browser such as icab, an error appears.There's no Internet connection. this is the log file:
Platform: BlueSCSI Pico
FW Version: 2023.11.16-release Nov 17 2023 03:26:27
Flash chip size: 2048 kB
 
=== SD Card Info ===
SD card detected, FAT32 volume size: 30436 MB
SD Name: SD32G, MID: 0x03, OID: 0x53 0x44
 
=== Global Config ===
Reading configuration from bluescsi.ini
Active configuration:
 
=== Finding images in / ===
== Opening /NE4.hda for ID: 4 LUN: 0
---- Configuring as network based on image name
---- Image ready
== Opening /HD10_512 PowerBook 520c 240MB.hda for ID: 1 LUN: 0
---- Image ready

=== ROM Drive ===
Platform supports ROM drive up to 1692 kB
---- Activating ROM drive, SCSI id 0 size 1692 kB
---- WARNING: This image does not appear to be a valid Macintosh Device image. See: https://github.com/BlueSCSI/BlueSCSI-v2/wiki/Disk-Images
---- Activated ROM drive, SCSI id 0 size 1692 kB
 
=== Configured SCSI Devices ===
* ID: 0, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 1692kB
* ID: 1, BlockSize: 512, Type: Fixed, Quirks: Apple, Size: 245824kB
* ID: 4, Type: Network, Quirks: Apple
 
=== Network Initialization ===
Wi-Fi MAC: 00:80:19:07:E3:38
Connecting to Wi-Fi SSID "" with no authentication
Wi-Fi connection failed: -4
 
Initialization complete!
INFO: Pico Voltage: 3.276V.

Someone can help me please???
thankyou!