PiSCSI/piscsi #1202: BlueSCSI and PiSCSI coexistence issues

ShadeDream, 2023-08-07:
# Info

- Which version of Pi are you using: 2b
- Which github revision of software: v23.04.01
- Which board version: 2.4a
- Which computer is the PiSCSI connected to: Mac SE, Quadra 700

# Describe the issue
Generally when a BlueSCSI (v1 or v2) is connected internally and PiSCSI externally on the same machine I run into issues with behavior of the two together.

Q700 - BlueSCSI v2 + PiSCSI - Bad

Using the two devices together in this configuration and booting from the BlueSCSI results in _very_ slow boot times and corruption of data on the BlueSCSI disk image. This blueSCSI I believe is running the latest firmware (v2023.07.03).

SE - BlueSCSI v1 + PiSCSI - Bad

In this case I was booting from the PiSCSI to install an OS on the BlueSCSI v1 images. The machine boots up fine from the PiSCSI but the BlueSCSI is completely inaccessible and doesn't show up. Running Apple SD HC to see the drives is _very_ slow and only finds the PiSCSI devices. BlueSCSI log file reports it's seeing the images and operating normally. The BlueSCSI v1 is an earlier model and running an older firmware because it requires a programmer to upgrade (which I don't have).

Molar G3 - BlueSCSI v2 + PiSCSI - Fine?

Out of curiosity, I tossed one of the BluSCSI v2s in a gossamer Molar G3 and connected the PiSCSI up. I was booting from my IDE CF card, but I was able to see and interact with both devices in that case for whatever reason.

IIsi - SCSI2SD 5.1 + PiSCSI - Good

No issues in this case, only including to note that I've tested this as well.

benjamink, 2023-08-07:
Performa 475 - External BlueSCSI v1 + Internal PiSCSI - Good

I was able to boot just fine with my Performa with a PiSCSI v2.7 internally & a BlueSCSI v1.1 externally.  BlueSCSI has the jumpers both on `0` termination DIP switches both set to `ON` on the PiSCSI.

erichelgeson, 2023-08-07:
I'll add to this as I test more machines but this is something I've not come across. My tests are with a 2.4 board and a Pi4 with BlueSCSI v2 

* Powerbook 160 - worked fine. PiSCSI external, internal BlueSCSI
*

rdmark, 2023-08-07:
What *hasn't* worked for me:

* IIsi with BlueSCSI v1.1 internally, and PiSCSI externally
* SE/30 with BlueSCSI v2 internally, and PiSCSI externally

uweseimet, 2023-10-02:
Looks as if a termination issue has been identified as the source of the problem. Can this ticket be closed?

closed 2023-10-15 by uweseimet