PiSCSI/piscsi #1204: Macintosh SE FDHD won't boot from piscsi on cold boot

kirbyfrugia, 2023-08-14:
Info
Which version of Pi are you using: 3 Model A
Which github revision of software: v23.02.01
Which board version: 2.4a
Which computer is the PiSCSI connected to: Mac SE FDHD

Describe the issue

I can't do a cold boot straight from an image on my piscsi. It enters into a weird infinite loop. See [this video](https://youtube.com/watch?v=Wi8jjQ_EbXM&feature=shareb).

What works:
0. Insert bootable floppy
1. Power up piscsi
2. Power up Macintosh
3. Boot up from floppy
4. Eject floppy
5. Run system picker and pick an image from the piscsi
6. Reboot from image
7. All is good

What doesn't work:
0. No bootable floppy in drive
1. Power up piscsi
2. Power up Macintosh
3. Watch it loop like in the video

rdmark, 2023-08-14:
Have you tried with other hard drive images? It could be a problem with the embedded driver, either corruption on the boot partition or a marginal compatibility problem. You can try using Disk Jockey to swap out with other drivers.

kirbyfrugia, 2023-08-19:
@rdmark I detached all drives except my main one and it's working now. There must have been something wrong with one of my other drive images. Thanks!

rdmark, 2023-08-19:
Good to hear! Yes Macs can be picky about specific drive images and the partition schemes on them.

closed 2023-08-19 by rdmark