PiSCSI/piscsi #1206: Undervoltage notification in Macintosh Desk Accessory

benjamink, 2023-08-31:
I'm proposing a new feature that provides indication of under voltage events from the PiSCSI to the host via some form of flag/data on the SCSI bus.  The idea being that utility applications on the host can monitor for this information on the bus & provide some sort of notification to the user when these events occur.

My initial plan would to be to develop a Desk Accessory for Macintosh System 6+ that could utilize this information appropriately as an initial proof of concept.

benjamink, 2023-08-31:
Related Discord conversation started here: https://discord.com/channels/749261651978748014/793579557197971488/1146861523520733374

uweseimet, 2023-09-24:
@benjamink Can you please provide more insight about the benefits of such an accessory? My (maybe naive) assumption is that if there is an undervoltage condition the PiSCSI board may not work anymore. If this is the case any SCSI-based solution would not help. If the board is still fully functional in such a condition, why is undervoltage an issue at all? How often does such an issue actually happen, and why can it happen? It should happen quite frequently in order to justify the implementation effort.

Note that any solution that requires frequent polling on the SCSI bus would have a negative impact on performance. A SCSI-based solution, if it makes sense at all, would most likely implement an additional mode page for the existing host services device. Provided that there is a C API that can inquire the undervoltage condition adding such a mode page would be quite simple.

If there was an accessory, how often should it poll, how should it indicate an issue to the user, and what is the user supposed to do if there is an undervoltage scenario? Would the user have to shut down PiSCSI or the Pi? In this case it might be better if piscsi shut itself down automatically.

In case the board does not work reliably anymore in an undervoltage condition, a client polling the piscsi remote interface (e.g. the PiSCSI Control App) would IMO be the only alternative. The app might send a push message in case of an undervoltage condition.
The client interface approach also works if SCSI does not work anymore, as long as there is a network connection to the Pi. It also requires a C API, and polling should not happen to often.

benjamink, 2023-09-24:
@uweseimet good points about the chicken/egg issue of detecting an issue when the SCSI bus is already likely unstable/unusable due to the undervoltage event.  One place where this might still be useful is if someone were to reboot due to a hang situation caused by undervoltage without knowing why things hung.  On next boot of the Mac they could detect the "Previous undervoltage" flag as per `vcgencmd` on the Pi & raise some kind of notification in the Mac UI.  

Honestly, I'm not sure if the idea makes total sense.  I was just thinking about it after working on https://github.com/PiSCSI/piscsi/issues/1199 & wanting to learn C on the Mac.  This seemed like an interesting idea for a project to learn how to do it.  I'm not a programmer AT ALL so I really don't know exactly how to go about doing it yet (other than what you said & what has been said in Discord) but it seemed like an idea that might be useful.  

Until I have the slightest idea how to even get started (again, I know VERY LITTLE about C or C++) I really can't even begin to prototype the idea.

uweseimet, 2023-09-24:
@benjamink Besides probably not being a valid use case (i.e. not really being implementable, at least not with SCSI-based means), I don't think this kind of ticket would help with learning C++. Besides learning object oriented programming and C++ there are several additional challenges:

- You need to know how SCSI works in general
- You need to know how the Mac's SCSI API works
- You need to know how to implement accessories on the Mac in general
- You need a suitable development environment for the old Macs we are dealing with

These items are even challenges for an experienced C++ developer. In order to learn C++ there is IMHO no other way than reading good books about the language in particular and OOP programming in particular.

This is why I would like to suggest to close this ticket as "wontfix", at least for now, because there are simply too many preconditions required, and too many open questions.

benjamink, 2023-09-24:
Fair enough.  No idea when I'd get to this anyhow.  Closing.

closed 2023-09-24 by benjamink

uweseimet, 2023-09-24:
@benjamink I would like to ask you for a bit more feedback regarding the alternative approaches I mentioned, i.e. to let the protobuf client interface report the undervoltage condition, so that an external (non-SCSI) client (like an app) can report an issue, or to let piscsi itself handle it. These are my questions:

1. What is a user supposed to do if such a condition is reported, e.g. by an app sending a push message?
2. Is there something piscsi could do automatically, e.g. to log this issue and then immediately shut down?
3. How frequently should the undervoltage condition be checked?

benjamink, 2023-09-25:
I don't have definitive answers for your questions.  Here's my uneducated thoughts: 

1. In the notification added to the web UI the user is given basic hints on improving their power supply or air-flow/cooling depending on the messages shown.   I would expect to give similar short hints with a link to the Raspberry Pi documentation for more details.
2. I don't think every occurance of low-voltage or high temperature will cause instability or unresponsiveness so I'd hesitate to take any automatic actions other than notification/awareness.  In some cases an IO action might be delayed slightly but succeed or it may hang the system altogether.  Forcing an automatic response/reboot of the Pi could be completely unnecessary & lead to data loss when it's avoidable.  I don't know enough about the conditions & how they affect operations to know what the best course of action is but I know that when things become unstable due to undervoltage or high temps it would be nice to have some indication of that.
3. Again, this is hard to answer as the event can be extremely intermittent, rare or unexpected.  Obviously checking too frequently would be a performance issue but checking too infrequently might miss the condition in time to warn about it before the system hangs.  Since I'm not that knowledgeable about how the polling would affect performance I'm not sure how to answer this.  If doing an API call it seems that polling every few 5-10 seconds (or even more frequently) for a single byte would be trivial but I could be misunderstanding the consequences.  If only monitoring for, say, current undervoltage, previous undervoltage, current overheat, previous overheat then you're looking at something that can be represented in 3 bits, right?  Add one more bit & you can represent all the states that `vcgencmd` provides.  It seems that the performance hit for this would be pretty miniscule.

uweseimet, 2023-09-25:
@benjamink Thank you for your detailed feedback. Regarding any API call, much more than 3 bits would have to be transferred. Regardless of the kind of API (e.g. protobuf or REST), there is always a protocol overhead (espcially with REST), and data structures and not just single bits are transferred.

uweseimet, 2023-09-25:
Some more thoughts on this. The process that would produce the least overhead when checking for such a condition would be piscsi, provided there is a C API that gives proper access to the required data. piscsi would not even have to check strictly periodically, which would require a separate thread. It would be sufficient to check, for instance, at the beginning of a bus free phase, where piscsi also checks whether the host services device was sent a SCSI shutdown message. This means that if there is no bus activity (no SCSI commands received) piscsi would not check for this condition, but this is acceptable because one may assume there will be regular bus activity. Otherwise you would not have any need for piscsi ;-). My guess is that any voltage issue will not pop up anyway when there is no bus activity, so checking right after each command (bus free phase) is probably the right place. piscsci would then simply remember that there was an undervoltage condition, or maybe how many of them.

Any client could get the respective status from piscsi (using the protobuf remote interface) at infrequent intervals, maybe just every few minutes. As long as piscsi remembers the undervoltage status, any client can learn about it any time, so there is no need to check frequently.

All in all this asynchronous approach combines a low polling overhead with the advantage of checking the status quite regularly and at times where is issue is quite likely. An external process cannot do that.

@benjamink Is there any way to properly inquire the status in C/C++, e.g. with a system call? If there is, the implementation effort may be rather low, and I might extend the remote interface accordingly.

Actually, I am a bit worried about regularly polling the status with an external process. I guess this is what the web UI does The overhead may not be neglectable, and the more polling of that kind is done, the more in particular Pis with only a single core or just 512 MB of memory may suffer. Polling might happen during an active data transfer and might cause the Pi to swap ...

benjamink, 2023-09-25:
I believe this is the C source for the `vcgencmd` utility that should have what you're looking for.  https://github.com/raspberrypi/userland/blob/master/host_applications/linux/apps/gencmd/gencmd.c

The web UI runs the `vcgencmd` on page refresh so not that frequently.   It's a static notification header based on the state of the command's output at web page refresh/reload time only.

uweseimet, 2023-09-25:
OK, so the overhead of running vcgencmd should be neglectable. On the other hand, does this mean that the condition is only reported if the user refreshes a page at exactly the right time?

uweseimet, 2023-09-25:
I am just having a look at the source code of vcgencmd. It's not much, but I cannot see how the tool is related to an undervoltage condition. (I'm not a Raspberry Pi expert.) Maybe you can explain in a few words, how the condition is related to the video core (this seems to be what the tool is dealing with) and how the command line for using this tool in order to detect an undervoltage condition looks like?

benjamink, 2023-09-25:
> OK, so the overhead of running vcgencmd should be neglectable. On the other hand, does this mean that the condition is only reported if the user refreshes a page at exactly the right time, i.e. at a time when such a condition actually exists?

That's correct.  It's far from an ideal notification but it was intended as a very simple implementation that at least gives some notice of an undervoltage situation having occured.  The command provides statuses for both currently occurring undervoltage as well as a persistent flag for when undervoltage has EVER been seen since the Pi was last rebooted.  The latter notification is more likely to be what is seen & the hope was that if you're seeing some odd behaviors & you go to the web UI you can see if there might have been some undervoltage occurring at some point.  If you want more details you can also look at the system logs (eg, `dmesg`) & see the events logged there.

> @benjamink I am just having a look at the source code of vcgencmd. It's not much, but I cannot see how the tool is related to an undervoltage condition. (I'm not a Raspberry Pi expert.) Maybe you can explain in a few words how the condition is related to the video core (this seems to be what the tool is dealing with) and how the command line for using this tool in order to detect an undervoltage condition looks like?

Yes, I've found this odd too.  I'm not an expert in any of this either so I can't tell you why a utility for looking at video details is also used for viewing system voltage/throttling details other than to go look at the docs online with you.  I found out about this command through a search on how to identify undervoltage & this is the command that's documented to do that in Raspberry Pi's official docs.  Sorry I can't be much more help than that - I wish I knew this stuff at a much lower level than I do.

Again, the idea with some kind of Desk App was entirely born out of the idea of learning some C & the thought that having this notification be more prevalant on the Mac you're using would be very handy.  It's gotten a lot more technical in this conversation (which is & has been awesome!).  I'm assuming because we're still having this conversation that you find the idea interesting as well?  I hope what little context I've been able to provide so far has been somewhat helpful.  

Here's the Raspberry Pi docs on `vcgencmd`: https://www.raspberrypi.com/documentation/computers/os.html#vcgencmd

uweseimet, 2023-09-25:
@benjamink Thanks again for the feedback. vcgencmd gets its data from the /dev/vcio device file. This is why vcgencmd is that small: it does not really do much. Using device files is not as nice as having a proper API, though.
Anyway, I don't know yet whether I will follow up on this, but I might. In that case I would create a new ticket, summarizing the approach. Essentially I would inquire the condition in piscsi when such a condition is most likely, as indicated above, and I would expand the protobuf interface so that any interested client can retrieve this information directly from piscsi. Further I would extend the PiSCSI Control app to display this information and/or optionally send a push message when the condition has occurred. This appears like a reasonable way to me of alarming a user.

uweseimet, 2023-09-25:
I think I will create a new ticket anyway, because it helps me with refining the approach, regardless of whether it will be implemented by somebody or not.

uweseimet, 2023-11-15:
@benjamink Just for completeness sake: #1349 exposes most of the commands piscsi can execute via SCSI. You can, for instance, send a SCSI command in order to get the piscsi statistics. If reporting undervoltage should ever be implemented the undervoltage error condition would become a part of these statistics.
This means that if you want to implement a tool that is running on the Mac and wants to get this information, this tool would have to send the custom SCSI command described in #1349. The command might still slightly change, but the approach works fine and has already been tested.