PiSCSI/piscsi #1226: Add Raspberry Pi 5 Support

akuker, 2023-10-01:
# Info

The Raspberry Pi 5 has been announced with a new SoC. Since the GPIOs are now managed by the RP1 chip, the low level software interface has likely changed.

The PiSCSI code should be updated to support the Raspberry Pi 5 platform.

Performance tests are recommended to compare the performance to RPi 4.

akuker, 2023-10-01:
Note: I've pre-ordered RPi 5's, but if anyone else beats me to this, just comment below :)

uweseimet, 2023-10-02:
The GPIO pins of the Pi 5 are said to be fully backwards compatible, so it should work out of the box. See https://kitronik.co.uk/blogs/resources/the-differences-between-raspberry-pi-4-model-b-raspberry-pi-5, for instance.

Regarding performance let me try to make a prediction :). There won't be much of a difference compared to a Pi 4. I'm saying that because even though the Pi 4 is much faster than a Pi Zero (Zero, not Zero 2), if I recall my tests correctly, rascsi was only about 10% faster with a Pi 4.
If this is the case, just for running piscsi the Pi 5 is IMO not very attractive, because it needs more power and is more expensive than previous models, and you pay for features (PCIe) you cannot use.

akuker, 2023-10-02:
We'll find out when my Pi 5 arrives. It'd be great if no changes are needed!

>Drop it straight into your old projects, it's the same general size and shape as its predecessors and the if features the same backwards compatible GPIO interface as its predecessors.

I would be surprised if the low-level register definitions didn't change, but we'll see!

akuker, 2023-10-02:
The Raspberry Pi utility `pinctl` had updates specifically to support the RPi 5.

Reference: https://github.com/raspberrypi/utils/commit/2bd898fecb10c7259fc16efa8da22472d410a1d5

uweseimet, 2023-10-02:
So it's probably just the pinout that has not changed.

rdmark, 2023-10-03:
> If this is the case, just for running piscsi the Pi 5 is IMO not very attractive, because it needs more power and is more expensive than previous models, and you pay for features (PCIe) you cannot use.

There will be users who want to use an RPi5 with piscsi because that's what they have on hand, or for a multi-purpose setup with many different pieces of software running concurrently on the Pi. 

I agree that it shouldn't be the recommended hardware though, unless we see a huge performance increase (unlikely).

akuker, 2023-10-03:
Agree it probably won't be the recommended solution. But, still think the software should support it.

uweseimet, 2023-10-03:
@akuker Yes, definitely. And maybe I am wrong and there is even a relevant gain of performance. Compile times will be shorter in any case ;-).

@rdmark Regarding a multi-purpose setup you may remember my concern from the past that easysetup is too intrusive. It installs/configures a lot that that you may not want. The more you use want to use your Pi for different things, the more this can become an issue.

rdmark, 2023-11-20:
A user has reported in Discord that the Bullseye based release images do not boot on an RPi5. We likely need to update the scripts in https://github.com/PiSCSI/piscsi-pi-gen and merge the latest changes from upstream pi-gen.

TheNightmanCodeth, 2023-12-31:
re-posting from discord for visibility:

I can't get PiSCSI to work at all on my pi 5 without modification. At `gpiobus.cpp:397`, epoll_wait will continuously fail, flooding the logs (leading to > 2GB log file). The calls to `mmap` in the systimer and gpiobus inits would also fail with `EINVAL` (We don't like addr, length, or offset (e.g., they are too large, or not aligned on a page boundary)).

I discovered this was because `bcm_host_get_peripheral_address` returns `16` on the pi 5. I was able to get around this by hardcoding the peripheral base address in `GPIOBUS_Raspberry::Init` to match what was said in [this discussion](https://github.com/jgarff/rpi_ws281x/issues/528). I don't as of yet know of a way to retrieve the base address for the RP1.

Additionally, in `GPIOBUS_Raspberry::Init`, I had to replace `/dev/gpiochip0` with `/dev/gpiochip4`. 

This cleared any errors from the logs, but I haven't been able yet to confirm if anything is actually working. Doing so as I type this out.

[This document](https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf) contains the addresses and offsets for the RP1 chip which is where the peripherals are now accessed (as opposed to via the `BCMXXXX` chip on the other pis). My gut says these offsets will need to be updated as well.

jrcichra, 2026-10-02:
If it's helpful, I put the hat on a Pi5 (because that's what I had lying around) let Claude trial-and-error for a few hours, and now am able to boot and use System 6 on a Mac Plus without issues.

Patches are here: https://gist.github.com/jrcichra/2e94a4c62f5647c0ad9cfd1b34d7cd39