PiSCSI/piscsi #1227: Timer Test in scsiloop is not reliable

akuker, 2023-10-01:
# Info

The Timer Test in the scsiloop utility is not reliable. It will fail on some versions of RPi 4.

Its only usefulness is when porting to new platforms, NOT for testing out PiSCSI boards. Recommend disabling the test for now. In the future, it can be re-enabled as an command-line option.

akuker, 2023-10-14:
Removing myself as the assignee. The next time this file is changed, the timer test could probably just be deleted. It is only useful for porting to new platforms IMHO.

rdmark, 2023-11-20:
This was done in https://github.com/PiSCSI/piscsi/pull/1228

closed 2023-11-20 by rdmark

uweseimet, 2023-11-20:
@rdmark This means the code with the test can be removed? I am asking because #1228 does not fix this bug, it just does not run the code that has this bug anymore.