PiSCSI/piscsi #1237: Protobuf client interface: Add read/write and cache miss statics

uweseimet, 2023-10-05:
The piscsi core shall remember the number of read/write accesses and how often there was a cache miss. A new remote interface operation STATISTICS_INFO shall be added. This method shall be extensible, so that other information can be added (backwards comptible) if need should be. The statistics shall be categorized by an enum denoting the severity, e.g. INFO, ERROR. 
scsictl shall be extended with statistics support.

For consistency the data returned by this method shall also be returned by the GET_SERVER_INFO method.

Suggested protobuf interface:
```
enum PbOperation {
  ...
  // Get statistics (PbStatisticsInfo)
  STATISTICS_INFO = 32;
}

enum PbStatisticsCategory {
  NONE = 0;
  INFO = 1;
  ERROR = 2;
}

message PbStatistics {
  PbStatisticsCategory category = 1;
  // A symbolic unique item name
  string key = 2;
  uint32 value = 3;
}

message PbStatisticsInfo {
  repeated PbStatistics statistics = 1;
}

message PbServerInfo {
  ...
  PbStatisticsInfo statistics_info = 10;
}

message PbResult {
  ...
  // The result of a STATISTICS_INFO command
  PbStatisticsInfo statistics_info = 15;
}
```

uweseimet, 2023-10-05:
```
>scsictl -S
Statistics:
  cache_miss_read_count: 0 (INFO)
  cache_miss_write_count: 0 (INFO)
  sector_read_count: 5698 (INFO)
  sector_write_count: 20 (INFO)

uweseimet, 2023-10-11:
I would appreciate feeback on this ticket. @rdmark You helped a lot with refining the initial protobuf interface, maybe you can have a look at this extension.
I wonder:

- Could there be any statistics data that require a non-integer type for the value, e.g. a boolean? In cannot think of any.
-  Is uint32_t sufficient, or should it be uint64_t? I think uint32_t is definitely sufficient.
- Do we need the WARNING category? I tend to say no, at least I cannot find any use case for it.

rdmark, 2023-10-14:
@uweseimet So the idea here is to build a foundation that you will later use to implement undervoltage stats, as well?

At a glance I agree with your three conclusions. The INFO type is sufficient for, for example, notifying that undervoltage happened since the last power cycle (but not right now).

>   // A symbolic unique item name
Do you have a unique naming scheme in mind already?

uweseimet, 2023-10-14:
@rdmark I don't know whether I am going to implement undervoltage detection, but regarding a foundation you are right. Even without undervoltage detection support there are some pieces of internal status data (not visible from the outside, like an undervoltage issue) that piscsi can easily collect and that are probably useful and give some insight that is currently not available.

Regarding the categories, besides INFO the ERROR category is definitely needed. What I am not sure about is WARNING. ERROR is important for clients that want to inform the user of an issue. I have already tested this with my app, where I plan to optionally show a notification message, telling you when something is wrong, and an overview of all available statistics. The notification, which also pops up if the app is not the foreground app, only pops up in the ERROR case.

The current unique names are the ones you can see in the scsictl output above. This is the actual output of the implementation of this feature in the testing_issues_1237_1238 branch. These names should not change anymore once there is a release with this feature, so that clients can use them for internationalization. I am not sure whether there can be a naming scheme, but the names should speak for themselves. This way old clients can at least display something meaningful even for new values they don't know yet, i.e. they do not have a translation for. Do you have a particular naming scheme in mind?

uweseimet, 2023-10-14:
I tend to make the count a 64 bit value. This is cheap and guarantees that it will always be big enough. I also tend to keep the WARNING category, because as suggested in #1220 it may still be of use. Adding it later would potentially cause issues when sorting the categories by severity, because the enum value for WARNING would then be bigger than the one for ERROR, which is inconvenient at least.

I will add the device ID and LUN to each item. This at least optionally enables clients to provide more specific information. Read/write errors, for instance, might only affect a particular device, and by knowing the device one can identify the image file that causes issues. In addition, read/write error data per device may help improving the SEND DIAGNOSTIC SCSI command. Currently it never returns an error status, but if there are read/write errors it could. This is less likely than with real drives, but you never know.

Anything that is not device specific, e.g. an undervoltage condition, might in addition to the remote interface also be reported for the host services device. This device is a good candidate because essentially it represents the piscsi process as a whole.

uweseimet, 2023-10-28:
There is now also a warning, namely when the printer device receives the PRINT command but no data for printing are available.
piscsi_interface.proto contains the full set of statistics.

closed 2023-10-30 by uweseimet