PiSCSI/piscsi #1265: Need Help Or Info About HDD Format .HDS For Sharp x68000, How Open This And See For Work With This In editdisk (Disk Explorer)?

IgorPetrovich, 2023-10-24:
Hello!
I use and working in emulators from Computer Sharp x68000:
xm6_typeg_332L30_20201231
xm6_typeg_336_20230523
XM6 Pro-68k

My 2 qesutions: 
I have already mastered a lot for a very long time and have gradually delved into a lot of things, understood them for myself, discovered them, and figured them out... But there is still something left that I can't figure out on my own...
I barely figured out on my own how to create a Virtual Hard Disk HDD in .HDF format and install and run games from HDD (.HDF) in the xm6_typeg_332L30_20201231 emulator - it wasn't difficult, but I just can't figure out the HDD format (.HDS) - it's easy to create, but not I may not see it through the editdisk program (Disk Explorer), and emulators don't see it - not in DOS, not in shells ala Windows (Human 68k v2.01 (1989) (Sharp - Hudson), what could be the matter? Can you tell me ? I haven't found any information on this on the Internet... Does it need to be formatted somehow or something? After all, when creating an HDD (. HDF) there is Logical Formatting - formatting, when creating .HDS - there is no... It's the same as on modern PCs, you need to start by formatting the Hard Disks before working with them... editdisk (Disk Explorer) after creating .HDS does not open it at all, or rather does not see it... I have not worked with them before... I also opened it through Human 68k HDD and through Human 68k HDD SCSI In the graphical shell (Human 68k v2.01 (1989)(Sharp - Hudson)) - it also doesn't see... I know, about the walking X68000_V4SP2.HDS created by someone with a bunch of games on it - the system sees it, but I wanted to create my own large HDD (.HDS), and work with it, create it - I created it, and then what?
Of course, I inserted it where necessary, I already learned what was what in the meantime... HDD (.HDF) in SASI (Everything is easy and clear there), I created a new HDD (.HDS) in SCSI (Selecting Internal\Internal) Specifying everything correctly in BIOS, well, boot from what\Boot, etc., etc. I have already mastered the basics over time (For a long time...) But with .HDS I still don't understand... :(

Tell me, who knows?
I visited foreign forums, and even Japanese ones, read a lot of information, downloaded thousands of files and programs on Sarp x68000 - all in vain...

In other, simple words, withou big text...
This short question:
How to do, what new, empty HDD (.HDS), which i create in \ by use emulators can be visible and usable in editdisk (Disk Explorer)? For work with it...
Need special programs for formating HDD ( .HDS ) before use it or what? (Then which programs? )
Why emulators have option "Do logical formatting" only for HDD ( .HDF ), and do not have for HDD ( .HDS ) ?

Any ideas, who have information about it?
How creators of popular in internet archives  " X68000 " ; "X68000V4" - " X68000_V4.HDS ", with many games in it on HDD ( .HDS ) , have sucesfull with thsi task? WHat they used for it?

Btw, editdisk (Disk Explorer) opened their X68000_V4.HDS...
My new created by me in Emulators of Sharp x68000 empty HDD ( .HDS ) - no... :(
Any ideas, information, TIP's, Hints?
Need your help, guys... (  Need your information, TIP's, Hints, Tutorials or Lessons )

I want to know, understand, and use HDD ( .HDS ) for - tests, and because this can be have huge sizes of space (Even in Gbs ), much more than  in .HDF ( Which limited just in 40 Mb. )
But, sies not main reason, just want understand - how use .HDS?!
This main, reason, and can't do this task only by myself need your help, who have knows about it and expirence with work with this...

I just know, whta it is possible ( On example with X68000_V4.HDS by developers of this ), just not know how...
Need your help, guys...

2.:
I also remembered the second question:
How to reconfigure controls for 2 players on one keyboard? It only worked out halfway... It was on the xm6_typeg_332L30_20201231 emulator, and it was on this version, because there are 2 Ports with 00 JoyKeys, customizable for 2 players, it is possible on the same keyboard! Only I haven't figured out how to make different arrows yet, but I was able to reassign the buttons to JoyKey Devices A btn.1 Port 1-1 A Button btn 2 Port 1-2 B Button and Devices B btn.1 Port 2-1 A Button btn 2 Port 2-2 B Button, all this gives me in one of the fighting games, different buttons for two players, on one PC keyboard! But, so far I can't reassign the arrows, no matter how hard I tried, and whatever I downloaded, even a certain Joyport U-kun, it is available, by the way, in the latest version (As of today, October 2023, when I am typing this text) of the xm6_typeg_336_20230523 emulator, but there for some reason the usual 00 JoyKeys are not available, it says not found, or the required resources are not available (for some reason) (A required resource was unavailable), although I downloaded a lot, a lot from different sites, even from Japanese ones, from Japanese developers, and from the authors themselves - just a lot of time...
For now, here are 2 questions for experts - people who know, have information, knowledge and experience in this, I'm getting used to the rest, I'm still delving into it on my own, and gradually, step by step, slowly I'm studying everything, experimenting, something new for myself I open it, it opens, and something happens in the knowledge of these old Japanese Sharp x68000 Computers...

P.S.:
I managed to launch games from the Japanese Computer Sharp x68000 on MAME right away...
There, different controls on one keyboard from a PC for two players are easily customizable, and everything else... Everything there is easy, simple, and not interesting...
I would like to fully understand the emulators:
xm6_typeg_332L30_20201231
xm6_typeg_336_20230523
XM6 Pro-68k

rdmark, 2023-10-24:
I'm sorry, but this is not a forum for questions about using X68k computers in general, but for the development and use of the SCSI emulation device PiSCSI. If you have specific issues when using **PiSCSI** with a X68k please don't hesitate to create a separate ticket, or start a Discussions thread.

closed 2023-10-24 by rdmark

IgorPetrovich, 2023-10-28:
After time i found solved on question 1, about how make HDD ( .HDS )
All sucesfully done by this method, which i was do, and typed full information, and guide \ tutorial for someone else who also will be need this ( In future maybe or right now )...:
What i was do step by step for make HDD ( .HDS )?!
1.: Open Emulator XM6 Pro-68k or xm6_typeg_336_20230523
2.: Chose in options - Tools - Create - Fixed Disc SCSI - HDD ( .HDS ) , with sizes for tests 100 \ 1000 Mb ( In Fact 99 \ 999 Mb)
Choose Sizes ( Capacity )
Other Options:
Options Must Be On "Attached to interface." And "Replace First Drive."
Press "OK"
Checking What In Options Of Emulator, In Option SCSI - SCSI Interface - Internal - SCSI Discs ( via SCSI Interface ) - Drives ( 1 ) - ID ( 0 ) - SCSI-HD Image File ( .HDS )
3.: After created HDD ( .HDS ), i opened Emulator XM6 Pro-68k \ xm6_typeg_336_20230523 , Options - Floppy Drive #0 - MasterDisk_V3
4.: In DOS , in first i typed Command "FORMAT"
5.: Chosed "SCSI Devices" - Enter - "Hard Disc" - Enter - "Initalise" - Enter ( Y ) - "Create" - Enter - "Size"- Enter - Chosed Sizes ( 99 Mb ), by Typed this in window ( In Down Of The Screen ) - Enter -"Start" - Enter - "Create new partition" - "Y" , and got this results in end:
https://i.ibb.co/M2pdx89/44444444.png
6.: "End" - Enter.
7.: Do you want to reset? Most changes to the Partitions need it... Chosed - "Reset" ( By Press "Y" )
Done!
HDD Disc, with format .HDS - already created, formatted, can be readable now, now must be visible( In Programs \ Utilities \ In Emulators ), and ready to work ( For example in Disc Explorer ) .
P.S.:
I really like to write a lot and in detail, describing each point in detail... :)