PiSCSI/piscsi #1270: Web UI: File uploads in progress stored in tmp dir

rdmark, 2023-10-25:
This is to store file uploads in progress into a temporary directory instead of directly into the target images dir; then move them to the image dir once the upload is successful.

uweseimet, 2023-10-25:
The temporary directory should be /tmp. A different approach (I think Firefox does it this way) is to store the file with a unique temporary name in the final target directory, and to rename it to its actual name after a successful upload. This may be better, because it is guaranteed that both the temporary and the final file are in the same filesystem.
Especially when the temporary file is in a different folder than the final file you have to check for errors/completeness twice. Once for the uploaded file, and a second time after moving the file to the final location. Having both in the same folder makes a failure (provided the upload itself was successful) when moving/renaming less likely, because you cannot run out of disk space. If you run out of disk space and the files are in different folders two files may have to be deleted in order to clean up.

rdmark, 2023-10-27:
There are pros and cons to both approaches. The web UI doesn't have logic presently to automatically clean up failed or aborted file uploads. 

Putting temporary files in the present dir with a temporary name, makes it easier for users to identify failed uploads and clean them up themselves. But manual intervention by the user will be needed.

By using the /tmp dir the system will eventually clean up the failed uploads automatically and no manual intervention needed. However, the latter option isn't ideal since large temp files may clog up the file system in the short term.

I will attempt the former approach.

closed 2023-10-31 by rdmark