PiSCSI/piscsi #1279: Run Sonar scanning upon changes to the Python codebase

rdmark, 2023-11-01:
Sonar scanning is currently set up to run only in cpp.yml

https://github.com/PiSCSI/piscsi/blob/develop/.github/workflows/cpp.yml

However, when triggered the actual scan does cover the entire codebase, and reports valuable improvements to the Python and html/css code. It would be helpful to have the scanning run upon changes to `python/` so that we can get immediate feedback on improvements or regressions to the code quality.

rdmark, 2023-11-01:
https://github.com/PiSCSI/piscsi/pull/1280

closed 2023-11-01 by rdmark

uweseimet, 2023-11-01:
@rdmark WIth the changes in the current develop branch merged into my issues_1263_1278 branch the C++ tests are not run anymore when changing the C++ code. You can see this in the current workflow runs for "Build ARM binaries #4918": There is no corresponding  C++ tests run anymore.

rdmark, 2023-11-02:
@uweseimet I understand your use case now. You want the workflows to run for all subsequent commits pushed to an existing PR? 

The reason I changed that setting, was because duplicate workflow jobs were always initiated on every PR, which seemed wasteful and noisy.

Before I file a PR to revert to the previous behavior, I wanted to run two ideas by you:

For one, you can always initiate the workflow runs on demand. Then your workflow could be to push a number of commits, and then initiate the runs when you think enough changes are in to warrant a round of static analysis.
[image]

Secondly, Sonar provides an IDE plugin that provides a similar level of static analysis on code in real time on your local machine. This way you get faster feedback, while putting less load on Github and Sonar's servers.

https://www.sonarsource.com/products/sonarlint/

uweseimet, 2023-11-02:
@rdmark Before your changes for each commit (even if there was no PR yet for the effected branch) a workflow was run. Yes, in case of a PR there was a duplicate run, but this is better than not having any run until there is at least a draft PR.
This was discussed with @nucleogenic some time ago, when he was changing workflows, and we agreed on living with two runs.
Regarding the plugin, I do use it, but it does not really work and reports false positives or results different than what SonarCloud says. This is why I added on not on just that to the Wiki long ago. This means that running the analysis is the only way for really knowing what's going on.
I would appreciate the previous behavior to be restored.