PiSCSI/piscsi #1282: Remove fullspec/standard sub-folder

uweseimet, 2023-11-01:
Does it make sense to have a fullspec or standard sub-folder for the object files and binaries? Usually (and this might even be required) you run would run "make clean" before re-building for a different board type anyway. And usually your Pi is either connected to the one or the other board type, i.e. it is unlikely that you need both binaries anyway.
Unless I am missing something this sub-folder should be removed, so that 'bin' and 'obj' are located in 'cpp'.

rdmark, 2023-11-02:
There is one Standard board design that I know of, the "Piscsi Drive" made by Levo_man. Here is his website: https://levosretrocomputerprojects.co.uk/piscsi-drive

Where on the wiki does it say Standard hasn't been tested? It's probably good to update that, and link to this board design.

The only reason I can think of for keeping the binaries in separate dirs would be for a theoretical future `make dist` target for binary distribution packaging. But the build system would have to be modified at that point anyways. I don't have a strong opinion either way with this subdirs.

uweseimet, 2023-11-02:
I was referring to the bottom part of https://github.com/PiSCSI/piscsi/wiki/Hardware-Versions, but I think I misunderstood it, because this is just about old hardware versions that might not work.

Regarding something like "make dist", one could simply run make twice, and a "make clean" might be required anyway after changing the connection type. This would have to be verified.
All in all this this means that there is currently no reason for the subdir, and you agree with this, am I right? In this case we should keep the ticket as it is, i.e. for removing the subdir only and not doing anything else.

rdmark, 2023-11-02:
Yes, I don't mind getting rid of the subdirs.

closed 2023-11-14 by uweseimet