PiSCSI/piscsi #1287: Remove delays that are not accounted for by the SCSI specification

uweseimet, 2023-11-02:
The existing code contains delays that are not covered by the SCSI specification. These delays shall be removed. Initially there shall be a configuration switch in config.h call NO_DELAY.

- For each byte during the handshakes there are settle bus delays. Thess delays are not covered by the SCSI specification. Comments say that the delays wait until the signal line stabilizes. This IMO does not make sense, because the delays are only applied before reading/writing data bytes and not for any other signal. Why would the data signals need a delay before they are stable, but the REQ, ACK and other signals would not?
- The SCSI controller class does not accept commands as fast as it could. Again, this is not covered by the SCSI specification. Why should the controller not process commands as fast as it could, i.e. as fast as the computers send them?

The issues_1263_1278_1283 branch currently does not contain these delays has successfully been tested against a second PiSCSI board, a hard drive, a card reader, a streamer (each time in initiator mode using scsidump) and an Atari TT in target mode. The speed gain in target mode is a bit less than 1% with a Pi 4.

uweseimet, 2023-11-09:
@benjamink This is the ticket for testing delays, e.g. for the daynaport emulation. If you would like to help, please check out the testing_issue_1287 branch. In the current code of this branch there is no delay. You already tested this with your SE and it failed. I would be interested in the log with trace:6. You already know the procedure ;-).

uweseimet, 2023-11-09:
@benjamink As soon as you have confirmed that the code in this ticket does *not* work (because it contains the offending delay change) I will revert this change (change 1. in the list above) and will apply change 2 instead. Testing should be done with your SE for now.

benjamink, 2023-11-09:
First test fails (as expected).  `piscsi` crashes & the Mac shows the following on the screen: 

[image]

I will be labeling any log files with the commit hash being tested from now on:

[piscsi-issue1287-branch-c61b5c003c84f0778a8d6fb4f8d9f5d8e5e67321.log](https://github.com/PiSCSI/piscsi/files/13308555/piscsi-issue1287-branch-c61b5c003c84f0778a8d6fb4f8d9f5d8e5e67321.log)

uweseimet, 2023-11-09:
@benjamink No screenshots, please. They make this ticket harder to read and usually do not provide any important information for this ticket.
From the log you added I do not see that piscsi was crashing. Did the console report something like "Segmentation fault"?

uweseimet, 2023-11-09:
@benjamink Regardless of the answer to my previous question (we definitely need to know whether there was an actual crash and piscsi *terminated with an error message*) I have already committed the next change set: I have reverted the delay change. We need to make this test in order to ensure that we have a clean starting point for any further change.

benjamink, 2023-11-09:
@uweseimet piscsi didn't error out, it just froze.  Want me to run with tracing or debug on for the HDD device as well & see what happens?  I only had tracing on for the Daynaport device.

benjamink, 2023-11-09:
@uweseimet I might have some contention going on here with another device.  Let me remove everything else & try again.  I'll get back to you.

uweseimet, 2023-11-09:
@benjamink OK. Please note that the *current* branch for this ticket is identical with what you successfully tested on your SE yesterday. So this should definitely be working.

benjamink, 2023-11-09:
Confirmed `a4b523b9163689ed3f018579b04c78143b1653d6` works

uweseimet, 2023-11-09:
@benjamink OK, so we have a clean starting point. I will need some time to prepare the next change set, because I will first test it with my Atari. It would not make sense if you were testing something that does not even work with my hardware ;-).

uweseimet, 2023-11-09:
@benjamink The next change set is available, tested with my Atari. Before you start testing ensure that you have a backup of the disk image file(s) you attach. The changes affect everything, not just the daynaport. If possibe, do not use the daynaport emulation at all but just the hard drive emulation.
In case of an issue please attach a log (no screenshot, but just a note what happens, e.g. Mac freezes or Mac reports an error). Limit logging to your hard drive, e.g. trace:0, depending on the SCSI ID you use for the drive. And test with your SE only.
My Atari is in a different room, by the way, but I don't need to climb any stairs to get there ;-).

benjamink, 2023-11-09:
Commit `0db39ea19a0a96b60f3d1669946e63288a7da9f2` worked fine.  I ran with this command so it was only the disk (`$HASH` is just the current git hash - this command is in a small script I wrote): 

```
sudo "bin/fullspec-1287-debug/piscsi-${HASH}" -L trace:1 -id1 /home/pi/images/test-HD0-BK_SYS_7.1.hda
```

uweseimet, 2023-11-09:
@benjamink Sounds good! What you have just tested is item 2. in my list. I just updated the change set to also include item 3. Please re-test. Running "make" without "make clean" after updating is fine for this ticket.

benjamink, 2023-11-09:
Commit `70b6e69e52379a8d69e169f42cdf3937864d5c54` worked fine (again with just disk attached)

uweseimet, 2023-11-09:
@benjamink Thank you. Can you please test the same binary also with the daynaport? Please confirm that you are using an optimized build, with "-O3" displayed when building.

benjamink, 2023-11-09:
Looks like they're optimized.  Example:

```
g++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c generated/piscsi_interface.pb.cpp -o obj/fullspec/piscsi_interface.pb.o
```

uweseimet, 2023-11-09:
Yes, they are optimized. I have already push a new change set for testing, but before you update please test your current binary also with the daynapor.t

benjamink, 2023-11-09:
Everything including networking worked fine with commit `70b6e69e52379a8d69e169f42cdf3937864d5c54`

uweseimet, 2023-11-09:
@benjamink OK, please proceed with the latest change set. This one I expect not to work with the daynaport, but we'll see.

benjamink, 2023-11-09:
Commit `50278522c1288431d6863384bfff974d0e512b23` 

- Works with disk only (no DP)
- Hangs during boot with DP attached - same as before, everything just freezes, no crash or error

uweseimet, 2023-11-09:
@benjamink Thanks, just as expected, but this needed verification. The next change set will take some time to prepare from a coding perspective, and I will also need to test it with my hardware.

benjamink, 2023-11-09:
Ok, no problem.  I can test anything you need today (I'm in US Eastern Time) but I'll be on holiday from tomorrow until Tuesday.

uweseimet, 2023-11-09:
@benjamink The next change set is available and I have tested it with my hardware. Please also test, disk and daynaport. Please also write data to the disk and verify that the operation was successful.

benjamink, 2023-11-09:
Commit `fc35587f0b4033d849cf25af80fece205b2ff6d9` worked both with & without Daynaport attached.  I was able to read/write a text file in both scenarios & browse the Internet when Daynaport was attached.

uweseimet, 2023-11-09:
@benjamink Very good! I just committed yet another change set, related to item 5. on the list. This change only affects the daynaport.

benjamink, 2023-11-09:
Commit `e829b2cd30a4e7916c5ca8c1b46f71f231b1c894` froze everything again including the Raspberry Pi (or at least its networking).  

[piscsi-issue1287-branch-e829b2cd30a4e7916c5ca8c1b46f71f231b1c894.log](https://github.com/PiSCSI/piscsi/files/13313070/piscsi-issue1287-branch-e829b2cd30a4e7916c5ca8c1b46f71f231b1c894.log)

benjamink, 2023-11-09:
These are the last logs I have then me rebooting the Raspberry Pi.  Looks like it took networking offline: 

```
Nov  9 04:36:04 piscsi-perf475 kernel: [20810.284850] piscsi_bridge: port 2(piscsi0) entered disabled state
Nov  9 04:36:55 piscsi-perf475 kernel: [20861.356532] device piscsi0 left promiscuous mode
Nov  9 04:36:55 piscsi-perf475 kernel: [20861.356613] piscsi_bridge: port 2(piscsi0) entered disabled state
Nov  9 04:37:33 piscsi-perf475 kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
Nov  9 04:37:33 piscsi-perf475 kernel: [    0.000000] Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubu
ntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
```

uweseimet, 2023-11-09:
@benjamink Can you please try again with the update I just pushed? No need for a log if the system freezes again.

benjamink, 2023-11-09:
Froze again - `07f9383bffd9206dd8748e007cbc7b90a1ddb465`

uweseimet, 2023-11-09:
@benjamink OK, this means that items 1. and 5. (both are related) need further investigation. But not now, and not before your vacation ;-). Thank you once again for spending your time for this analysis. Enjoy your vacation, see you next week!

benjamink, 2023-11-09:
@uweseimet Thank you!  Looking forward to you having everything 100% fixed on Tuesday 😄

uweseimet, 2023-11-14:
@benjamink I hope you enjoyed your vacation. I'm afraid not everything is fixed ;-). But there is an updated testing_issue_1287 branch at least. After testing, please just provide your verdict. No screenshots or logs, in order to keep this ticket clear. Thank you!

benjamink, 2023-11-14:
@uweseimet thanks!  I gave `2f4354095df3577618257dae0e00e48ef6575bed` a test.  It works fine with HDD only but it freezes on boot when I include Daynaport.  Same thing as before, just freezes, no crash or error.

uweseimet, 2023-11-14:
@benjamink Thanks. Any change with my latest commit?

uweseimet, 2023-11-14:
@benjamink This is excellent news! It means that we do not need the proprietary Pi hardware timer to get the work-around needed for the DaynaPort right.

benjamink, 2023-11-14:
Confirmed commit `7b8ae25005ce0f5ead725010c25163ad761d1916` in the `improve_daynaport_delay` branch works fine with Daynaport.

benjamink, 2023-11-14:
Commit `3bf0a06ead38bae9516a50cf4ba783bb1db48762` still works with Daynaport

uweseimet, 2023-11-14:
@benjamink Please note that further testing should happen again on the testing_issue_1287 branch. There is a new changeset, with the latest commits to the develop branch included. Note that the binaries are no longer in cpp/fullspec/bin but in cpp/bin. Nothing else should have changed regarding the daynaport.

benjamink, 2023-11-14:
Commit `22dfa0c1958734ed65516bb9418d7b319ebd7bc3` of the `testing_issue_1287` works fine with networking enabled.

uweseimet, 2023-11-15:
@benjamink There is a new branch scsi_compliant_handshake. This branch is identical with the current develop branch except for changes covering item 2. of my list, related to the SCSI handshake.
You have already tested these changes, but before creating a PR, please test scsi_compliant_handshake (both hard drive and daynaport), to ensure that everything works before a potential merge. Thank you!

benjamink, 2023-11-15:
Worked fine with networking enabled.

Branch: scsi_compliant_handshake - Hash: 08e0f447b4da516342f0cb1a9ba9fa7069d94688

uweseimet, 2023-11-16:
@benjamink There is a new branch testing_issue_1287_2, which is identical with the current develop branch except for one kind of change. When you have time, I would appreciate a test with hard drive and daynaport.

benjamink, 2023-11-16:
@uweseimet worked fine w/ Daynaport enabled

Branch: testing_issue_1287_2 - Hash: f02d0d95b52b255596a71ee27539458e4460f7ad

rdmark, 2023-11-20:
I recommend asking for testing on more platforms in Discord. We have users on f.e. DEC workstations, RISCOS, Apple II etc. Sampler devices may be impacted as well as they are notoriously quirky.

If the suspicious delays originate in the RaSCSI codebase, they may have been workarounds for NEC PC-98, X68k or MSX2 quirks.

Is there anything concrete that we gain in compatibility or performance by making these low-level changes? I'm thinking of the tradeoff between standards compliance and the risk of breaking something for the broader user base.

uweseimet, 2023-11-20:
@rdmark Please see my notes in the related ticket: Why are the data signal lines of the PI considered unstable, but all the other lines are considered stable? This IMO does not make any sense from a technical perspective, doesn't it?
This change just gives a1% performance improvement, and a very important benefit is that one can completely get rid of using the system timer. This makes porting PiSCSI to other hardware (Pi 5) much easier. And as far as I can tell this was also one of the obstacles of porting to the Banana Pi.

What I suggest is to introduce a switch (macro) to enable/disable this change. For the develop part the change should be *enabled* (so that anybody can easily test it and tests it by default), but by enabling the macro in config.h you can disable it.

uweseimet, 2023-11-20:
@rdmark I added the switch mentioned above to the scsi_compliant_handshake branch.

closed 2023-11-24 by uweseimet