PiSCSI/piscsi #1306: Dyanaport networking no longer working after latest build of 'develop'

benjamink, 2023-11-04:
# Info

- Which version of Pi are you using:

Raspberry Pi 4

- Which github revision of software:

`develop` branch (built 2023-11-04 @ ~12pm UTC)

- Which board version:

Current

- Which computer is the PiSCSI connected to:

Macintosh Performa 475 w/ System 7.5.3

# Describe the issue

Raspberry Pi connected via wired ethernet (`eth0`) but wireless interface (`wlan0`) is also active & gets an IP from DHCP.

PiSCSI is configured to bridge `eth0`:

```
root@piscsi-perf475:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master piscsi_bridge state UP group default qlen 1000
    link/ether dc:a6:32:1f:8c:8b brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether dc:a6:32:1f:8c:8c brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.11/16 brd 192.168.255.255 scope global dynamic noprefixroute wlan0
       valid_lft 6257sec preferred_lft 5357sec
    inet6 fe80::3bfe:ab:8ed7:3cbc/64 scope link
       valid_lft forever preferred_lft forever
4: piscsi_bridge: <BROADCAST,MULTICAST,ALLMULTI,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 7e:47:31:f0:27:b0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.14/16 brd 192.168.255.255 scope global dynamic piscsi_bridge
       valid_lft 6233sec preferred_lft 6233sec
    inet6 fe80::dea6:32ff:fe1f:8c8b/64 scope link
       valid_lft forever preferred_lft forever
6: piscsi0: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast master piscsi_bridge state DOWN group default qlen 1000
    link/ether 7e:47:31:f0:27:b0 brd ff:ff:ff:ff:ff:ff
```

Dyanaport networking was working on the Performa prior to rebuilding PiSCSI today but it hasn't been used for a couple months.  Only new change is the domain used for my LAN but this is provided by DHCP as expected.

I'll provide additional details including screenshots later when I'm able to get back on the Macintosh.

uweseimet, 2023-11-04:
@benjamink Can you please provide logs from v23.04.01 and the develop branch? Please restrict the logs to the SCSI ID of the daynaport by launching piscsi with "-L trace:SCSI_ID" and please use debug builds, not optimized builds.

Fixing daynaport issues can be quite an effort, so please ensure that this is definitely not a problem with your environment. Just replacing the respective piscsi binaries without changing anything else should be how you can switch between a working and non-working setup.

benjamink, 2023-11-04:
@uweseimet how can I build piscsi with debugging?  I assume you're asking for something different than what I get when I run option #1 from `easyinstall.sh`?

I don't have specific logs for each version.  I have a huge `piscsi.log` that goes back to August though so it includes logs from when things were working but not specific logs like you're asking for.  I'm going to revert to `v23.04.01` & see if I can get networking working again.  Possibly with a bare-bones disk image & then re-build with current `develop` to see if it breaks again.  If things are still broken when I go back to `v23.04.01` then we know it's my setup & I'll need to get that working first.

I'll capture logs from both versions when I have things all setup.

uweseimet, 2023-11-04:
@benjamink In the cpp folder run these commands to compile with debug information:
```
DEBUG=1
make clean; make piscsi
```
The resulting binary will be placed in cpp/fullspec/bin. The binaries built this way are the ones to use for any testing. Please run any tests on a regular Raspberry Pi OS, bullseye or bookworm. But I assume you are already doing this, because the develop branch does not compile anymore on buster.
Please also see the build notes on https://github.com/PiSCSI/piscsi/wiki/Setup-Instructions.

benjamink, 2023-11-05:
I was able to revert back to `v23.04.01` & things worked again.  I did have to force an old version of the `Werkzeug` library (due to the older version of Flask) & rebuild the web-ui but that should be unrelated to anything I think.  I did this by adding the following to `python/web/requirements.txt`, deleting the `venv` & running `sudo ./start.sh` to rebuild the `venv`: 

```
Werkzeug==2.3.7
```

Attaching screenshot & relevant logs/configs from test run: 

[image]

[20231105-issue-1306.tar.gz](https://github.com/PiSCSI/piscsi/files/13259915/20231105-issue-1306.tar.gz)

I'm re-compiling current `develop` now & will test that later when I have time.

uweseimet, 2023-11-05:
@benjamink OK. For testing make a backup of your 23.04.01 piscsi binary, and then replace it by the piscsi binary from the develop build. Do not change anything else (!!!) If the develop binary false, in order to double-check replace it by the 23.04.01 binary, and do not change anything else (!!!). If things start working again we can be sure that this is a regression issue with the binary.

benjamink, 2023-11-06:
I'm suspecting that my real problem is with my netatalk configuration.  When I originally setup this PiSCSI I was using `wlan0` but have since moved the PiSCSI externally & am using `eth0`.  However it appears that regardless of my configurations AFP shares are only working over `wlan0` still.  

I didn't note your comment about compiling with the `-g` option until just now.  I can do that & re-run my tests later if you'd like.  `DEBUG` should be enabled in the builds I used below, however.

Again, with `wlan0` enabled I have a networking config that looks like the following: 

```
$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master piscsi_bridge state UP group default qlen 1000
    link/ether dc:a6:32:1f:8c:8b brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether dc:a6:32:1f:8c:8c brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.11/16 brd 192.168.255.255 scope global dynamic noprefixroute wlan0
       valid_lft 7199sec preferred_lft 6299sec
    inet6 fe80::3bfe:ab:8ed7:3cbc/64 scope link
       valid_lft forever preferred_lft forever
4: piscsi_bridge: <BROADCAST,MULTICAST,ALLMULTI,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether dc:a6:32:1f:8c:8b brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.14/16 brd 192.168.255.255 scope global dynamic piscsi_bridge
       valid_lft 7022sec preferred_lft 7022sec
    inet6 fe80::dea6:32ff:fe1f:8c8b/64 scope link
       valid_lft forever preferred_lft forever
```

I took `wlan0` offline with this command & confirmed `wlan0` was marked `DOWN` afterwards: 

```
sudo ifconfig wlan0 down
```

I performed a number of different iterations of tests & attached the logs in a tarball. I ran my tests by running the `piscsi` command manually with the following command line (changed as appropriate for each iteration - trace device, disk image & daynaport all remaining the same): 

```
sudo fullspec-v23.04.01-debug/piscsi -L trace:6 -id1 /home/pi/images/HD0-Performa-7.5.3.hda -id6 daynaport| tee piscsi-v23.04.01-fixatalkd-wlan0up.log
```

`scsiutil` shows the following output: 

```
$ scsictl -i 6 -c show
  6:0  SCDP  Dayna:SCSI/Link:1.4a  inet=10.10.20.1/24:interface=eth0,wlan0
```

## atalkd mis-configured for 'wlan0'

`atalkd.conf`:

```
wlan0 -phase 2 -net 0-65534 -addr 65280.76
```

- `piscsi-v23.04.01-debug.log` 
  - Works as expected but the AppleShare mount is using the wlan0 IP instead of the eth0 IP 
  - Both `piscsi-perf475` & `piscsi-macse` shares are visible in Chooser
- `piscsi-v23.10.01-debug.log` 
  - Stored AppleShare mount hangs at boot & times out 
  - nothing visible in Chooser 

## atalkd reconfigured to use `piscsi_bridge`

`atalkd.conf`:

```
piscsi_bridge -phase 2 -net 0-65534 -addr 65280.76
```

- `piscsi-v23.04.01-wlan0down.log`
  - `wlan0` forced down/offline
  - Stored AppleShare mount hangs at boot & times out
  - Only `piscsi-macse` is visible in Chooser
  - Attempting to connecto to `eth0` IP in Chooser times out/fails
- `piscsi-v23.04.01-fixatalkd-wlan0up.log`
  - `wlan0` online w/ expected IP address
  - Boots just fine w/ stored AppleShare mounted automatically on desktop
  - Both `piscsi-perf475` & `piscsi-macse` shares are visible in Chooser
  - Attempt to mount `PiSCSI Images` from `piscsi-perf475` share works but mount is using the `wlan0` IP address instead of `eth0`
  - Attempt to manually use `eth0` IP address in Chooser shows AppleShares but mounting a share still shows the share is using the `wlan0` IP address
- `piscsi-v23.10.01-fixatalkd-wlan0down.log`
  - `wlan0` forced down/offline
  - Stored AppleShare mount hangs at boot & times out
  - nothing visible in Chooser
  - Attempting to connecto to `eth0` IP in Chooser times out/fails
- `piscsi-v23.10.01-fixatalkd-wlan0up.log`
  - `wlan0` online w/ expected IP address
  - Stored AppleShare mount hangs at boot & times out
  - nothing visible in Chooser
  - Attempting to connecto to `eth0` IP in Chooser times out/fails

[piscsi-testing-logs-2023-11-06.tar.gz](https://github.com/PiSCSI/piscsi/files/13266956/piscsi-testing-logs-2023-11-06.tar.gz)

benjamink, 2023-11-06:
@uweseimet Understood.  I included all of the above moreso because I found it interesting the `v23.04.01` seems to tolerate whatever my netatalk config issues are just fine but `v23.10.01` does not.  I'm not sure why that makes a difference but there is definitely something different about the two versions related to this.

rdmark, 2023-11-06:
Benjamin, didn’t you say that web browsing doesn’t work either when we talked about this initially?

Let’s take netatalk out of the equation for now and reconfirm that DaynaPort…
- TCP/IP networking functions (can you talk to the internet?)
- AppleTalk networking functions (can your two Macs talk to each other)

For the latter, you can turn on personal file sharing on one of the Macs and have the other Mac connect to it, for instance.

For the former, try to just load a web page in a browser.

Let’s start with this to get a baseline assessment of your situation.

benjamink, 2023-11-06:
@rdmark I have access to the Internet when I run the `v23.04.01` build but not when I run the `v23.10.01` build (with nothing else changed).

Works (`v23.04.01`): 

```
sudo fullspec-v23.04.01-debug/piscsi -L trace:6 -id1 /home/pi/images/HD0-Performa-7.5.3.hda -id6 daynaport
```

Does not work (`v23.10.01`): 

```
sudo fullspec/piscsi -L trace:6 -id1 /home/pi/images/HD0-Performa-7.5.3.hda -id6 daynaport
```

All I've changed is the `piscsi` binary being run.  I was even able to switch from `v23.10.01` not working to `v23.04.01` working while the Mac was on & powered up.  Chooser instantly populated when I switched the `piscsi` binaries under the covers.

benjamink, 2023-11-06:
Internet continues to work even after doing the following: 

**Mac running the whole time booted into System 7.5.3**

1. Stop `picsci` daemon (`ctrl-c` from foreground process)
2. Take down `wlan0` w/ `ifconfig wlan0 down`
3. Start `piscsi` daemon again w/ the same commands in the previous comment
4. Navigate to a random URL I haven't been to before in iCab

uweseimet, 2023-11-06:
@benjamink Please provide separate logs for each piscsi binary. It is important that the logs are as short as possible and that logging is only enable for ID 6. Your previous logs show that you tried to create two daynport devices:
```
Nov  5 14:26:15 piscsi-perf475 PISCSI[16835]: [2023-11-05 14:26:15.406] [error] Duplicate ID 6, unit 0
```
This cannot work.

benjamink, 2023-11-06:
@uweseimet I don't understand where that duplicate came from.  I'm not seeing that when I run the `v23.10.01` binary right now with this command: 

```
sudo fullspec/piscsi -L trace:6 -id1 /home/pi/images/HD0-Performa-7.5.3.hda -id6 daynaport | grep -i duplicate
```

I ran each binary command independently & sent the output to a separate log file each time.  I ran the `piscsi` binary with the `-L trace:6` (daynaport device) as you requested.  It spews A LOT of output constantly so I'm not sure how to limit it without editing the log file & I wasn't sure if you'd get everything you needed if I did that.

uweseimet, 2023-11-06:
@benjamink I suggest that you add the two logs to this ticket. I will have a look at them and then let's see what the next steps might be. Thank you for spending your time on this.

benjamink, 2023-11-06:
More info... I find this very odd.  When I run the `v23.04.01` binary & run `nmap` against what I believe is it's virtual IP(?) the `nmap` completes in about 10s: 

```
$ nmap -Pn 192.168.15.141
Starting Nmap 7.94 ( https://nmap.org ) at 2023-11-06 18:30 EST
Nmap scan report for 192.168.15.141
Host is up (0.036s latency).
All 1000 scanned ports on 192.168.15.141 are in ignored states.
Not shown: 1000 closed tcp ports (conn-refused)

Nmap done: 1 IP address (1 host up) scanned in 10.05 seconds
```

If I run the `v23.10.01` binary with the same command & run `nmap` again it takes 85s: 

```
$ nmap -Pn 192.168.15.141
Starting Nmap 7.94 ( https://nmap.org ) at 2023-11-06 18:31 EST
Nmap scan report for 192.168.15.141
Host is up (0.000031s latency).
All 1000 scanned ports on 192.168.15.141 are in ignored states.
Not shown: 1000 filtered tcp ports (no-response)

Nmap done: 1 IP address (1 host up) scanned in 85.30 seconds
```

I am watching `tcpdump` when I run the above nmaps & confirm the nmap is runnag against piscsi: 

```
tcpdump -i piscsi0 tcp
```

I'm not sure if that suggests anything but the difference seems odd to me.

benjamink, 2023-11-06:
@uweseimet I'm happy to run whatever commands you'd like or reconfigure things however.  Whatever I can do to help determine if this is a problem with the new piscsi version of it is me.  I'm in no rush & I'll keep digging as well.

uweseimet, 2023-11-06:
I cannot really comment on this, but it might be helpful to have an overview on the initial setup created by both binaries. Can you please do this:

1. Ensure that piscsi is not automatically launched on your Pi, e.g. by sstemd
2. Reboot your Pi
3. Launch the 23.04.01 binary with iD 6 daynaport and trace enabled, but do not do anything else, do not use your Mac
4. Now run "ifconfig -a" to get the complete network information
5. Save the piscsi log and the ifconfig output

Now do the same steps *including the reboot* with the develop binary.

Please attach the resulting outputs to this ticket. This is just to ensure that when setting up the bridge both piscsi binaries do the same. The logs should reflect that they do the same, and the ifconfig output should also reflect this.

benjamink, 2023-11-07:
Disabled `piscsi` & `piscsi-web` services with:

```
systemctl disable piscsi
systemctl disable piscsi-web
```

Rebooted

**v23.04.01 tests**

Ran `ps ax` before running anything else:

[v23.04.01-ps-output.txt](https://github.com/PiSCSI/piscsi/files/13273704/v23.04.01-ps-output.txt)

Ran `ifconfig -a` before running anything else:

[v23.04.01-pre-ifconfig-a.txt](https://github.com/PiSCSI/piscsi/files/13273706/v23.04.01-pre-ifconfig-a.txt)

Started `piscsi` with this command (Mac is off): 

```
sudo fullspec-v23.04.01-debug/piscsi -L trace:6 -id1 /home/pi/images/HD0-Performa-7.5.3.hda -id6 daynaport | tee piscsi-v23.04.01.log
```

[piscsi-v23.04.01.log](https://github.com/PiSCSI/piscsi/files/13273716/piscsi-v23.04.01.log)

Ran `ifconfig -a` again from a different terminal while `piscsi` was running: 

[v23.04.01-post-ifconfig-a.txt](https://github.com/PiSCSI/piscsi/files/13273722/v23.04.01-post-ifconfig-a.txt)

Rebooted

**v23.10.01 tests**

Ran `ps ax` before running anything else:

[v23.10.01-ps-output.txt](https://github.com/PiSCSI/piscsi/files/13273730/v23.10.01-ps-output.txt)

Ran `ifconfig -a` before running anything else:

[v23.10.01-pre-ifconfig-a.txt](https://github.com/PiSCSI/piscsi/files/13273734/v23.10.01-pre-ifconfig-a.txt)

Started `piscsi` with this command (Mac is off): 

```
sudo fullspec/piscsi -L trace:6 -id1 /home/pi/images/HD0-Performa-7.5.3.hda -id6 daynaport | tee piscsi-v23.10.01.log
```

[piscsi-v23.10.01.log](https://github.com/PiSCSI/piscsi/files/13273736/piscsi-v23.10.01.log)

Ran `ifconfig -a` again from a different terminal while `piscsi` was running: 

[v23.10.01-post-ifconfig-a.txt](https://github.com/PiSCSI/piscsi/files/13273737/v23.10.01-post-ifconfig-a.txt)

uweseimet, 2023-11-07:
The pre-ifconfig logs show that the piscsi_bridge is already available after the reboot. Where does it come from? Please ensure that nothing is running that creates the bridge before you manually launch piscsi.

benjamink, 2023-11-07:
It's being setup by the file in `/etc/network/interfaces.d/piscsi_bridge` that the installer creates: 

```
pi@piscsi-perf475:/etc $ cat network/interfaces.d/piscsi_bridge
#
# Defines the 'piscsi_bridge' bridge that connects the PiSCSI network
# interface (ex DaynaPort SCSI/Link) to the outside world.
#
# Depending upon your system configuration, you may need to update this
# file to change 'eth0' to your Ethernet interface
#
# This file should be place in /etc/network/interfaces.d

auto piscsi_bridge
iface piscsi_bridge inet dhcp
	bridge_ports eth0
```

I'll disable that & re-run the tests.

uweseimet, 2023-11-07:
@rdmark I don't think anything else than piscsi should deal with the bridge, because piscsi sets up the bridge automatically, depending on the interface parameter settings.

benjamink, 2023-11-07:
I shouldn't be surprised but commenting out the `piscsi_bridge` file means `eth0` never gets an IP from DHCP (it's never brought up).

```
root@piscsi-perf475:/etc# ifconfig -a
eth0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:1f:8c:8b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 64  bytes 5144 (5.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 64  bytes 5144 (5.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.3.11  netmask 255.255.0.0  broadcast 192.168.255.255
        inet6 fe80::3bfe:ab:8ed7:3cbc  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:1f:8c:8c  txqueuelen 1000  (Ethernet)
        RX packets 1209  bytes 300942 (293.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 244  bytes 31841 (31.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

benjamink, 2023-11-07:
Same exact process as above but with the `piscsi_bridge` file commented out so it doesn't get setup.

**v23.04.01 tests***

[v23.04.01-ps-output.txt](https://github.com/PiSCSI/piscsi/files/13273855/v23.04.01-ps-output.txt)
[v23.04.01-pre-ifconfig-a.txt](https://github.com/PiSCSI/piscsi/files/13273857/v23.04.01-pre-ifconfig-a.txt)
[piscsi-v23.04.01.log](https://github.com/PiSCSI/piscsi/files/13273858/piscsi-v23.04.01.log)

**v23.10.01 tests***

[v23.10.01-ps-output.txt](https://github.com/PiSCSI/piscsi/files/13273896/v23.10.01-ps-output.txt)
[v23.10.01-pre-ifconfig-a.txt](https://github.com/PiSCSI/piscsi/files/13273923/v23.10.01-pre-ifconfig-a.txt)
[piscsi-v23.10.01.log](https://github.com/PiSCSI/piscsi/files/13273932/piscsi-v23.10.01.log)
[v23.10.01-post-ifconfig-a.txt](https://github.com/PiSCSI/piscsi/files/13273957/v23.10.01-post-ifconfig-a.txt)

benjamink, 2023-11-07:
Should I configure `eth0` to get an IP address as a normal interface but leave the `piscsi_bridge` script commented out so that only the `piscsi` binary creates the bridge?

rdmark, 2023-11-07:
What does your bridge configuration look like now?

Please see this wiki page and work your way backwards. https://github.com/PiSCSI/piscsi/wiki/Dayna-Port-SCSI-Link#manual-setup

https://github.com/PiSCSI/piscsi/wiki/Dayna-Port-SCSI-Link#manual-setupIn particular, the state of these two files:

/etc/network/interfaces.d/piscsi_bridge
/etc/dhcpcd.conf

rdmark, 2023-11-07:
I can’t test right now, but if I’m not mistaken piscsi_bridge is started by interfaces.d independently from piscsi… required since we configure `denyinterfaces` for the bridged interface.

benjamink, 2023-11-07:
@rdmark I went through that page & confirmed the original setup was correct.  I've also disabled `wlan0` completely by setting `dtoverlay=disable-wifi` in `/boot/config.txt`.  I re-enabled the bridge interface & restored things back to how the wiki describes.  All networking in`v23.04.01` works perfectly, `v23.10.01` does not at all.

`/etc/network/interfaces.d/piscsi_bridge`:

```
auto piscsi_bridge
iface piscsi_bridge inet dhcp
	bridge_ports eth0
```

`/etc/dhcpcd.conf` (at the bottom):

```
#interface eth0
#fallback static_eth0
denyinterfaces eth0
```

`/etc/sysctl.conf` still has forwarding commented out as expected: 

```
#net.ipv4.ip_forward=1
```

Rebooted & `ifconfig -a` looks like this (remember, `wlan0` is disabled in the `/boot/config.txt`): 

```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether dc:a6:32:1f:8c:8b  txqueuelen 1000  (Ethernet)
        RX packets 22833  bytes 1943809 (1.8 MiB)
        RX errors 0  dropped 34  overruns 0  frame 0
        TX packets 19917  bytes 6797434 (6.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 102  bytes 9161 (8.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 102  bytes 9161 (8.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::14a2:fcff:fecb:6bc  prefixlen 64  scopeid 0x20<link>
        ether 16:a2:fc:cb:06:bc  txqueuelen 1000  (Ethernet)
        RX packets 219  bytes 18295 (17.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2266  bytes 315931 (308.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi_bridge: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.3.14  netmask 255.255.0.0  broadcast 192.168.255.255
        inet6 fe80::dea6:32ff:fe1f:8c8b  prefixlen 64  scopeid 0x20<link>
        ether 16:a2:fc:cb:06:bc  txqueuelen 1000  (Ethernet)
        RX packets 22780  bytes 1610311 (1.5 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 19123  bytes 6751008 (6.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

At the top of the logs when running `piscsi` is this: 

```
[2023-11-07 01:16:51.879] [info] piscsi_bridge is already available
[2023-11-07 01:16:51.882] [info] Tap device piscsi0 created
```

Without the `/etc/network/interfaces.d/piscsi_bridge` in place `eth0` will not get an IP address & thus nothing can work.

uweseimet, 2023-11-07:
@benjamink In the last set of logs the 23.04.01 post-ifconfig is missing. Can you please add it? No need for the ps output, by the way.

uweseimet, 2023-11-07:
@benjamink Besides adding the missing logfile (see my previous comment), please also run this test:

- Reboot the Pi and ensure that the bridge is not set up (no need to add logfiles for this)
- Launch the develop piscsi manually and stop it again. The bridge should now be available, please verify with ifconfig. No need for attaching a log.
- Without a reboot now launch the 23.04.01 piscsi
- Check whether accessing the network with your Mac works. Does this work?

This way we have the develop piscsi set up the bridge and have the 23.04.01piscsi use it.

uweseimet, 2023-11-07:
@rdmark I just noticed that the template for new issues does not indicate to add the Raspberry Pi os version, e.g. bullseye or bookworm. Can you please add this?
@benjamink Are you using buster or bullseye?

uweseimet, 2023-11-07:
@benjamink In order to proceed faster I added a new branch "testing_daynaport". The code in this branch is identical to 22.04.01, but I will step by step apply the daynaport changes (there are hardly any) from the develop branch there. Sooner or later the piscsi binary built with this branch should fail, and then we know what's wrong
From now on please use testing_daynaport instead of 22.04.01 for testing. There is already a first set of changes available in "testing_daynaport". Can you please check out this branch and test it? I don't need any logs (yet) from this branch, just your verdict whether it works or not.

benjamink, 2023-11-07:
> @benjamink In the last set of logs the 23.04.01 post-ifconfig is missing. Can you please add it? No need for the ps output, by the way.

```
eth0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:1f:8c:8b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 64  bytes 5144 (5.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 64  bytes 5144 (5.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::c08e:6ff:fea9:8200  prefixlen 64  scopeid 0x20<link>
        ether c2:8e:06:a9:82:00  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi_bridge: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.10.20.1  netmask 255.255.255.0  broadcast 10.10.20.255
        inet6 fe80::88ca:92ff:fe56:91cc  prefixlen 64  scopeid 0x20<link>
        ether c2:8e:06:a9:82:00  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 37  bytes 6296 (6.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.3.11  netmask 255.255.0.0  broadcast 192.168.255.255
        inet6 fe80::3bfe:ab:8ed7:3cbc  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:1f:8c:8c  txqueuelen 1000  (Ethernet)
        RX packets 2504  bytes 546446 (533.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 642  bytes 100335 (97.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

uweseimet, 2023-11-07:
@benjamink Thank you for the missing log. @benjamink @rdmark Do we all agree that the bridge setup created by develop and 23.04.01 is identical?

benjamink, 2023-11-07:
> @benjamink Besides adding the missing logfile (see my previous comment), please also run this test:
> 
>     * Disconnect your Mac
> 
>     * Reboot the Pi and ensure that the bridge is not set up (no need to add logfiles for this)
> 
>     * Launch the develop piscsi manually and stop it again. The bridge should now be available, please verify with ifconfig. No need for attaching a log.
> 
>     * Without a reboot now launch the 23.04.01 piscsi
> 
>     * Please add logs of this piscsi startup sequence and with the ifconfig output after launching the 23.04.01 piscsi
> 
>     * Check whether accessing the network with your Mac works. Does this work?
> 
> 
> This way we have the develop piscsi set up the bridge and have the 23.04.01piscsi use it.

I've re-enabled `wlan0` so I can access the pi when `eth0` isn't getting a DHCP IP.

I've commented out the contents of the `/etc/network/interfaces.d/piscsi_bridge` & reverted `/etc/dhcpcd.conf` so it has this at the bottom (as it is originally): 

```
# fallback to static profile on eth0
interface eth0
fallback static_eth0
#denyinterfaces eth0
```

When I rebooted I confirmed the `piscsi_bridge` **IS NOT** created.  Then I started the `v23.10.01` binary.  This time when I look at `ifconfig -a` I noticed thet that `picsci_bridge` that gets created **DOES NOT** have a `10.10.20.1` IP address now.  I tried to be very diligent with organizing my output files above but this is different: 

```
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:1f:8c:8b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 63  bytes 5080 (4.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 63  bytes 5080 (4.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::b804:4cff:fe02:877d  prefixlen 64  scopeid 0x20<link>
        ether ba:04:4c:02:87:7d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi_bridge: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::b804:4cff:fe02:877d  prefixlen 64  scopeid 0x20<link>
        ether ba:04:4c:02:87:7d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 23  bytes 3438 (3.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4675<UP,BROADCAST,RUNNING,ALLMULTI,MULTICAST>  mtu 1500
        inet 192.168.3.11  netmask 255.255.0.0  broadcast 192.168.255.255
        inet6 fe80::3bfe:ab:8ed7:3cbc  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:1f:8c:8c  txqueuelen 1000  (Ethernet)
        RX packets 863  bytes 182358 (178.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 262  bytes 41508 (40.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

I rebooted again & this time started the `v23.04.01` binary.  Now the `piscsi_bridge` is setup **WITH** the `10.10.20.1` IP again: 

```
piscsi_bridge: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.10.20.1  netmask 255.255.255.0  broadcast 10.10.20.255
        inet6 fe80::b46b:8fff:feed:a9d8  prefixlen 64  scopeid 0x20<link>
        ether ca:ce:58:ec:3e:8c  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 27  bytes 4404 (4.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Rebooted again & ran the `v23.10.01` binary & confirmed again the missing `10.10.20.1` IP address: 

```
piscsi_bridge: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::9454:91ff:fef0:4e82  prefixlen 64  scopeid 0x20<link>
        ether 96:54:91:f0:4e:82  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 14  bytes 1772 (1.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

benjamink, 2023-11-07:
> @rdmark I just noticed that the template for new issues does not indicate to add the Raspberry Pi os version, e.g. bullseye or bookworm. Can you please add this? @benjamink Are you using buster or bullseye?

```
pi@piscsi-perf475:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

benjamink, 2023-11-07:
> @benjamink In order to proceed faster I added a new branch "testing_daynaport". The code in this branch is identical to 22.04.01, 

>There is already a first set of changes available in "testing_daynaport". Can you please check out this branch and test it? 

I'll build the binary from this & test again.

uweseimet, 2023-11-07:
@rdmark I have to admit I don't understand the network setup part. It's important to ensure that whatever setup there is, the result after launching piscsi (develop or 23.04.01) while there is no traffic (Mac disconnected) is exactly the same.
@benjamink Maybe you can add two logs each (develop and 22.04.01) with all your dhscp stuff enabled, of launching piscsi after a reboot and (with the Mac disconnected) and with the ifconfig output after lauching piscsi.
We definitely have to ensure that that as long as the Mac is disconnected everything is identical. Otherwise we will waste a lot of time.

benjamink, 2023-11-07:
I just did my last test with the `testing_daynaport` branch build & it does setup the `10.10.20.1` IP address on the `piscsi_bridge` interface when it starts.

uweseimet, 2023-11-07:
@benjamin Please note that testing_daynaport is not meant to test the setup, but to test everything else. But we first definitely need to know that 23.04.1 and develop result in the same setup after you have enable all your dhcp setup scripts etc. This is why I was asking for the two sets of logfiles showing that. (Please see my previous comment.)
As soon as we know that the final setup is always the same we do not have to consider the setup being broken in develop.

benjamink, 2023-11-07:
> @benjamin Please note that testing_daynaport is not meant to test the setup, but to test everything else. But we first definitely need to know that 23.04.01 (or testing_daynaport) and develop result in the same setup after you have enabled all your dhcp setup scripts etc. This is why I was asking for the two sets of logfiles showing that. (Please see my previous comment.) As soon as we know that the final setup is always the same we do not have to consider the setup being broken in develop.

When you say "all of my dhcp stuff setup" I assume you mean re-enabling the `piscsi_bridge` configs & changes to `/etc/dhcpcd.conf` that the `easyinstall.sh` currently does?  With the `easyinstall.sh` changes I get an IP address (`192.168.3.14`) for my LAN on the `piscsi_bridge` interface that bridges `eth0`.

I'm re-enabling those changes now & will test all 3 binaries with the log & `ifconfig -a` outputs as I have time today.

uweseimet, 2023-11-07:
@benjamink Yes, this is what I meant. We have to ensure that with your regular setup what piscsi does when it is launched is still the same. No need for *this* test to also test testing_daynaport. Just test develop and 22.04.01.

benjamink, 2023-11-07:
`piscsi_bridge` network script, DHCP on `eth0`, Mac disconnected

**v23.04.01**

```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether dc:a6:32:1f:8c:8b  txqueuelen 1000  (Ethernet)
        RX packets 1330  bytes 206119 (201.2 KiB)
        RX errors 0  dropped 6  overruns 0  frame 0
        TX packets 373  bytes 219183 (214.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 64  bytes 5157 (5.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 64  bytes 5157 (5.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::a800:f3ff:fe40:6b86  prefixlen 64  scopeid 0x20<link>
        ether aa:00:f3:40:6b:86  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi_bridge: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.3.14  netmask 255.255.0.0  broadcast 192.168.255.255
        inet6 fe80::dea6:32ff:fe1f:8c8b  prefixlen 64  scopeid 0x20<link>
        ether aa:00:f3:40:6b:86  txqueuelen 1000  (Ethernet)
        RX packets 1324  bytes 187223 (182.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 285  bytes 213375 (208.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4675<UP,BROADCAST,RUNNING,ALLMULTI,MULTICAST>  mtu 1500
        inet 192.168.3.11  netmask 255.255.0.0  broadcast 192.168.255.255
        inet6 fe80::3bfe:ab:8ed7:3cbc  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:1f:8c:8c  txqueuelen 1000  (Ethernet)
        RX packets 409  bytes 86964 (84.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 79  bytes 8303 (8.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Reboot

**v23.10.01**

```
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether dc:a6:32:1f:8c:8b  txqueuelen 1000  (Ethernet)
        RX packets 746  bytes 129229 (126.2 KiB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 217  bytes 81354 (79.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 63  bytes 5093 (4.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 63  bytes 5093 (4.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::c4c1:26ff:feb4:808d  prefixlen 64  scopeid 0x20<link>
        ether c6:c1:26:b4:80:8d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

piscsi_bridge: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.3.14  netmask 255.255.0.0  broadcast 192.168.255.255
        inet6 fe80::dea6:32ff:fe1f:8c8b  prefixlen 64  scopeid 0x20<link>
        ether c6:c1:26:b4:80:8d  txqueuelen 1000  (Ethernet)
        RX packets 744  bytes 118693 (115.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 194  bytes 79836 (77.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4675<UP,BROADCAST,RUNNING,ALLMULTI,MULTICAST>  mtu 1500
        inet 192.168.3.11  netmask 255.255.0.0  broadcast 192.168.255.255
        inet6 fe80::3bfe:ab:8ed7:3cbc  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:1f:8c:8c  txqueuelen 1000  (Ethernet)
        RX packets 381  bytes 83046 (81.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 81  bytes 8304 (8.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Only difference is that the MAC addresses change for the `piscsi0` & `piscsi_bridge` interfaces (not sure if that's expected/normal).

uweseimet, 2023-11-07:
@benjamink I don't think the mac address change is relevant. Or is it @rdmark ?

rdmark, 2023-11-07:
> @benjamink I don't think the mac address change is relevant. Or is it @rdmark ?

I don't know enough about Linux networking to say for sure. It seems unusual, since a MAC address is theoretically tied to a physical interface? 

As a side note and reminded, the emulated daynaport interface has a hard coded MAC address. If you have multiple computers using piscsi daynaport on the same LAN subnet you may run into issues.

benjamink, 2023-11-07:
Scenario - I have an AppleShare that is set to auto-mount at boot.

- v23.04.01 - boots & mounts share without problems
- v23.10.01 - long delay during boot while trying to mount share - share never mounts but boot completes to desktop
- `testing_daynaport` branch build appears to mount the share during boot but actually crashes both the piscsi & the Mac (had to power cycle the Pi) when the desktop starts to show up.  Double arrows in the top left show it is attempting network communication when it crashes.

Final lines of log from `testing_daynaport` branch before crash: 

```
[2023-11-07 13:09:35.571] [trace] (ID 6) - Entering Data In phase
[2023-11-07 13:09:35.571] [trace] (ID 6) - Sending data, offset: 0, length: 6
[2023-11-07 13:09:35.571] [trace] (ID 6) - Moving to next phase: datain
[2023-11-07 13:09:35.571] [trace] (ID 6) - Status Phase, status is $00
[2023-11-07 13:09:35.571] [trace] (ID 6) - Sending data, offset: 0, length: 1
[2023-11-07 13:09:35.572] [trace] (ID 6) - Moving to next phase: status
[2023-11-07 13:09:35.572] [trace] (ID 6) - Message In phase
[2023-11-07 13:09:35.572] [trace] (ID 6) - Sending data, offset: 0, length: 1
[2023-11-07 13:09:35.572] [trace] (ID 6) - Moving to next phase: msgin
[2023-11-07 13:09:35.572] [trace] (ID 6) - Bus free phase
[2023-11-07 13:09:35.572] [trace] virtual bool GPIOBUS::PollSelectEvent()
[2023-11-07 13:09:35.572] [trace] static bool SBC_Version::IsBananaPi()
[2023-11-07 13:09:35.577] [trace] static bool SBC_Version::IsBananaPi()
[2023-11-07 13:09:35.577] [trace] (ID 6) - ++++ Starting processing for initiator ID 7
[2023-11-07 13:09:35.577] [trace] (ID 6) - Selection phase
[2023-11-07 13:09:35.577] [trace] (ID 6) - Command phase
[2023-11-07 13:09:35.577] [debug] (ID 6) - Controller is executing Read6/GetMessage10, CDB $08000005f4c0
[2023-11-07 13:09:35.577] [debug] (ID:LUN 6:0) - Device is executing Read6/GetMessage10 ($08)
[2023-11-07 13:09:35.577] [trace] (ID:LUN 6:0) - READ(6) command, record: $00000005
[2023-11-07 13:09:35.577] [trace] (ID:LUN 6:0) - Read maximum length: 244
[2023-11-07 13:09:35.577] [trace] bool CTapDriver::HasPendingPackets() const 1 revents
[2023-11-07 13:09:35.577] [debug] int CTapDriver::Receive(uint8_t*) CRC is 5C8A4D62 - 62 4D 8A 5C

[2023-11-07 13:09:35.577] [trace] (ID:LUN 6:0) - Packet Size 64, read count: 1
[2023-11-07 13:09:35.577] [trace] bool CTapDriver::HasPendingPackets() const 1 revents
[2023-11-07 13:09:35.577] [trace] (ID:LUN 6:0) - Length is 70
[2023-11-07 13:09:35.577] [trace] (ID 6) - Entering Data In phase
[2023-11-07 13:09:35.577] [trace] (ID 6) - Sending data, offset: 0, length: 70
```

Logs:

[piscsi-testing-daynaport-branch.log](https://github.com/PiSCSI/piscsi/files/13280291/piscsi-testing-daynaport-branch.log)
[piscsi-v23.04.01.log](https://github.com/PiSCSI/piscsi/files/13280292/piscsi-v23.04.01.log)
[piscsi-v23.10.01.log](https://github.com/PiSCSI/piscsi/files/13280293/piscsi-v23.10.01.log)

uweseimet, 2023-11-07:
@rdmark The bridge is not a physical interface, this is why I think the address may change. At least there is nobody setting it, right?

benjamink, 2023-11-07:
MAC address is the same between `piscsi0` & `piscsi_bridge` for the `testing_daynaport` branch build as well (after starting piscsi).

uweseimet, 2023-11-07:
@benjamink Is the mac addres with 23.04 always the same after a reboot? What I mean is, if you reboot 2 or 3 times, will the address always be the same? Same question for develop. The address differs compared with 23.04, but is it always the same?

benjamink, 2023-11-07:
Checking.  When I re-ran v23.10.01 after stopping the `testing_daynaport` branch just now I ended up with matching MAC addresses still (they were different from the MAC from the `testing_daynaport` run).  Rebooting & trying a few more times.

benjamink, 2023-11-07:
Now they're consistently matching (`piscsi0` vs. `piscsi_bridge` MAC addresses) when running the v23.10.01 multiple times.  The MAC address for those virtual interfaces does change on each run though but they consistently match between themselves.

uweseimet, 2023-11-07:
testing_daynaport does exactly the same as 23.04.01 when creating the bridge. Whatever differences there are, they cannot be related to the piscsi code.
I will check now why testing_daynaport may have crashed. This is unexpected and unrelated to the problem in this ticket. I cannot test this myself, but only look at the code.

benjamink, 2023-11-07:
Ran v23.04.01 10x (still no reboot) & 100% of the time the MAC addresses matched

uweseimet, 2023-11-07:
Note that without rebooting the results are not comparable.

rdmark, 2023-11-07:
@uweseimet Regarding adding "Raspberry Pi OS version" the issue template, this template is managed in this file:

https://github.com/PiSCSI/piscsi/blob/develop/.github/ISSUE_TEMPLATE.md

If you put up a PR with the additions you would like to see, I can review & approve.

benjamink, 2023-11-07:
I am getting mis-matched MAC addresses on v23.04.01 after reboot but at a lower frequency (about 1 in 10 restarts)

uweseimet, 2023-11-07:
@benjamink I thought so. The address is not stable, but this is normal.

@rdmark Just create the PR. Please don't forget to approve #1286, as already discussed.

uweseimet, 2023-11-07:
@benjamink Please update your testing_daynaport branch and re-try.

uweseimet, 2023-11-07:
@benjamink There is also a new branch "testing_delay". It is identical with 22.04.01 except for a minor change. This change is completely unrelated to this ticket, but it would nevertheless be great if you could test it, this time with a lot of traffic. I would like to know if there are any issues, or if it works just like 22.04.01. Thanks a lot!

benjamink, 2023-11-07:
> @benjamink Please update your testing_daynaport branch, recompile with 'make piscsi' (no 'make clean' required) and re-try. If there is no crash, please provide the logfile after accessing the network with your Mac.

Build of `testing_daynaport` at commit `514294c3487491c9b612bacc7d8449866df13613` (from the time you commented) works.  

[piscsi-testing-daynaport-branch.log](https://github.com/PiSCSI/piscsi/files/13283978/piscsi-testing-daynaport-branch.log)

> @benjamink There is also a new branch "testing_delay". It is identical with 22.04.01 except for a minor change. This change is completely unrelated to this ticket, but it would nevertheless be great if you could test it, this time with a lot of traffic. I would like to know if there are any issues, or if it works just like 22.04.01. No need for any logs, regardless whether this branch works or not. I would just need your verdict. Thanks a lot!

Sure, I'll build & give it a test later today. 👍

uweseimet, 2023-11-07:
@benjamink Thank you! Regarding testing_daynaport we have a good point to start the step-by-step-change approach. I am going to get back to you when I have changes for this branch.

benjamink, 2023-11-07:
Only regrets are that my Performa is in the basement & my office is on the 2nd floor 😆

[32 more replies]

closed 2023-11-08 by uweseimet