PiSCSI/piscsi #1313: First boot fails. "Not running from a valid git repository." and "No module named 'bjoern'"

sqwirral, 2023-11-07:
# Info

- Which version of Pi are you using: 3 B+
- Which github revision of software: April 2023 release (
image_2023-04-25-PiSCSI-v23.04.01-arm64-lite)
- Which board version: 2.3B
- Which computer is the PiSCSI connected to: N/A

# Describe the issue

First boot fails, hanging on Web UI start.
[image]

Log:
`Nov 07 19:13:41 piscsi systemd[1]: Started PiSCSI-Web service.
Nov 07 19:13:49 piscsi PISCSIWEB[2240]: Warning: Not running from a valid git repository. Will not be able to update the code.
Nov 07 19:13:50 piscsi PISCSIWEB[2245]: compiling catalog src/translations/zh/LC_MESSAGES/messages.po to src/translations/zh/LC_MESSAGES/messages.mo
Nov 07 19:13:50 piscsi PISCSIWEB[2245]: compiling catalog src/translations/fr/LC_MESSAGES/messages.po to src/translations/fr/LC_MESSAGES/messages.mo
Nov 07 19:13:50 piscsi PISCSIWEB[2245]: compiling catalog src/translations/sv/LC_MESSAGES/messages.po to src/translations/sv/LC_MESSAGES/messages.mo
Nov 07 19:13:50 piscsi PISCSIWEB[2245]: compiling catalog src/translations/es/LC_MESSAGES/messages.po to src/translations/es/LC_MESSAGES/messages.mo
Nov 07 19:13:50 piscsi PISCSIWEB[2245]: compiling catalog src/translations/de/LC_MESSAGES/messages.po to src/translations/de/LC_MESSAGES/messages.mo
Nov 07 19:13:51 piscsi PISCSIWEB[2240]: Starting PiSCSI Web UI ...
Nov 07 19:13:51 piscsi PISCSIWEB[2247]: Traceback (most recent call last):
Nov 07 19:13:51 piscsi PISCSIWEB[2247]:   File "/home/pi/piscsi/python/web/src/web.py", line 12, in <module>
Nov 07 19:13:51 piscsi PISCSIWEB[2247]:     import bjoern
Nov 07 19:13:51 piscsi PISCSIWEB[2247]: ModuleNotFoundError: No module named 'bjoern'
Nov 07 19:13:51 piscsi systemd[1]: piscsi-web.service: Main process exited, code=exited, status=1/FAILURE
Nov 07 19:13:51 piscsi systemd[1]: piscsi-web.service: Failed with result 'exit-code'.
Nov 07 19:13:51 piscsi systemd[1]: piscsi-web.service: Consumed 10.326s CPU time.
Nov 07 19:13:51 piscsi systemd[1]: piscsi-web.service: Scheduled restart job, restart counter is at 156.
Nov 07 19:13:51 piscsi systemd[1]: Stopped PiSCSI-Web service.
Nov 07 19:13:51 piscsi systemd[1]: piscsi-web.service: Consumed 10.326s CPU time.
`

benjamink, 2023-11-08:
Add this to the bottom of the file `/home/pi/piscsi/python/web/requirements.txt`:

```
Werkzeug==2.3.7
```

Then restart `piscsi-web`.

If it still fails to start, remove the `venv` directory again & restart `piscsi-web`

```
sudo rm -rf /home/pi/piscsi/python/web/venv
sudo systemctl start piscsi-web
systemctl status piscsi-web
```

closed 2023-11-08 by rdmark

juchong, 2023-11-08:
@benjamink the old tags should be updated to build successfully. None of the precompiled images will work!

benjamink, 2023-11-08:
> @benjamink the old tags should be updated to build successfully. None of the precompiled images will work!

I agree - I think the current v23.04.01 should have the library pinned as a bug fix.  I'm not a maintainer though.  @rdmark  I'll let you reply more officially.

rdmark, 2023-11-09:
For starters, I added the following section to the top of the last release notes (April 2023):

---

## LIBRARY DEPENDENCY WORKAROUND
If the Web UI won't start up on your RPi, it is most likely because the Python library dependencies are broken. With the release of [Python Flask 3.0.0](https://pypi.org/project/Flask/3.0.0/) on Sep 30 2023, the PyPI dependency map for old versions of Flask broke down and `pip` ends up installing incompatible libraries.

The recommended workaround follows below. After booting your RPi, issue the following commands from the terminal command line:

```
echo "werkzeug==2.3.7" >> /home/pi/piscsi/python/web/requirements.txt
sudo rm -rf /home/pi/piscsi/python/web/venv
sudo systemctl restart piscsi-web
```

The Web UI should now start up, and load correctly after installing its dependencies with `pip`.

rdmark, 2023-11-09:
I went back and checked how far back our releases are impacted. This is the commit where the explicit `werkzeug` library version was removed from requirements.txt, which is from Aug 1 2021: https://github.com/PiSCSI/piscsi/commit/ea4486d1c099bdda42a687f97b8e7f0c766c5fbc

Looking at our releases, we have 11 of them. 10 are affected by this dependency issue. Only the very first May 2021 release has a fixed werkzeug library dependency (which was fine as long as the Flask 2.x release series was current in PyPI). Going back and retagging a reimaging 10 releases would be a significant effort, with many pitfalls.

We are planning to tag and release the next stable version on Nov 11 (Saturday this week) as long as no further blocker bugs are discovered last minute. I propose that we leave historical releases as-is and "roll forward" with a future-secured release.

As a side note: The Python dependencies may break again in the future. There are 22 Python library dependencies for the Web UI alone, and we only freeze 9 presently. Raised https://github.com/PiSCSI/piscsi/issues/1320 to get the entire dependency map frozen to (hopefully) avoid this situation in the future.

benjamink, 2023-11-09:
I think having the solution is sufficient but I was thinking it might be good to fix the current v23.04.01 version since it's still an active version.  That said, I'm not sure what all is involved in patching & rebuilding images, etc. so if just having the solution written is the plan I wouldn't be mad at that either. 

Bravo for pinning all packages going forward though.  When's the rewrite in Go so it's a single self-contained binary? 😀

rdmark, 2023-11-09:
Preparing those release images takes about 2h and is a semi-automated process. Let's hear what @akuker thinks! (When he's back from his biz trip.)

And yes, please by all means write a client in Golang or node.js -- save us all from this dependency hell ;)

sqwirral, 2023-11-09:
The notice on the release page is all you need, IMO, if the new version is coming out so soon. Thanks for all your hard work on this btw.

rdmark, 2023-11-09:
@sqwirral Thank you for the feedback. And sorry for the trouble!