PiSCSI/piscsi #1315: Aibom and Gamernium code got removed with the bananapi cleanup

rdmark, 2023-11-08:
The bananapi cleanup seems to have inadvertently removed some of the code that builds and manages connections with Aibom and Gamernium rascsi board types.

These are board types that has an install base and it's valuable for piscsi to support these, in my opinion.

The commit that removed code: https://github.com/PiSCSI/piscsi/commit/43088ab3bcfef07ea891437755a82a1e41c7a0e4

uweseimet, 2023-11-08:
I see. I thought this code was related to the Banana Pi. I am going to re-add it.

uweseimet, 2023-11-08:
@rdmark Looking closer at this code it actually IS related to the Banana Pi. It relies on files required for the Banana Pi:
```
#include "hal/pi_defs/bpi-m2p.h"
#include <string>
```
bpi-mp2.h says:
```
Copyright (c) 2014-2017 Banana Pi
```

rdmark, 2023-11-08:
@uweseimet That header is only needed for bananapi related symbols. So connection_gamernium.h and connection_aibom.h need to be restored, but with the bananapi lines removed. Like you did for connection_fullspec.h for instance.

uweseimet, 2023-11-08:
@rdmark OK, thank you for spotting this.

closed 2023-11-08 by uweseimet