PiSCSI/piscsi #1325: Create a new sub-folder for IDE setup/configuration files

uweseimet, 2023-11-09:
IDE-specific files, e.g. formatting rules, should be added to the repository. A new top-level folder "ide_setup" shall be used. The "cpp/.vscode" folder should also be moved there, in order not to force local preferences/setup onto other users with different needs.

uweseimet, 2023-11-09:
@rdmark I suggest to remove the recently created .clang-format file, because it cannot be used for IDE configurations and also does not cover all the settings required.

rdmark, 2023-11-13:
> @rdmark I suggest to remove the recently created .clang-format file, because it cannot be used for IDE configurations and also does not cover all the settings required.

Yes, please go ahead. Let's avoid multiple sets of files with overlapping purpose.

uweseimet, 2023-11-13:
Just removed it.

closed 2023-11-14 by uweseimet