PiSCSI/piscsi #1345: Evaluate implementation of the SCSI medium changer device type

uweseimet, 2023-11-12:
SCSI-2 defines a medium changer device type. This type of device manages several media and there are SCSI commands to switch between these media. This device type might be useful. Instead of ejecting/inserting removable media with scsictl, the web UI or the PiSCSI Control app, changing a medium would be triggered by a SCSI command. A tool running on the client computer could, for instance, offer a choice of all available media (image files) and the user would chose which of those to insert.
In case the medium changer device is not a candidate for such a feature, a custom device type might be an alternative.

rdmark, 2023-11-13:
This sounds vaguely like those CD changer drives that were popular in the 90s. I wonder if an actual SCSI CD changer product used this spec?

Does the spec allow for any drive type to be changed, or does it have to be an optical drive?

uweseimet, 2023-11-13:
No, this does not have to be an optical drive. I guess typical devices are tape drives with several tapes, where the tape is switched each night for the next backup cycle.
I'm not yet sufficiently familiar with the details in the spec, but in a piscsi context #1349 is probably more interesting, because it is more flexible and not only restricted to switching media.
In both cases a tool has to run on the computer in order to use the respective feature, of course. In the case of a media changer device should tools must have existed, maybe for some workstations.

uweseimet, 2023-11-14:
The approach of #1349 works, see the sample JSON input/output there. Because #1349 is more useful and flexible than #1345, this ticket can be discarded.

closed 2023-11-14 by uweseimet