PiSCSI/piscsi #1347: scsictl: Add a simple batch mode

uweseimet, 2023-11-13:
A simple batch mode shall be added to scsictl. The sequence of commands to be executed shall be read from a file. The filename shall be provided as a command line parameter.
When any command reports an error (in the return status of the protobuf interface call) batching shall stop.

rdmark, 2023-11-13:
Just out of curiosity, do you have particular use case in mind when built-in scsictl batching enables something that was impossible before?

If you have a shell environment you can script a batch of scsictl commands. And we also script batched calls to the protobuf interface.

Not to dispute that it could be a handy feature, but I wanted to check if this was the right thing to add complexity in scsictl for?

uweseimet, 2023-11-13:
The part of being able to execute commands that are not in plain text but in serialized format is still required. I am going to update this ticket accordlingly.

uweseimet, 2023-11-14:
Most likely not needed.

closed 2023-11-14 by uweseimet