PiSCSI/piscsi #1349: piscsi: Expose protobuf interface via SCSI

uweseimet, 2023-11-13:
The host services devices shall support a custom SCSI command for executing protobuf interface commands. These commands shall process serialized protobuf data (see #1348) and shall return serialized protobuf data.
With this feature SCSI clients can execute the same commands as supported by the regular protobuf interface. In the ideal case a client tool can generate the protobuf data on its own, e.g. because it can be linked against the protobuf library. In other cases the client can send predefined commands, which beforehand have been serialized by scsictl based on #1348. It might also be possible that clients create simple commands by proprietary means, i.e. without using a protobuf library. The format of the protobuf protocol data is documented in detail.

uweseimet, 2023-11-14:
This approach works in practice.

JSON sent via SCSI:
```
{
 "operation": "NETWORK_INTERFACES_INFO",
 "devices": [
  {
   "id": -1
  }
 ],
 "params": {
  "locale": "en_US.UTF-8"
 }
}
```
JSON result obtained via SCSI:
```
{"status":true,"networkInterfacesInfo":{"name":["eth0","wlan0"]}}
```

uweseimet, 2023-11-16:
@rdmark The SCHS device represents the piscsi process, i.e. there should not be more than one. This ticket enforces this. Would you agree that there can also only be a single SCBR and SCDP device?

rdmark, 2023-11-16:
Yes I think allowing only one network device to be attached at a time is appropriate. In fact, I think this is already the de-facto situation, because the code for initializing the piscsi_bridge will throw an error when you attempt to double initialize a bridge.

uweseimet, 2023-11-16:
@rdmark Yes, there already is an error. By enforcing only a single instance we get a better error handling, becuse the problem is detected before trying to initialize the network a second time.

A new localized error messages is needed for that. The "issue_1349" branch already contains the required changes (ERROR_UNIQUE_DEVICE_TYPE). If anybody wants to add the missing translations, please go ahead and just add them to this branch.

uweseimet, 2023-11-16:
Successfully tested with an Atari TT as client computer.

uweseimet, 2023-11-17:
> Yes I think allowing only one network device to be attached at a time is appropriate. In 

By the way, the DaynaPort emulation reserves 16 MB of buffer space per instance. It's the device that consumes the most memory by far. If you want to bring a small Pi to its knees, just create several of them ;-).

rdmark, 2023-11-20:
@uweseimet Regarding the intention of this feature: I understand the technical concept of being able to send protobuf commands from the host over the SCSI bus to the PiSCSI. 

If you can help me understand your vision for this feature, however: Would you then develop a native application on, say the vintage Atari or Mac computer that makes use of this new interface? Do you already have such an application for Atari that could demonstrate the end-user feature set that you have in mind?

I'm just trying to understand the justification and tradeoff for the adding of a large amount of code (and the scsiexec tool) to an already complex software package, that takes hours to compile on an RPi Zero, as an example.

So in the end, while I see the potential utility, I'm just not sure about the benefit to the broader PiSCSI user base right now!

uweseimet, 2023-11-20:
@rdmark For the Atari a simple tool already exists: https://github.com/PiSCSI/piscsi.

Regarding compile times, one should consider making compiling the initiator mode tools, i.e. scsidump and scsiexec, optional. They should be contained in the binary PiSCSI distribution (this one should be complete), but easyinstall might not compile them by default.

uweseimet, 2023-11-20:
@rdmark One more think about which I have already been thinking, regarding compile times. I don't know whether this would already work yet, but the code is getting more modular all the time, and sooner or later and might be possible to ask the user before compiling which devices she/he wans to use. Then you would only compile what you need.

closed 2023-11-24 by uweseimet