PiSCSI/piscsi #1351: Expose Webmin as a sub-path rather than custom port & link to it from piscsi web ui settings page

benjamink, 2023-11-13:
Feature idea - Instead of exposing Webmin via port 10000 (or whatever custom port the user chooses) serve it from a path like `http://piscsi.domain/webmin`.  This should be able to be done with some simple Nginx configurations as part of the Webmin install.  

Additionally add a button to the PiSCSI "Settings" page to Webmin only if it is detected that the Webmin feature has been installed.

benjamink, 2023-11-13:
Perhaps just adding it to the list here with a link would be cleanest:

`<icon> Webmin configuration utility [enabled|disabled] <open_in_new_tab/window_icon>`

[image]

rdmark, 2023-11-14:
Sure, I guess it makes sense to list the Webmin service status in that list explicitly as well. 

This reminds me of that age-old request to allow users to actually install Netatalk/Samba/FTP/Webmin servers from within the Web UI. I wonder if this is worth revisiting now when we have this rudimentary UI for listing up the available services. Something like an "Install" button do each, and logic for testing whether the software is installed or not, and just not not running...?

benjamink, 2023-11-14:
It's also possible to only list it if it's been installed via `easyinstall` & otherwise just not have it show here at all.  

If it's installed then you can show it in the list with an indicator of whether it's currently running or not & a simple icon/link to the Webmin UI (maybe open in a new tab/window?).

rdmark, 2024-01-30:
@benjamink Are you still planning to try to get this working, or can it be discarded?

benjamink, 2024-02-01:
Nope, we can close this.

closed 2024-02-01 by benjamink