PiSCSI/piscsi #1358: PiSCSI 'develop' branch failing to build in Docker environment

benjamink, 2023-11-14:
Currently the `develop` branch is failing to build in the Docker Compse environment: 

```
41.02 Compiling FULLSPEC with clang++ on 4 simultaneous cores...
41.11 rm -rf obj bin generated coverage piscsi.dat
41.20 -- Generating protobuf-based source files
41.20 -- Creating directory obj
41.20 mkdir -p generated
41.20 mkdir -p obj
41.20 -- Creating directory bin
41.21 protoc --cpp_out=generated piscsi_interface.proto
41.21 mkdir -p bin
41.21 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./piscsi/piscsi_image.cpp -o obj/piscsi_image.o
41.21 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./piscsi/localizer.cpp -o obj/localizer.o
41.22 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./piscsi/command_context.cpp -o obj/command_context.o
41.24 mv generated/piscsi_interface.pb.cc generated/piscsi_interface.pb.cpp
41.24 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./piscsi/piscsi_executor.cpp -o obj/piscsi_executor.o
50.46 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./piscsi/piscsi_service.cpp -o obj/piscsi_service.o
59.15 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./piscsi/piscsi_response.cpp -o obj/piscsi_response.o
61.70 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./piscsi/piscsi_core.cpp -o obj/piscsi_core.o
62.84 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./controllers/phase_handler.cpp -o obj/phase_handler.o
63.38 In file included from ./piscsi/piscsi_response.cpp:16:
63.38 ./piscsi/piscsi_response.h:28:2: error: explicitly defaulted default constructor is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
63.38         PiscsiResponse() = default;
63.38         ^
63.38 ./piscsi/piscsi_response.h:50:44: note: default constructor of 'PiscsiResponse' is implicitly deleted because field 'device_factory' of const-qualified type 'const DeviceFactory' would not be initialized
63.38         [[no_unique_address]] const DeviceFactory device_factory;
63.38                                                   ^
65.24 clang++ -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++20 -iquote . -D_FILE_OFFSET_BITS=64 -DFMT_HEADER_ONLY -DSPDLOG_FMT_EXTERNAL -MD -MP  -DCONNECT_TYPE_FULLSPEC -c ./controllers/controller_manager.cpp -o obj/controller_manager.o
66.83 In file included from ./piscsi/piscsi_core.cpp:24:
66.84 In file included from ./piscsi/piscsi_core.h:16:
66.84 ./piscsi/piscsi_response.h:28:2: error: explicitly defaulted default constructor is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
66.84         PiscsiResponse() = default;
66.84         ^
66.84 ./piscsi/piscsi_response.h:50:44: note: default constructor of 'PiscsiResponse' is implicitly deleted because field 'device_factory' of const-qualified type 'const DeviceFactory' would not be initialized
66.84         [[no_unique_address]] const DeviceFactory device_factory;
66.84                                                   ^
66.87 In file included from ./piscsi/piscsi_core.cpp:24:
66.87 ./piscsi/piscsi_core.h:34:2: error: explicitly defaulted default constructor is implicitly deleted [-Werror,-Wdefaulted-function-deleted]
66.87         Piscsi() = default;
66.87         ^
66.87 ./piscsi/piscsi_core.h:70:39: note: default constructor of 'Piscsi' is implicitly deleted because field 'response' has a deleted default constructor
66.87         [[no_unique_address]] PiscsiResponse response;
66.87                                              ^
66.87 ./piscsi/piscsi_response.h:28:2: note: explicitly defaulted function was implicitly deleted here
66.87         PiscsiResponse() = default;
66.87         ^
66.87 ./piscsi/piscsi_response.h:50:44: note: default constructor of 'PiscsiResponse' is implicitly deleted because field 'device_factory' of const-qualified type 'const DeviceFactory' would not be initialized
66.87         [[no_unique_address]] const DeviceFactory device_factory;
66.87                                                   ^
68.32 1 error generated.
68.35 make: *** [Makefile:168: obj/piscsi_response.o] Error 1
68.35 make: *** Waiting for unfinished jobs....
71.11 2 errors generated.
71.13 make: *** [Makefile:168: obj/piscsi_core.o] Error 1
------
failed to solve: process "/bin/sh -c ./easyinstall.sh --run_choice=16 --cores=`nproc`" did not complete successfully: exit code: 2
```

benjamink, 2023-11-15:
@uweseimet does this look like something that might have been introduced recently?   I don't know enough about C++ to know how to fix this easily.

uweseimet, 2023-11-15:
@benjamink Yes, I also stumbled about this tonight. It depends on the compiler and the options being used. The cross compilers, for instance, do not report this. There is already a work around in the testing_issue_1287 branch. I think it's the last commit, but that's not yet part of develop.

benjamink, 2023-11-15:
Just added the fix to the Docker environment to work with that temporary work around.  Is this work around something that you expect to have merged in?

uweseimet, 2023-11-15:
Merged, but the next release most likely will not need this work-around. But having it now is the quickest solution.

closed 2023-11-15 by uweseimet