PiSCSI/piscsi #1365: Install vsftpd Webmin module

benjamink, 2023-11-15:
As per discussion in #1354 there is a Webmin module for `vsftpd` that should be installed to round out the ability to manage the various PiSCSI services.  It looks like it's a couple years old but maybe it's still worth having to round out Webmin support for all of the sharing services.

https://www.webmin.com/cgi-bin/search_third.cgi?modules=1

[image]

uweseimet, 2023-11-15:
@benjamink Be careful with installing old software. It might not be maintained anymore, or with the next Pi OS upgrade the web UI does not work anymore.

rdmark, 2023-11-16:
@uweseimet This is a plugin for Webmin. While we provide a link to Webmin (when installed) from the PiSCSI Web UI, it does not impact the functionality of PiSCSI or its Web UI directly. Additionally, the vsftp Webmin plugin specifically looks for the existence of the vsftpd.conf file and then provides a GUI for modifying it. It does not interfere with any other FTP server process.

I agree that it's slightly risky to install an unmaintained piece of software. In the worst case we can fork it and update it when something breaks. :)

rdmark, 2023-11-17:
Sooo, I tested the vsftpd module a little bit. It’s pretty buggy after just trying two different workflows. It’s not properly updating the conf file, but duplicates multiple sections and breaks the file. We need to fix the bugs before using this.

rdmark, 2023-11-18:
I prepped a fork of the vsftpd webmin module here: https://github.com/rdmark/vsftpd-webmin

Let me see if I can fix the bugs and roll a new tarball for the module.

rdmark, 2023-11-18:
Actually, after a bit more googling I found yet another vsftpd webmin module, last updated in 2009. https://sourceforge.net/projects/vsftpdwebmin/

It is a LOT better structured and has a tighter feature set, but uses some outdated syntax. (The other one tries to do fancy stuff with mysql and ssl certs, all quite half-baked.)

rdmark, 2023-11-18:
@benjamink Would you be able to test the latest `main` branch for my module fork? https://github.com/rdmark/vsftpd-webmin

The way to install the module goes something like:

```
cd vsftpd-webmin
./make_tar.sh vsftpd
sudo /usr/share/webmin/install-module.pl ~/vsftpd-[date and time].wbm.gz
```

I'm experiencing one weird bug where it's impossible to save the module configuration. Once you hit the Save button it just stalls there without saving the changes to /etc/webmin/vsftpd/config or transitioning back to the index page.

On the file system you can observe that /etc/webmin/vsftpd/config.lock gets created, but not deleted (which it should once the edit has been made.)

It's odd because this part of the logic and UI are all in Webmin proper (`config.cgi`) and not at all in the module code. So I'm starting to wonder if there's something environmental in my Ubuntu VM...

benjamink, 2023-11-18:
I will when I figure out a bug I think I introduced with moving webmin to a
sub-path location under Nginx.  Now authentication to Webmin isn't working
for some reason.

benjamink, 2023-11-18:
@rdmark I reverted back to the v23.11.01 tag & re-build both the `piscsi-web` & re-installed `webmin` from `easyinstall.sh`.  Then I installed your `vsftpd-webmin` module as per your instructions.  It seems to work fine for me.

I went into the `vsftpd` module & selected the profile "Standalone virtual users" & saved.  Then I enabled `xfer logs` & saved.  It saved the config change & also saved backups of the `/etc/vsftpd.conf` file.  Resulting `/etc/webmin/vsftpd` looks like this: 

```
pi@piscsi-perf475:~ $ sudo ls -la /etc/webmin/vsftpd/
total 28
drwx--x--x   2 root bin  4096 Nov 18 15:27 .
drwxr-xr-x 114 root root 4096 Nov 18 15:23 ..
-rw-------   1 root bin   246 Nov 18 15:23 config
-rw-r--r--   1 root root  319 May  7  2014 pam_d_backup_webmin
-rw-r--r--   1 root root  127 Nov 18 15:26 vsftpd.conf.webmin_backup
-rw-r--r--   1 root root 5850 Mar  6  2019 vsftpd.conf.webmin_backup_perm
```

`/etc/vsftpd.conf` looks like this after above changes: 

```
pi@piscsi-perf475:~ $ cat /etc/vsftpd.conf
xferlog_enable=YES
chroot_local_user=YES
pasv_max_port=30999
pasv_min_port=30000
anonymous_enable=NO
listen=YES
local_enable=YES
guest_enable=YES
```

rdmark, 2023-11-18:
@benjamink Sorry I wasn't specific enough. I meant the module configurations that you access through the gear icon in the top left of the title bar.

[image]

benjamink, 2023-11-19:
@rdmark ah!  Yes, I do see that same behavior when clicking "Save" from the gear config page.  `config.lock` gets created & the Save button just has a neverending spinner.

rdmark, 2023-11-19:
Thanks for testing! So there must be some code somewhere that interferes with that CGI script. Weird...

rdmark, 2023-12-09:
I fixed the bug... See https://forum.virtualmin.com/t/endless-loop-in-foreign-require/123882

The module seems to work well now. The one final thing I want to do is to make sure all the configuration options are still accurate and up to date with the latest vsftpd.

rdmark, 2023-12-10:
@benjamink Several more bugs fixed! I would appreciate if you could test the module now. :)

https://github.com/rdmark/vsftpd-webmin/tree/main

Note that I simplified the make_tar.sh script. You just run it without parameters now.

benjamink, 2023-12-10:
@rdmark sure, I'll take a look at it today.

rdmark, 2023-12-10:
Cheers! This PR should do the trick with installing the vsftpd module via easyinstall: https://github.com/PiSCSI/piscsi/pull/1396

benjamink, 2023-12-15:
@rdmark sorry it took me so long, work has been nuts.  I finally tested & reviewed your vsftpd module install & it looks fine.

rdmark, 2023-12-15:
Cheer!

closed 2023-12-15 by rdmark