PiSCSI/piscsi #1376: Feature request: be able to set the TYPE/CREATOR resource fork attributes of file(s) inside newly-created cd-rom ISO images of HFS type

i-to-z, 2023-11-18:
# Info

- Which version of Pi are you using:
- Which github revision of software:
- Which board version:
- Which computer is the PiSCSI connected to:
- Which OS you are using (output of 'lsb_release -a'):

# Describe the issue

i-to-z, 2023-11-18:
Hello piscs contributors!
I forked the "develop" branch and implemented this feature request ( filed by me ):
See it at https://github.com/i-to-z/piscsi/tree/issue_1376
The two files that I touched were:
- easyinstall.sh
- python/web/src/web.py

The code changes are small and well described.

Please do a pull request and merge these changes into the "develop" branch.

rdmark, 2023-11-18:
Thanks for reporting the improvement idea and filing the PR for it! This would definitely make the ISO image generation a lot more useful for Mac users, even providing a new bootstrapping path since you can embed a self-extracting archive for installing stuffit or the likes.

Now, do you know the source, authorship and license of that file extension mapping file? As a pure text file, storing it in our own git repo, or pulling it in through a submodule would be ideal.

It's interesting to note that the file may be derivative (or the other way around) of the AppleVolumes.system config file that has been distributed with Netatalk since the mid-90s.

https://github.com/Netatalk/netatalk/blob/branch-netatalk-2-3/config/AppleVolumes.system

i-to-z, 2023-11-18:
Hi Daniel, I don't remember where I found [my mapping file](https://drive.google.com/uc?export=download&id=1fUHyY8puxw3cWa9mH17G-f5--JiqplAR), it's been a while.
Doing some google search did not help me.

But I compared my file with the one you found at https://github.com/Netatalk/netatalk/blob/branch-netatalk-2-3/config/AppleVolumes.system (which comes with a free GNU license).

The two files contain the same data, but some of the comments are different.
These comments are not used by the genisoimage library, so these diffs are irrelevant.

So perhaps we can add something like this to it: 
"This file was derived from the open source file at https://github.com/Netatalk/netatalk/blob/branch-netatalk-2-3/config/AppleVolumes.system and the file was formatted to work with genisoimage".

And yes, the file is so small, it should be stored in the github repository directly.
I can make this change if you give me a suitable folder name.

rdmark, 2023-11-18:
IANAL but I think a collection of configuration data lacks the creative element that makes them copyrightable in the first place. The code comments may be covered by copyright however.

One example of such a discussion in https://groups.drupal.org/node/17555

In addition to your idea, I suggest rewriting the code comments from scratch, and add a note that the format is based on standards defined in the genisoimage man page (including the * for defining a blanket fallback mapping, which is novel compared to the Netatalk file.)

How about we store the file in `python/web` next to all the other configuration files for 3rd party apps that we rely on?

i-to-z, 2023-11-19:
Daniel I made the changes as we discussed:
- Completely rewrote and reformatted file `genisoimage_hfs_resource_fork_map.txt` based on the mappings at  https://github.com/Netatalk/netatalk/blob/branch-netatalk-2-3/config/AppleVolumes.system.
- Stored the file `genisoimage_hfs_resource_fork_map.txt` in the project repository under python/web.
- Undid my changes in `easyinstall.sh` (that were downloading the file from the web), they are not needed.
- Updated `web.py` to find the file under `python/web`.
Take a look.

rdmark, 2023-11-19:
Looks good. Merged!

closed 2023-11-19 by rdmark