PiSCSI/piscsi #1380: Webmin login doesn't work after moving to /webmin sub-path

benjamink, 2023-11-18:
After the change in #1354 the Webmin authentication no longer works.  Any attempts to login with the `pi` user (or any other user) are just redirected back to the Webmin login page.  The `root` user login will work if you use the Webmin command to generate a local `root` user password: 

```
/usr/share/webmin/changepass.pl /etc/webmin root <password>
```

But this isn't the desired behavior.  Webmin should allow any users with `sudo` permissions to login as Webmin admins automatically (this is how it used to work).  Something about moving Webmin to the `/webmin` sub-path under Nginx is breaking things.

benjamink, 2023-11-18:
Adding some configuration guides for reference.  I've tried these changes & they still don't work: 

https://webmin.com/faq/#can-i-run-webmin-or-usermin-behind-reverse-proxy

https://superuser.com/a/1709050

rdmark, 2023-11-20:
@benjamink Have you tried asking for advice at the Virtualmin forums? https://forum.virtualmin.com/

rdmark, 2023-11-20:
The mother ticket is here: https://github.com/PiSCSI/piscsi/issues/1351

Closing since the regression bug has been fixed

closed 2023-11-20 by rdmark