PiSCSI/piscsi #1402: ModeSelect6 fails with DEC Alpha system

kkaempf, 2023-12-20:
# Info

- Which version of Pi are you using: 3
- Which github revision of software: HEAD (tag 23.11.01)
- Which board version: 2.3B
- Which computer is the PiSCSI connected to: DEC Personal Workstation 433au, running OpenVMS/Alpha 7.3
- Which OS you are using (output of 'lsb_release -a'): Debian 11 (bullseye)

# Describe the issue

Booting from piscsi fails in the SRM console (aka 'BIOS') with `failed to open`
Running with debug reveals
~~~
[2023-12-20 16:40:20.184] [debug] (ID:LUN 0:0) - Device is executing TestUnitReady ($00)
[2023-12-20 16:40:20.194] [debug] (ID:LUN 0:0) - Device is executing ModeSelect6 ($15)
[2023-12-20 16:40:20.194] [debug] (ID 0) - Error status: Sense Key $05, ASC $26
~~~

kkaempf, 2023-12-20:
Digging through the code (and adding more log messages) reveals that ModeSelect6 is called with a page code of 0 (== "supported diagnostic pages") but the code in `cpp/devices/scsi_command_util.cpp` only recognizes page 3.

rdmark, 2023-12-21:
Thanks for reporting, and tracking down the code that’s responsible.

The next step here would be to track down what the SCSI specification says about ModeSelect6 page code 0. 

We don’t have an active core developer in this project right now, so if you’re up for it we would very much welcome your contribution.

rdmark, 2023-12-21:
Additionally, it would be helpful if you could test with the previous stable release version, April 2023. There was a good amount of refactoring that happened, which also touched the ModeSelect code. 

https://github.com/PiSCSI/piscsi/releases/tag/v23.04.01

kkaempf, 2023-12-21:
v23.04.01 shows the same behaviour.

Digging deeper now and trying to come up with a PR 🤞🏻

kkaempf, 2023-12-25:
So turns out it was actually three bugs, one generic, two specific to DEC Alpha, in ModeSelect6 😆

rdmark, 2024-01-09:
https://github.com/PiSCSI/piscsi/pull/1406 has been merged. Thanks for the patches!

PR #1405 (2024-01-07): Multiple fixes for ModeSelect
This PR covers multiple issues

- The DEC Alpha SRM console (aka BIOS) calls ModeSelect6 with no payload

- OpenVMS Alpha calls ModeSelect6 with a page code of 1

- ModeSelect had an off-by-one error in computing the length

Fixes #1402
closed 2024-01-09 by rdmark