PiSCSI/piscsi #1415: Communication error (broken pipe) when attaching/detaching a device

uweseimet, 2024-01-15:
This bug can affect any remote interface client, not only scsictl.

When attaching/detaching a device an error is reported because piscsi sends two protobuf responses instead of only the one with the current device list:
```
>scsictl -c a -i 0 -f test.hds
```
Root cause is a broken pipe because scsictl closes the socket after receiving the first (wrong) response, so that piscsi cannot send the second (correct) response:
```
[2024-01-15 16:24:30.999] [info] Attached SCHD 0:0
[2024-01-15 16:24:31.008] [warning] Can't write protobuf message data
```

uweseimet, 2024-01-15:
@rdmark The issue_1415 branch fixes this bug. (Just one line has changed.)

I noticed that at least on my PC the unit tests are failing with the current develop branch:
```
[ RUN      ] ScsiCdTest.ModeSelect
make: *** [Makefile:194: test] Aborted
```
Whatever the reason, this is not related to the changes for this ticket.

rdmark, 2024-01-16:
Thanks for the fix. It has been merged to develop now.

So if assertions are enabled only in debug mode, I propose we run the CI job for unit tests in debug mode: https://github.com/PiSCSI/piscsi/pull/1417

uweseimet, 2024-01-16:
Running the tests in debug mode also has the advantage that compiling does not take that long.

closed 2024-01-16 by uweseimet